MickMake/GoSungrow #2: Inverter data point

danielrdeoliveira, 2022-04-26:
@MickMake , incredible work! 
I've managed to gather some data of our power plants through this endpoint: '/v1/commonService/queryMutiPointDataList'.
The thing is, it works only for the ps_key xxxxxxx_11_0_0; where xxxx is the psid (the power station id), 11 seems to stand for the device type (in this case: 'unit', i.e. the power station itself) and the 0s probably channel id or something.
I have been trying to get the same data but for each inverter (our power plants have more than one inverter each - there it differs from yours).
In order to get the devices, I tried this endpoint: "/v1/devService/queryDeviceListForApp"
one example of response was: "ps_key" : "1112873_1_2_1", for a given inverter (type 1 - 'inverter', channeld etc)
but when I pass the inverter ps_key as parameter it doesn't work (null response). 
Same for this endpoint: '/v1/commonService/queryDevicePointMinuteDataList'.
It only works with ps_key = "1112873_11_0_0".
Do you have any idea how I could solve it? I've tried changing several parameters on the requests, no success so far.
Any help would be appreciated, although you have already helped as a great deal!!

MickMake, 2022-05-04:
Hey mate,
Yup, I suspect that there's another field that's required.

I was looking for someone who had more than one device to help debugging issues.

So I might add a debug option where you can set your own json request, (independent of GoLang structures).

MickMake, 2022-10-13:
Hey Daniel,
The latest version now supports more than one unit.

All the high level commands will default to searching for a list of all possible ps_ids and use those when fetching data.

closed 2022-10-13 by MickMake