MickMake/GoSungrow #3: Can't install

Azza1979, 2022-04-30:
Failed to install add-on
The command '/bin/ash -o pipefail -c chmod a+x /usr/local/bin/run.sh /usr/local/bin/getUrl.sh && /usr/local/bin/getUrl.sh' returned a non-zero code: 1

MickMake, 2022-05-04:
What device are you installing on?
Is it a Pi, PC, Mac, etc?

Azza1979, 2022-05-04:
Rpi 4

MickMake, 2022-05-11:
Which version of HASSIO?

I'm running
- Home Assistant Core: 2022.4.2
- Home Assistant Operating System: 7.5
- Mosquitto broker: 6.0.1

Azza1979, 2022-05-11:
Home Assistant Core: 2022.5.1
Home Assistant Operating System: 7.6
Mosquitto broker: 6.1.1

GazAt105, 2022-06-06:
Same issue.
Hardware: Odroid XU4 (armv7l)
Home Assistant Supervised: core-2022.6.2
OS: Linux 5.4.191-242 - Ubuntu 20.04.4 LTS
Mosquitto broker: 6.1.2

MickMake, 2022-06-08:
Yup, definitely an issue with the way I determine which binary to pull from GitHub.
Fortunately, I have pretty much every SBC on the planet - so I'll test.

arttwentyfourseven, 2022-07-05:
Any luck with this, mate?

MickMake, 2022-11-21:
I've been working on v3 of GoSungrow which is a major update that adds a lot of features.

MickMake, 2022-12-15:
v3.0.1 of the HASSIO addon has been released - https://github.com/MickMake/HomeAssistantAddons
Go check it out there.

closed 2022-12-15 by MickMake

Azza1979, 2022-12-15:
Still getting that same error mate. 
[image]

knzadmin, 2023-06-17:
same problem. any news on this issue?

Jogi284, 2023-10-30:
Hi,

I also have this issue. has someone found a solution for this?

nithikalisamy, 2023-12-11:
Hi, I am also facing similar issue as above. Please suggest if any workarounds available.

"The command '/bin/ash -o pipefail -c chmod a+x /usr/local/bin/setup.sh && /usr/local/bin/setup.sh' returned a non-zero code: 1"

nithikalisamy, 2023-12-11:
> v3.0.1 of the HASSIO addon has been released - https://github.com/MickMake/HomeAssistantAddons Go check it out there.

This issue is occurring again, please see if it can be addressed. Thanks.