MickMake/GoSungrow #4: Update 8 HA --> S6-Overlay 3.x update

antonyclarage, 2022-05-20:
Hello,

With the new 8.x update of HA Core

The addon won't start.

Error: s6-overlay-suexec: fatal: can only run as pid 1

See also: https://developers.home-assistant.io/blog/2022/05/12/s6-overlay-base-images/

jblattner-github, 2022-10-02:
Hello, i meet same Error,
Any news here? 
using Core Version 2022.9.6

Ralphy140, 2022-10-24:
Hey Mick, Still seeing this issue, I really appreciate the work you are putting in

JoakimS4, 2022-11-01:
Hello, maybe i am doing something wrong here. I see that you have updated to 2.3.3. But on Home Assistant in the add-on store i can still only find 2.2.0. Am i looking in the wrong place or how do i update?

Best regard, Jocke

MickMake, 2022-11-04:
Hi guys,
I've completely refactored a lot of the code and this will be officially released as v3.0.0.
Any other releases are just interim ones for me to keep track of changes.

This next release will now:
- support better api changes. This allows a much quicker change in my code when the api changes.
- multiple device support. Previous versions only allowed query of one device.
- improved cli for queries. Now can fetch endpoints in either "list" or "table" formats, the latter showing point data in a similar fashion to pivot tables in spreadsheets.
- multiple output formats supported - not only csv, png and ascii, but markdown, xml, HTML, sql.
- lots of bug fixes. :-).

jblattner-github, 2022-11-21:
I have running V 2.2.1, where to find the latest version 3.x?

MickMake, 2022-11-21:
I've been working on v3 of GoSungrow before updating this hassio addon.
V3 is a major update that adds a lot of features.
I didn't provide a release for the last version.

MickMake, 2022-12-15:
v3.0.1 of the HASSIO addon has been released - https://github.com/MickMake/HomeAssistantAddons
Go check it out there.

closed 2022-12-15 by MickMake