MickMake/GoSungrow #5: Using European Server?

hopfi2k, 2022-05-28:
Does GoSungrow support access via European Server as well? If so, how to specify EU-Server?

Using the default server (host) gives me a 

`ERROR: Account does not exist '账号不存在'`

MickMake, 2022-12-17:
Could you try the latest version just released? v3.0.0

hopfi2k, 2022-12-17:
Tried latest release and getting:

```
./GoSungrow api login                                                                                       
Error: API httpResponse is 404 Not Found
Usage:
  GoSungrow api login [flags]

Examples:
	GoSungrow api login  

Flags: Use "GoSungrow help flags" for more info.

Additional help topics:

ERROR: API httpResponse is 404 Not Found
```

host is correctly set to: https://api.isolarcloud.eu

Created the config file before of course...

Any idea?

DrLEEi, 2022-12-17:
I myself use a European server.
It didn't work for me with the previous version either, but the current version (Current version: 3.0.1) started and works!
In my Home Assistant configuration: https://gateway.isolarcloud.eu

I can extract real-time and daily data.

I am very grateful for the work, because until now I have not been able to do this with anything else.
Unfortunately, the ModbusTCP communication did not work either.

hopfi2k, 2022-12-18:
Changing host to https://gateway.isolarcloud.eu did the trick. Many thanks!

MickMake, 2022-12-18:
Fantastic! Great to hear it's working well.

closed 2022-12-18 by MickMake