MickMake/GoSungrow #6: Addon will not start - fatal error

AUSPhoenix78, 2022-06-12:
The addon won't start.

Error: s6-overlay-suexec: fatal: can only run as pid 1

Could we please get this fixed asap.

joergtimm, 2022-06-12:
Same Problem

Kawobei, 2022-06-17:
Same Where does it even install to?

AUSPhoenix78, 2022-06-30:
have no idea, but i had another plugin that did the same thing, developer done some work and we got an update, i think this developer might be on leave or something lol i have got sick of waiting so i am using MQTT Broker and TCPtoMQTT addon along with Lovelace Power Flow card and it works a treat now, i am just going to sit back and wait patiently for this project to be updated and in the mean time use my work around, we need to remember that this is all free work so i am willing to wait.

k158, 2022-08-22:
@AUSPhoenix78, any chance you can detail what you did?

mikkaat, 2022-09-02:
> 

@AUSPhoenix78 I would also beinterested in how you got this workaround to function. Thanks

MickMake, 2022-09-08:
Hi guys,
I'll be back next week, so can check it out. I suspect some update in HA has caused the issue.
It's working fine on my current version of HA that I've been upgrading over time - so something with a fresh install.

snaven1, 2022-09-27:
I have the same problem, is this any closer to be resolved. Thanks

Kawobei, 2022-10-11:
> CAUTION: This email originated from outside your organization. Exercise caution when opening attachments or on clicking links from unknown senders.

OFFICIAL

Can you detail how to do your work around a little more please? It might help a few of us get it going while we wait.

OFFICIAL

RTO Code: 4693 CRICOS no. 01985A
This email and any attachments may contain privileged and confidential information. If you are not the intended recipient, you may not distribute reproduce this e-mail or the attachments. If you have received this message in error, please notify us by return e-mail.

Please consider the environment before printing this email.

Ralphy140, 2022-10-24:
I have the same issue. Someone has forked the code as described above to resolve the PID issue. but still getting errors with $home not defined. Would be great if this can be resolve in the main branch :)

MickMake, 2022-12-15:
v3.0.1 of the HASSIO addon has been released - https://github.com/MickMake/HomeAssistantAddons

This fixes a lot of issues. Go check it out there.

closed 2022-12-15 by MickMake