MickMake/GoSungrow #8: Any command I run say "Error: option 'PsKey' is empty"

hdymling, 2022-08-25:
I cannot see that we can write a PsKey in the config file and also how to find the plants PsKey.
We have a setup with four inverters added to our account as a plant.

Any ideas how to solve this?

MickMake, 2022-09-08:
I only have one inverter, so it resolves on the first one found.
I'll be updating this next week and give the ability to perform some dumps so I can do debugging.

hdymling, 2022-09-26:
Hi,

GoSungrow api get getPsList
_<  I removed the logon info >_
{
        "pageList": null,
        "rowCount": 0
}

MickMake, 2022-10-07:
Hi, I'm just about to push a major release that fixes this issue.

My inverter actually died and they replaced it with a completely new system. So I got a chance to debug multi-inverter setups.

closed 2022-10-07 by MickMake