MickMake/GoSungrow #9: Method Not Allowed

Hickori, 2022-09-15:
Hi!

I am trying to use postman to make an interface, but I get this response:

```
{
    "error": "Method Not Allowed",
    "message": "Request method 'GET' not supported",
    "path": "/getCloudList",
    "req_serial_num": "2022091605354858891fde42b2eb84cb",
    "status": 405,
    "timestamp": "2022-09-16 00:46:10"
}
```

Thanks in advance!
Best regards Max

Hickori, 2022-09-16:
I have done some more error checking and my conclusion is that my POST requests gets this error:

```
{
    "result_code": "000",
    "result_data": null,
    "result_msg": "er_unknown_exception"
}
```

Any idea?

MickMake, 2022-10-07:
It's a pretty simple thing. The process is:
1. Obtain a token from /v1/userService/login - have a look at iSolarCloud/AppService/login/data.go
2. Use that token in all subsequent calls.

It is possible to refresh the token, but it seems to last from a couple of days to a month.

Hickori, 2022-10-15:
Thanks for the response!

I still cannot get it to work properly, I get the same error, my postman looks like this:

geoffishere, 2022-11-04:
It should look like this

[image]

Hickori, 2023-02-07:
Worked perfecly!

I have another question :D, which requests do you do to get this data 
[image]

I think it looks amazing and Id love to build it into my home controlelr :D