MickMake/GoSungrow #10: Build Instructions

entropical, 2022-09-24:
Hey, 

I have a sungrow, and just rebuild my HomeAssistant. I have HA in Docker, so no add-on store, so I am giving this project a go.

I was a bit confused initially, because GoSungrow (i.e. the binary) did not exist. I have it now, via installing go 1.18, go build, go install.

Might be helpful to include these steps in your README.

Cheers, Brock

MickMake, 2022-10-07:
Thanks. The goal is for people to not have to worry about building.
I'm about to push a major release along with pre-built binaries.... stay tuned.

closed 2022-10-07 by MickMake

mmierke, 2023-07-11:
@MickMake: Hi, it would be super cool if you could provide a standalone docker image to run goSungrow as a standalone container for those of us who run HA in a docker container. Being not a docker expert it is kinda hard for me to start from the Dockerfile.

EvoSems, 2024-02-01:
> @MickMake: Hi, it would be super cool if you could provide a standalone docker image to run goSungrow as a standalone container for those of us who run HA in a docker container. Being not a docker expert it is kinda hard for me to start from the Dockerfile.

yes please! or can you give us some instruction to build the docker file for rpi. I have been using methods as follows.

sudo docker run -d --name hass-configurator --privileged --restart=unless-stopped -v /home/evo/hass-config:/config -v /home/evo/homeassistant:/hass-config --network=host causticlab/hass-configurator-docker:latest

currently stuck on what to do on this page and it seems like the perfect project for my use case