MickMake/GoSungrow #11: Error building new version

hdymling, 2022-10-09:
Thanks for the work you are doing!
I have tried to build the new version, but when building it seems there are missing one thing in the "GoUnify" package.
This is the "/GoUnify/cmdPathiSolar"
The error looks like bellow!
Cloud\api\file.go:9:2: github.com/MickMake/GoUnify/cmdPath@v0.0.0-00010101000000-000000000000: replacement directory ../GoUnify/cmdPath does not exist

When checking GoUnify I cannot find cmdPath!

MickMake, 2022-10-13:
Hi,
I recently moved a large chunk of code off on to its separate repo.
So there may be a few teething issues.

I'll push the latest of GoUnify to GitHub and if you remove all the redirects in go.mod you'll be good to go.

There's always the releases though!

Let me know if I'm missing an architecture.

hdymling, 2022-10-16:
Thanks,
I could now build without a problem.
However I got a new problem, will register a new Issue for this.
Thanks a lot for your work!!!!

closed 2022-10-16 by hdymling