MickMake/GoSungrow #13: go install to build from source

jpillora, 2022-11-01:
It'd be awesome if we could install from source with `go install github.com/MickMake/GoSungrow@v2.3.0`. It looks like the replace directives enforce a local dependency?

MickMake, 2022-12-21:
I'll tackle this properly in v3.0.3 release.

I've just released v3.0.2 which addresses a few important bugs.

MickMake, 2022-12-23:
Had to shift to v3.0.4 - not enough time before Christmas!

jpillora, 2022-12-23:
No worries Mick, have a good break!

monojk, 2023-05-12:
I have a problem installing the code on a Raspberry Pi:
    go install github.com/MickMake/GoSungrow
    can't load package: package github.com/MickMake/GoSungrow: cannot find package "github.com/MickMake/GoSungrow" in any of:
	/usr/lib/go-1.11/src/github.com/MickMake/GoSungrow (from $GOROOT)
	/home/pi/go/src/github.com/MickMake/GoSungrow (from $GOPATH)

Paraphraser, 2023-05-13:
I'm only interested in `go build` rather than `go install` so I'm not sure if this will help. Environment:

- 4GB Raspberry Pi 4 Model B Rev 1.1 running Debian GNU/Linux 11 (bullseye) as full 64-bit OS
- Clean install of compiler
- Clean clone of GoSungrow

```
pi@sec-dev:~$ uname -a
Linux sec-dev 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

pi@sec-dev:~$ SHASUM=105889992ee4b1d40c7c108555222ca70ae43fccb42e20fbf1eebb822f5e72c6
pi@sec-dev:~$ IMAGE=go1.20.4.linux-arm64.tar.gz
pi@sec-dev:~$ wget -q "https://go.dev/dl/$IMAGE"
pi@sec-dev:~$ shasum -a 256 -c <<< "$SHASUM *$IMAGE"
go1.20.4.linux-arm64.tar.gz: OK

pi@sec-dev:~$ sudo rm -rf /usr/local/go
pi@sec-dev:~$ sudo tar -C /usr/local -xzf "$IMAGE"

pi@sec-dev:~$ which go
/usr/local/go/bin/go
pi@sec-dev:~$ go version
go version go1.20.4 linux/arm64

pi@sec-dev:~$ mkdir go
pi@sec-dev:~$ file $GOPATH
/home/pi/go: directory

pi@sec-dev:~$ mkdir go-projects
pi@sec-dev:~$ cd go-projects
pi@sec-dev:~/go-projects$ git clone https://github.com/MickMake/GoSungrow.git
Cloning into 'GoSungrow'...
remote: Enumerating objects: 60387, done.
remote: Counting objects: 100% (975/975), done.
remote: Compressing objects: 100% (351/351), done.
remote: Total 60387 (delta 660), reused 920 (delta 622), pack-reused 59412
Receiving objects: 100% (60387/60387), 32.82 MiB | 9.97 MiB/s, done.
Resolving deltas: 100% (35138/35138), done.
Updating files: 100% (2505/2505), done.

pi@sec-dev:~/go-projects$ cd GoSungrow
pi@sec-dev:~/go-projects/GoSungrow$ go build
go: downloading github.com/MickMake/GoUnify/Only v0.0.0-20221125023651-ff4a37b1928a
go: downloading github.com/MickMake/GoUnify/Unify v0.0.0-20221125023651-ff4a37b1928a
go: downloading github.com/MickMake/GoUnify/cmdConfig v0.0.0-20221125023651-ff4a37b1928a
go: downloading github.com/MickMake/GoUnify/cmdHelp v0.0.0-20221125023651-ff4a37b1928a
go: downloading github.com/MickMake/GoUnify/cmdLog v0.0.0-20221125023651-ff4a37b1928a
go: downloading github.com/MickMake/GoUnify/cmdPath v0.0.0-20221125023651-ff4a37b1928a
go: downloading github.com/eclipse/paho.mqtt.golang v1.4.2
go: downloading github.com/go-co-op/gocron v1.18.0
go: downloading github.com/spf13/cobra v1.6.1
go: downloading github.com/spf13/viper v1.14.0
go: downloading github.com/wcharczuk/go-chart/v2 v2.1.0
go: downloading github.com/olekukonko/tablewriter v0.0.5
go: downloading go.pennock.tech/tabular v1.1.3
go: downloading github.com/MickMake/GoUnify/cmdCron v0.0.0-20221125020154-1b15b8d20735
go: downloading github.com/MickMake/GoUnify/cmdDaemon v0.0.0-20221125023651-ff4a37b1928a
go: downloading github.com/MickMake/GoUnify/cmdShell v0.0.0-20221125023651-ff4a37b1928a
go: downloading github.com/MickMake/GoUnify/cmdVersion v0.0.0-20221125023651-ff4a37b1928a
go: downloading github.com/ivanpirog/coloredcobra v1.0.1
go: downloading github.com/spf13/pflag v1.0.5
go: downloading github.com/MichaelMure/go-term-markdown v0.1.4
go: downloading github.com/robfig/cron/v3 v3.0.1
go: downloading golang.org/x/sync v0.1.0
go: downloading github.com/gorilla/websocket v1.4.2
go: downloading golang.org/x/net v0.0.0-20221014081412-f15817d10f9b
go: downloading gopkg.in/yaml.v2 v2.4.0
go: downloading github.com/agrison/mxj v0.0.0-20160310142625-1269f8afb3b4
go: downloading github.com/tealeg/xlsx v1.0.5
go: downloading github.com/fsnotify/fsnotify v1.6.0
[... 27 lines cut ...]
go: downloading github.com/pelletier/go-toml v1.9.5
go: downloading gopkg.in/yaml.v3 v3.0.1
go: downloading github.com/rivo/uniseg v0.2.0
go: downloading github.com/fatih/structs v1.1.0
go: downloading github.com/MichaelMure/go-term-text v0.3.1
go: downloading github.com/alecthomas/chroma v0.7.1
go: downloading github.com/eliukblau/pixterm/pkg/ansimage v0.0.0-20191210081756-9fb6cf8c2f75
go: downloading github.com/gomarkdown/markdown v0.0.0-20191123064959-2c17d62f5098
go: downloading github.com/kyokomi/emoji/v2 v2.2.8
go: downloading github.com/abiosoft/readline v0.0.0-20180607040430-155bce2042db
go: downloading github.com/flynn-archive/go-shlex v0.0.0-20150515145356-3f9db97f8568
go: downloading github.com/mattn/go-isatty v0.0.14
go: downloading github.com/inconshreveable/go-update v0.0.0-20160112193335-8152e7eb6ccf
go: downloading github.com/ulikunitz/xz v0.5.9
go: downloading github.com/mattn/go-colorable v0.1.12
go: downloading github.com/google/go-querystring v1.0.0
go: downloading golang.org/x/crypto v0.0.0-20220525230936-793ad666bf5e
go: downloading github.com/dlclark/regexp2 v1.1.6
go: downloading github.com/disintegration/imaging v1.6.2
go: downloading github.com/lucasb-eyer/go-colorful v1.0.3
go: downloading github.com/danwakefield/fnmatch v0.0.0-20160403171240-cbb64ac3d964

pi@sec-dev:~/go-projects/GoSungrow$ file GoSungrow
GoSungrow: ELF 64-bit LSB executable, ARM aarch64, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-aarch64.so.1, Go BuildID=60gysPfCJSuKtACy42o1/vJNB-XsBkL0KzlK05j5N/ccUaTydsvLUpta78E94G/zL2fXqQEsi3kvSyXGC_r, with debug_info, not stripped
```

jpillora, 2023-05-13:
From a go module directory

go install is equivalent to go build -o $GOPATH/bin/<dir name>

From outside a go module, you need the version

go install ***@***.*** (or v3xxxx tag)

monojk, 2023-05-13:
So I had an old go version and installed the newest (go1.20.4), but unfortunately still no success:
```
pi@rp4b:~ $ go install github.com/MickMake/GoSungrow@v3.0.4
go: github.com/MickMake/GoSungrow@v3.0.4: github.com/MickMake/GoSungrow@v3.0.4: invalid version: module contains a go.mod file, so module path must match major version ("github.com/MickMake/GoSungrow/v3")

pi@rp4b:~ $ go install github.com/MickMake/GoSungrow@v3
go: github.com/MickMake/GoSungrow@v3: no matching versions for query "v3"
```

Paraphraser, 2023-05-13:
Did you recursively remove the **entire** `/usr/local/go` folder? Most web tutorials omit that step but I've found I get really weird errors if I just unpack the tar over the top of an existing installation.

I don't claim to understand why it should be necessary. It's just a fix that seems to work.

jpillora, 2023-05-13:
Yep, need to rm -rf /usr/local/go

Here's my `install-go` script that i use to install/upgrade go

jpillora, 2023-05-13:
https://github.com/jpillora/dotfiles/blob/main/bin/install-go

monojk, 2023-05-13:
I did, I followed the instructions in https://go.dev/doc/install
```
sudo su
rm -rf /usr/local/go && tar -C /usr/local -xzf go1.20.4.linux-amd64.tar.gz
export PATH=$PATH:/usr/local/go/bin
```

Paraphraser, 2023-05-13:
You could also try erasing whatever $GOPATH points too.

monojk, 2023-05-13:
Hm, GOPATH isn't mentioned anymore in the new install instructions. Yet, I had it set for the old Go version. Now I unset it (and don't set it again in .profile). In the go directory (~/go as was pointed to by previous GOPATH) I renamed the 2 subdirectories pkg and src with a .0 suffix, but kept the bin subdir, because I have 2 Go programs there (they still work with the new Go version).
Yet, it doesn't help (pkg sub dir is also created anew upon the go install), got the same error as before.
What about the message about a mismatch of the major version?
```
pi@rp4b:~ $ go install github.com/MickMake/GoSungrow@v3.0.4
go: github.com/MickMake/GoSungrow@v3.0.4: github.com/MickMake/GoSungrow@v3.0.4: invalid version: module contains a go.mod file, so module path must match major version ("github.com/MickMake/GoSungrow/v3")
```

Paraphraser, 2023-05-13:
I'm really only a casual dilettante user of go. As in about the second page in the tutorial after the "hello world". Keeping that disclaimer in mind, as I read that error, it is saying that the `go.mod` file has to be in a subdirectory of the repo named `v3`. The `go.mod` file is at the root of the repo and there is no `v` anything subdirectory. I interpret all that as implying that this `go install <github URL>` syntax can't actually work until the GitHub side of things is fixed.

> You could probably test this by forking the repo and making the changes yourself.

My earlier replies were based on (a) making sure I had the latest and greatest compiler and (b) that I could clone and compile from scratch - so you'd know that someone else could at least compile successfully on a Raspberry Pi.

The idea that GOPATH might be deprecated is something I wasn't aware of. I just observed that when I erased `~/go`, it was repopulated during the build. So 🤷‍♂️

In one of your earlier comments you showed the output from `go version` as "go version go1.20.4 linux/arm" which, to me, says the compiler matching ARM 32-bit system. Then you showed installing from "go1.20.4.linux-amd64.tar.gz" which is AMD (Intel) not ARM.

Although I would've expected different errors, maybe that's something to double-check:

- is this a Pi? If yes, then it's "Linux" + either "ARMv6" or "ARM64".
- is it a full 32-bit or mixed mode 32-bit user with 64-bit kernel? If yes then it's "Linux+ARMv6".
- is it full 64-bit? If yes, then it's "Linux+ARM64".

I haven't done any experiments to confirm this but my understanding is the 32-bit compiler (ARMv6) will run on anything (full 32-bit, mixed mode, full 64-bit) whereas the 64-bit compiler (ARM64) needs a full 64-bit system.

Is there any particular reason why you can't do what I did, and just finish off with a `go install`? Again, I haven't tested it but it seems to me that if a `build` works, an `install` likely will too.

monojk, 2023-05-13:
@Paraphraser I  unfortunately did cite the go1.20.4.linux-amd64.tar.gz directly from the go install webpage. I really used the go1.20.4.linux-armv6l.tar.gz. Sorry for the confusion.

I have run the `git clone https://github.com/MickMake/GoSungrow.git` now and the following commands you listed. Now I was able to build, check and call the command. Thanks for your support!
```
pi@rp4b:~/go-projects/GoSungrow $ file GoSungrow
GoSungrow: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, Go BuildID=X8VACstuNbJNYTpSl_PM/vvM5vEN6YAwEcwlbGlFK/WA6lCjnQaHR2HvGdrRdV/iBk0xNnZFI1kv6boxbx4, with debug_info, not stripped
```