MickMake/GoSungrow #14: Small programmatic example

jpillora, 2022-11-01:
From an existing Go program, I'd like to get my current Sungrow inverter's current kw/hr output. I could use the binary and run a shell script, though it'd be awesome if I could call a function in my Go program. Is there a small example of this?

This is what I've tried, but it crashes

```go
package main

import (
	"GoSungrow/iSolarCloud"
	"GoSungrow/iSolarCloud/AppService/login"
	"log"
)

func main() {
	sg := iSolarCloud.NewSunGro("https://augateway.isolarcloud.com", "/tmp/")
	if err := sg.Login(login.SunGrowAuth{
		UserAccount:  "MY_EMAIL",
		UserPassword: "MY_PASSWORD",
	}); err != nil {
		log.Fatal(err)
	}
	ep := sg.GetEndpoint("queryDeviceList")
	rargs := ep.GetRequestArgNames()
	log.Printf("args:%s\n", rargs)
	// var request iSolarCloud.SunGrowDataRequest
}
```

MickMake, 2022-11-02:
I'm in the middle of doing a major rewrite - adding that feature wouldn't be too hard. So I'll tack it on to the end of release 2.0.0.

But if you're impatient - you'll need to follow:
- Setup
```
ca.Api.SunGrow = iSolarCloud.NewSunGro(ca.Api.Url, ca.CacheDir)
ca.Error = ca.Api.SunGrow.Init()
```
The Init() will load all the endpoints up. If you don't want to do that, then you'll need to load them up manually.
Each endpoint is managed within itself, so you can do that easily.

- login
```
ca.Error = ca.Api.SunGrow.Login(login.SunGrowAuth{
	AppKey:       ca.Api.AppKey,
	UserAccount:  ca.Api.Username,
	UserPassword: ca.Api.Password,
	TokenFile:    ca.Api.ApiTokenFile,
	Force:        false,
})
```

- Query
Have a look at `highlevel.go`, which shows examples of how to query some endpoints.
```
ep := sg.GetByStruct(
"AppService.queryMutiPointDataList",
queryMutiPointDataList.RequestData {
	// PsId:           api.SetIntegerValue(psId),
	PsId:           psId,
	PsKey:          pskeys,
	Points:         valueTypes.SetStringValue(pointNames.PrintPoints()),	// @TODO - Fixup!
	MinuteInterval: valueTypes.SetStringValue("5"),
	StartTimeStamp: valueTypes.SetStringValue(when.GetDayStartTimestamp()), // @TODO - Fixup!
	EndTimeStamp:   valueTypes.SetStringValue(when.GetDayEndTimestamp()),	// @TODO - Fixup!
},
DefaultCacheTimeout,
)
```

Note, you'll need to Assert the endpoint to be able to see the locally scoped data within the endpoint interface.
```
data := queryMutiPointDataList.Assert(ep)
data.GetData()
table := data.GetPointDataTable(pointNames)

```

MickMake, 2022-11-02:
BTW... I recently found a new endpoint which returns static data on what data points are supported for your device.

```
GoSungrow data table WebIscmAppService.modelPointsPage DeviceModelId:714 DeviceType:14
```