MickMake/GoSungrow #15: Failure to start in HA.

JoakimS4, 2022-11-05:
Hello! I have today updated youre application in Home Assistant, but when i try to start the addon it gives me this. 

Can you please help me understand what is going on here.

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[09:45:24] INFO: Setting up GoSungrow config ...
parse error: Expected string key before ':' at line 1, column 4
[09:45:24] ERROR: Unknown HTTP error occured
[09:45:25] INFO: Writing GoSungrow config ...
ERROR: $HOME is not defined
[09:45:25] INFO: Login to iSolarCloud using gateway null ...
ERROR: $HOME is not defined
[09:45:25] INFO: Syncing data from gateway null ...
ERROR: $HOME is not defined
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Best regards, Jocke

vinayp84, 2022-11-07:
same issue on this end, sadly.

nollettnoll, 2022-11-08:
same here after last update, worked fine before, possible to downgrade?

or fix it, my config, any hints?

sungrow_host: gateway.isolarcloud.eu
sungrow_user: <removed>
sungrow_password: <removed>
sungrow_mqtt_host: 192.168.0.167
sungrow_mqtt_port: "1883"
sungrow_mqtt_password: <removed>
sungrow_mqtt_user: homeassistant
sungrow_debug: false
sungrow_appkey: <removed>
sungrow_timeout: 60
log_level: debug

snaven1, 2022-11-09:
Running the latest version of home assistant (supervised) 20221108.0   This HA install update seems to have resolved the S6 docker container problem (which I had previously had). On trying to start the Sungrow Module, I am now getting the same problem mentioned above. Additionally, on trying to do a full uninstall/reinstall of the module, HA is only finding the V2.2.1 module, not the 2.3 module is indicated on Github. Anyone with any suggestions, would be appreciated.

Die-Sebis, 2022-11-09:
Hi, im running HA in Version 2022.11.2 and GoSungrow in the lates Version 2.2.1.
I see in the Logs the issue : --> parse error: Expected string key before ':' at line 1, column 4
[16:41:35] ERROR: Unknown HTTP error occured

changing the config "sun grow_host"  like : 
https://gateway.isolarcloud.eu 
or
https://portaleu.isolarcloud.com
or
https://augateway.isolarcloud.com

or something stupid : https://HelloServer.eu

dos not change the log comment.
Does anyone have an idea what I could change ?

iliapis, 2022-11-12:
same here today

Ralphy140, 2022-11-30:
I have the same issue as above, I know V3 is being worked on but would be keen to see if we can find a fix for the current version while we wait.

MickMake, 2022-12-15:
v3.0.1 of the HASSIO addon has been released - https://github.com/MickMake/HomeAssistantAddons

This fixes a lot of issues. Go check it out there.

closed 2022-12-15 by MickMake