MickMake/GoSungrow #16: MQTT error: invalid character '"' after object key:value pair

jimmybish, 2022-12-15:
Hi! 👋 
As a round of perfect timing, I found and decided to give GoSungrow a try approximately 20 mins after you updated to v3.0 😁 
Unfortunately, I've hit an error when attempting to sync MQTT to HA.

The connection works and I can get devices:
```
./GoSungrow show ps list
┏━━━━━━━━━━━━━━━━━━┳━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━┓
┃ Ps Key           ┃ Ps Id   ┃ Device Type ┃ Device Code ┃ Channel Id ┃ Serial #    ┃ Factory Name ┃ Device Model   ┃
┣━━━━━━━━━━━━━━━━━━╇━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━┫
┃ [FILTERED]    │ 1181868 │ 1           │ 1           │ 2          │ [FILTERED] │ SUNGROW      │ SG5K-D         ┃
┃ [FILTERED]    │ 1181868 │ 1           │ 1           │ 1          │ [FILTERED] │ SUNGROW      │ SG5K-D         ┃
┃ [FILTERED] │ 1181868 │ 22          │ 247         │ 2          │ [FILTERED] │ SUNGROW      │ WiFi V31       ┃
┃ [FILTERED] │ 1181868 │ 22          │ 247         │ 1          │ [FILTERED] │ SUNGROW      │ WiFi V31       ┃
┃ [FILTERED]    │ 1181868 │ 7           │ 1           │ 1          │ [FILTERED] │ SUNGROW      │ SG Smart Meter ┃
┗━━━━━━━━━━━━━━━━━━┷━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━━━┷━━━━━━━━━━━━━━━━┛
```

But when I go to perform an MQTT sync, I receive the following:

```
./GoSungrow mqtt sync
Email:	[FILTERED]@gmail.com
Create Date:	Thu Nov 17 11:47:54 CST 2022
Login Last Date:	2022-12-15 10:45:52
Login Last IP:
Login State:	1
User Account:	[FILTERED]
User Id:	[FILTERED]
User Name:	[FILTERED]
Is Online:	false
Token:	[FILTERED]
Token File:	/home/ubuntu/.GoSungrow/AppService_login.json
2022/12/15 02:52:39 Connecting to MQTT HASSIO Service...
2022/12/15 02:52:39 Connecting to SunGrow...
2022/12/15 02:52:39 Found SunGrow 5 devices
2022/12/15 02:52:39 Caching Sungrow metadata...
Error: invalid character '"' after object key:value pair
```

My config, if that helps:
```
{
  "appkey": "[FILTERED]",
  "config": "/home/ubuntu/.GoSungrow/config.json",
  "debug": true,
  "host": "https://augateway.isolarcloud.com",
  "mqtt-host": "homeassistant.home.local",
  "mqtt-password": "",
  "mqtt-port": "1883",
  "mqtt-user": "",
  "out": "",
  "password": "[FILTERED]",
  "quiet": false,
  "save": false,
  "timeout": "30s",
  "token-expiry": "2022-12-15T10:45:52",
  "user": "[FILTERED]@gmail.com"
}
```

MickMake, 2022-12-15:
Yeah, I made a slight error with the `mqtt_endpoints.json` file. If you add a , just between the the two last array elements, it'll fix it.

```
	}
	"getPsDetail": {
		"include": [
			"virtual.*"
		],
		"exclude": [
		]
	}
```

Change it to this:
```
	},
	"getPsDetail": {
		"include": [
			"virtual.*"
		],
		"exclude": [
		]
	}
```

jimmybish, 2022-12-16:
Ah, that'll do it! I even looked over that JSON file and the template in `cmd_mqtt.go` multiple times and didn't pick that up 🤦 

Fantastic, all working now. Thanks!

closed 2022-12-16 by jimmybish