MickMake/GoSungrow #17: ERROR: No internal MQTT service found. Please install Mosquitto broker

Surkel, 2022-12-15:
Hi,

i was happy to see that a new version was provided, but unfortunately it doesn't work for me. I got an error Message "ERROR: No internal MQTT service found. Please install Mosquitto broker" although Mosquito Broker is running:
[08:23:53] INFO: Successfully send discovery information to Home Assistant.
[08:23:53] INFO: Successfully send service information to the Supervisor.

Any Ideas?

Log from Addon:
[07:35:12] INFO: Login to iSolarCloud using gateway https://gateway.isolarcloud.eu ...
Email:	surkel@gmx.net
Create Date:	Wed Apr 06 20:56:21 CST 2022
Login Last Date:	2022-12-15 15:35:13
Login Last IP:	
Login State:	1
User Account:	ebzys9ma1g
User Id:	47700
User Name:	sg
Is Online:	false
Token:	<xyz>
Token File:	/data/.GoSungrow/AppService_login.json
[07:35:13] INFO: Syncing data from gateway https://gateway.isolarcloud.eu ...
No internal MQTT service found. Please install Mosquitto broker
[07:35:13] ERROR: No internal MQTT service found. Please install Mosquitto broker
parse error: Expected string key before ':' at line 1, column 4
[07:35:13] ERROR: Unknown HTTP error occured
host: 
password: 
port: 
username: 
[07:35:14] INFO: host: 
[07:35:14] INFO: password: 
[07:35:14] INFO: port: 
[07:35:15] INFO: username: 
__BASHIO_LIB_DIR=/usr/lib/bashio
PWD=/
GOSUNGROW_MQTT_HOST=core-mosquitto
GOSUNGROW_APPKEY=93D72E60331ABDCDC7B39ADC2D1F32B3
HOME=/data
__BASHIO_BIN=/usr/lib/bashio/bashio
GOSUNGROW_DEBUG=
CONFIG_PATH=/data/options.json
GOSUNGROW_PASSWORD=<xyz>
GOSUNGROW_USER=<xyz>
SHLVL=2
GOSUNGROW_HOST=https://gateway.isolarcloud.eu
PATH=/command:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
GOSUNGROW_TIMEOUT=60s
OLDPWD=/
_=/usr/bin/env
PING core-mosquitto (172.30.33.2): 56 data bytes
64 bytes from 172.30.33.2: seq=0 ttl=64 time=0.963 ms
64 bytes from 172.30.33.2: seq=1 ttl=64 time=0.195 ms

Surkel, 2022-12-15:
Update to Version 3.0.1..then I got no errors.

Last Lines of Log:
[10:13:52] INFO: Syncing data from gateway https://gateway.isolarcloud.eu ...
[10:13:52] INFO: GoSungrow exited without error ...

is it working now?
If Yes, how can I implement it?

MickMake, 2022-12-15:
OK. I had a few false starts with this release.

You'll need to:

1. Uninstall the addon.
2. Refresh the plugin cache.
3. Reinstall the addon.
4. Once installed configure, (make sure to enter in your HA username/password). You can optionally create a local user in HA for this.

Surkel, 2022-12-15:
thanks for quick response, but now we have another problem, see
https://github.com/MickMake/GoSungrow/issues/19

Surkel, 2022-12-15:
3.0.1 fixed

closed 2022-12-15 by Surkel