MickMake/GoSungrow #18: Still no PS are seen

hdymling, 2022-12-15:
Hi,
First I would like to thank you for all the work you are doing regarding this!

I have now updated to version 3.0. This is done on my windows 10 workstations as I plan to run and export daily graphs.
The bellow are what is happening when I run the list of PS. We have four inverters.
[image]

MickMake, 2022-12-15:
Hi @hdymling, could you try to pull the latest release again? I re-released as the same version by mistake.

hdymling, 2022-12-15:
I have now pulled the latest relase and recompiled, still same issue!

MickMake, 2022-12-15:
Hmmm. Interesting. What about the command `show ps tree`

Just wondering there might be some different result because you have 4 inverters.

hdymling, 2022-12-16:
running this command the result are as bellow
[image]

To valdiate the version I run the command "gosungrow version" and it shows 3.0.0, is this correct?

hdymling, 2022-12-20:
I could share the credentials if you want to debug this. At least i guess this is the most easy way to see what the issue are.
What do you think?

/Helge

MickMake, 2022-12-21:
Hmmm. I'd rather not do that. Could you tell me your solar install name? (That is the name you called it when installing. Or the name that appears when you use the iSolarCloud app.)

I can query using that.

hdymling, 2022-12-21:
The name is "20395 BRF Slottet".
I guess this is the name you want?

Didainius, 2022-12-31:
I have hit such problem as well. What I see is that if I use the account, that is used in my iPhone App -  it works well and lists devices. Now I was trying to use the account one can create for read only purposes - and then it is unable to list inverters.

hdymling, 2023-01-01:
I have now tried the HomeAssistant integration and getting the following in the log file.
Token File:	/data/.GoSungrow/AppService_login.json
2023/01/01 14:25:04 INFO: Connecting to MQTT HASSIO Service...
**2023/01/01 14:25:04 INFO: Connecting to SunGrow...
2023/01/01 14:25:05 INFO: Found SunGrow 0 devices
Error: identifier rejected
Usage:
  GoSungrow mqtt run [flags]
Aliases:
  run, 
Examples:
	GoSungrow mqtt run  
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: identifier rejected**
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping

hdymling, 2023-02-21:
[Didainius](https://github.com/Didainius)
> I have hit such problem as well. What I see is that if I use the account, that is used in my iPhone App - it works well and lists devices. Now I was trying to use the account one can create for read only purposes - and then it is unable to list inverters.

I have now got more access rights on Isolarcloud and our Plant, but still I cannot see any devices found.
Error message: "Found SunGrow 0 devices" Error: Identifier rejected"

Do you know what access rights you have when i worked?

Didainius, 2023-02-26:
@hdymling ,
There aren't much rights in the app - in my case I can only "share" with either `Management` or `Browse` rights. None of these worked in my case so I tried with the original main account I got from my provider - that one works.

zachkont, 2023-09-20:
I have the same issue, I only have 1 inverter in a Residential PV plant (my home installation) and `show ps list` returns an empty list. Using prebuilt binary for windows, version 3.0.7