MickMake/GoSungrow #19: API problems?

bartvanmoer, 2022-12-15:
Hi,

You did a terrific job, but still I can not get it working.
Maybe it is me who does something wrong.

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[10:59:07] INFO: Setting up GoSungrow config ...
parse error: Expected string key before ':' at line 1, column 4
[10:59:08] ERROR: Unknown HTTP error occured
[10:59:09] INFO: Writing GoSungrow config ...
Using config file '/data/.GoSungrow/config.json'
New config:
+-----------------+------------+-------------------------+--------------------------------+------------------------------------------------------------------+
|      FLAG       | SHORT FLAG |       ENVIRONMENT       |          DESCRIPTION           |                       VALUE (* = DEFAULT)                        |
+-----------------+------------+-------------------------+--------------------------------+------------------------------------------------------------------+
| --config        |            | GOSUNGROW_CONFIG        | GoSungrow: config file.        | /data/.GoSungrow/config.json                                     |
| --debug         |            | GOSUNGROW_DEBUG         | GoSungrow: Debug mode.         | false *                                                          |
| --quiet         |            | GOSUNGROW_QUIET         | GoSungrow: Silence all         | false *                                                          |
|                 |            |                         | messages.                      |                                                                  |
| --timeout       |            | GOSUNGROW_TIMEOUT       | Web timeout.                   | 1m0s                                                             |
| --user          | -u         | GOSUNGROW_USER          | SunGrow: api username.         | ****                                        |
| --password      | -p         | GOSUNGROW_PASSWORD      | SunGrow: api password.         | ****                                                        |
| --appkey        |            | GOSUNGROW_APPKEY        | SunGrow: api application key.  | ****                                 |
|                 |            |                         |                                | *                                                                |
| --host          |            | GOSUNGROW_HOST          | SunGrow: Provider API URL.     | https://portaleu.isolarcloud.com                                 |
| --token-expiry  |            | GOSUNGROW_TOKEN_EXPIRY  | SunGrow: last login.           |  *                                                               |
| --save          | -s         | GOSUNGROW_SAVE          | Save output as a file.         | false *                                                          |
| --mqtt-user     |            | GOSUNGROW_MQTT_USER     | HASSIO: mqtt username.         | homeassistant                                                    |
| --mqtt-password |            | GOSUNGROW_MQTT_PASSWORD | HASSIO: mqtt password.         | oChisi8ieX2aeso3Baqu8ro0ZaeNool0riez7miu1tici5ziiDoo1cai5AhCee2u |
| --mqtt-host     |            | GOSUNGROW_MQTT_HOST     | HASSIO: mqtt host.             | core-mosquitto                                                   |
| --mqtt-port     |            | GOSUNGROW_MQTT_PORT     | HASSIO: mqtt port.             |                                                             1883 |
+-----------------+------------+-------------------------+--------------------------------+------------------------------------------------------------------+
[10:59:09] INFO: Login to iSolarCloud using gateway https://portaleu.isolarcloud.com ...
Error: API httpResponse is 405 Method Not Allowed
Usage:
  GoSungrow api login [flags]
Examples:
	GoSungrow api login  
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: API httpResponse is 405 Method Not Allowed
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Surkel, 2022-12-15:
same Problem after these steps:

1. Uninstall the addon.
2. Refresh the plugin cache.
3. Reinstall the addon.
4. Once installed configure

bartvanmoer, 2022-12-15:
Nope, does nothing differtent.
Is it normal the API key is already filled in the cofiguration page?

Surkel, 2022-12-15:
still problems.
a parse error and identifier error

Parts of the Log:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[11:50:53] INFO: Setting up GoSungrow config ...
**parse error: Expected string key before ':' at line 1, column 4**
[11:50:54] ERROR: Unknown HTTP error occured
[11:50:55] INFO: Writing GoSungrow config ...
Using config file '/data/.GoSungrow/config.json'
.....
[11:50:56] INFO: Login to iSolarCloud using gateway https://gateway.isolarcloud.eu ...
Email:	<mail>
Create Date:	Wed Apr 06 20:56:21 CST 2022
Login Last Date:	2022-12-15 19:50:56
Login Last IP:	
Login State:	1
User Account:	<account>
User Id:	47700
User Name:	sg
Is Online:	false
Token:	<token>
Token File:	/data/.GoSungrow/AppService_login.json
[11:50:57] INFO: Syncing data from gateway https://gateway.isolarcloud.eu ...
2022/12/15 11:50:58 Connecting to MQTT HASSIO Service...
2022/12/15 11:50:58 Connecting to SunGrow...
2022/12/15 11:50:58 Found SunGrow 2 devices
**Error: identifier rejected**
Usage:
  GoSungrow mqtt sync [flags]
Aliases:
  sync, 
Examples:
	GoSungrow mqtt sync  
	GoSungrow mqtt sync  all
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
**ERROR: identifier rejected**
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

MickMake, 2022-12-15:
OK. That’s a whole lot closer.

   2022/12/15 11:50:58 Found SunGrow 2 devices

it found your 2 devices, but now we have another problem.

  ERROR: identifier rejected

JoakimS4, 2022-12-19:
Hello, i also have this problem.. Dont really know what i am doing wrong here..

I have tried different sungrow_host and still get an error.

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[06:18:10] INFO: Setting up GoSungrow config ...
parse error: Expected string key before ':' at line 1, column 4
[06:18:11] ERROR: Unknown HTTP error occured
[06:18:12] INFO: Writing GoSungrow config ...
Using config file '/data/.GoSungrow/config.json'
New config:
+-----------------+------------+-------------------------+--------------------------------+----------------------------------+
|      FLAG       | SHORT FLAG |       ENVIRONMENT       |          DESCRIPTION           |       VALUE (* = DEFAULT)        |
+-----------------+------------+-------------------------+--------------------------------+----------------------------------+
| --config        |            | GOSUNGROW_CONFIG        | GoSungrow: config file.        | /data/.GoSungrow/config.json     |
| --debug         |            | GOSUNGROW_DEBUG         | GoSungrow: Debug mode.         | false *                          |
| --quiet         |            | GOSUNGROW_QUIET         | GoSungrow: Silence all         | false *                          |
|                 |            |                         | messages.                      |                                  |
| --timeout       |            | GOSUNGROW_TIMEOUT       | Web timeout.                   | 1m0s                             |
| --user          | -u         | GOSUNGROW_USER          | SunGrow: api username.         | Joakim.Sundh4@gmail.com          |
| --password      | -p         | GOSUNGROW_PASSWORD      | SunGrow: api password.         | Gilerark50                       |
| --appkey        |            | GOSUNGROW_APPKEY        | SunGrow: api application key.  | 93D72E60331ABDCDC7B39ADC2D1F32B3 |
|                 |            |                         |                                | *                                |
| --host          |            | GOSUNGROW_HOST          | SunGrow: Provider API URL.     |                                  |
|                 |            |                         |                                | https://gateway.isolarcloud.eu   |
| --token-expiry  |            | GOSUNGROW_TOKEN_EXPIRY  | SunGrow: last login.           |  *                               |
| --save          | -s         | GOSUNGROW_SAVE          | Save output as a file.         | false *                          |
| --mqtt-user     |            | GOSUNGROW_MQTT_USER     | HASSIO: mqtt username.         |  *                               |
| --mqtt-password |            | GOSUNGROW_MQTT_PASSWORD | HASSIO: mqtt password.         |  *                               |
| --mqtt-host     |            | GOSUNGROW_MQTT_HOST     | HASSIO: mqtt host.             | core-mosquitto                   |
| --mqtt-port     |            | GOSUNGROW_MQTT_PORT     | HASSIO: mqtt port.             |                             1883 |
+-----------------+------------+-------------------------+--------------------------------+----------------------------------+
[06:18:13] INFO: Login to iSolarCloud using gateway  https://gateway.isolarcloud.eu ...
Error: Sungrow API EndPoint not yet implemented
Sungrow API EndPoint not yet implemented
Usage:
  GoSungrow api login [flags]
Examples:
	GoSungrow api login  
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: Sungrow API EndPoint not yet implemented
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Thank you for any assistance!

Best regards, Joakim

Surkel, 2022-12-21:
> OK. That’s a whole lot closer.
> 
> 2022/12/15 11:50:58 Found SunGrow 2 devices
> 
> it found your 2 devices, but now we have another problem.
> 
> ERROR: identifier rejected

Hi,
any new information about that issue?

MickMake, 2022-12-21:
I've just released v3.0.2. Could you try that version and see if it's fixed your issue?

Surkel, 2022-12-21:
> I've just released v3.0.2. Could you try that version and see if it's fixed your issue?

new error:
[06:06:35] INFO: Setting up GoSungrow config ...
/usr/local/bin/run.sh: line 52: GOSUNGROW_MQTT_HOST: unbound variable
s6-rc: info: service legacy-services: stopping

mosquito broker is running and throw this log event:
2022-12-21 06:44:27: New connection from 172.30.33.3:43242 on port 1883.
error: received null username or password for unpwd check

MickMake, 2022-12-23:
Just released v3.0.3!

Surkel, 2022-12-23:
> Just released v3.0.3!

Hi,

still the Problem with identifier rejected.

[10:12:10] INFO: Syncing data from gateway https://gateway.isolarcloud.eu ...
2022/12/23 10:12:11 INFO: Connecting to MQTT HASSIO Service...
2022/12/23 10:12:11 INFO: Connecting to SunGrow...
2022/12/23 10:12:11 INFO: Found SunGrow 2 devices
Error: identifier rejected
Usage:
  GoSungrow mqtt run [flags]
Aliases:
  run, 
Examples:
	GoSungrow mqtt run  
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: identifier rejected
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Kungkarl86, 2023-01-05:
Hello, would love to implement your project but I have the same issue as Surkel,

[16:55:12] INFO: Syncing data from gateway https://gateway.isolarcloud.eu ...
2023/01/05 16:55:12 INFO: Connecting to MQTT HASSIO Service...
2023/01/05 16:55:12 INFO: Connecting to SunGrow...
2023/01/05 16:55:13 INFO: Found SunGrow 2 devices
Error: identifier rejected
Usage:
  GoSungrow mqtt run [flags]
Aliases:
  run, 
Examples:
	GoSungrow mqtt run  
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: identifier rejected
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

dbr18210, 2023-01-21:
Hello, I get to a similar point (it finds two devices) but then the following error:

2023/01/21 01:22:39 INFO: Connecting to MQTT HASSIO Service...
2023/01/21 01:22:39 INFO: Connecting to SunGrow...
2023/01/21 01:22:39 INFO: Found SunGrow 2 devices
2023/01/21 01:22:39 INFO: Caching Sungrow metadata...
2023/01/21 01:22:40 INFO: Cached 236 Sungrow data points...
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0xc8 pc=0x441a60]
goroutine 1 [running]:
GoSungrow/iSolarCloud/AppService/queryDeviceList.(*EndPoint).SetPvPoints(_, {{_, _, _}}, {0x40000d7380, {0x400066e4a7, 0x7}, {0xc0eaed7850d590c9, 0x801d8077, 0x411eca0}, ...})
	/Users/mick/Documents/GoLang/Solar/GoSungrow/iSolarCloud/AppService/queryDeviceList/data.go:419 +0x630
GoSungrow/iSolarCloud/AppService/queryDeviceList.(*EndPoint).GetEnergyStorageSystem(_, {0x40000d7380, {0x400066e4a7, 0x7}, {0xc0eaed7850d590c9, 0x801d8077, 0x411eca0}, {{0x400047aa00, 0x2, 0x2}}, ...})
	/Users/mick/Documents/GoLang/Solar/GoSungrow/iSolarCloud/AppService/queryDeviceList/data.go:299 +0x2e0
GoSungrow/iSolarCloud/AppService/queryDeviceList.(*EndPoint).GetData(_)
	/Users/mick/Documents/GoLang/Solar/GoSungrow/iSolarCloud/AppService/queryDeviceList/data.go:259 +0x94
GoSungrow/iSolarCloud/AppService/queryDeviceList.EndPoint.GetEndPointData(...)
	/Users/mick/Documents/GoLang/Solar/GoSungrow/iSolarCloud/AppService/queryDeviceList/struct.go:367
GoSungrow/iSolarCloud.(*SunGrowData).CallEndpoint(_, {_, _}, {{0x4000b54a20, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, ...}, ...})
	/Users/mick/Documents/GoLang/Solar/GoSungrow/iSolarCloud/data.go:161 +0x344
GoSungrow/iSolarCloud.(*SunGrowData).getDataSinglePsIdRequired(0x4000a65460, {0x27661a8, 0x400049ea80})
	/Users/mick/Documents/GoLang/Solar/GoSungrow/iSolarCloud/data.go:281 +0x254
GoSungrow/iSolarCloud.(*SunGrowData).GetDataSingle(0x4000a65460, {0x4000b56040, 0xf})
	/Users/mick/Documents/GoLang/Solar/GoSungrow/iSolarCloud/data.go:236 +0x124
GoSungrow/iSolarCloud.(*SunGrowData).GetData(0x4000a65460)
	/Users/mick/Documents/GoLang/Solar/GoSungrow/iSolarCloud/data.go:207 +0x134
GoSungrow/cmd.(*CmdMqtt).Cron(0x4000536aa0)
	/Users/mick/Documents/GoLang/Solar/GoSungrow/cmd/cmd_mqtt.go:370 +0x200
GoSungrow/cmd.(*CmdMqtt).CmdMqttRun(0x4000536aa0, 0x0?, {0x0?, 0x0?, 0x0?})
	/Users/mick/Documents/GoLang/Solar/GoSungrow/cmd/cmd_mqtt.go:268 +0x70
github.com/spf13/cobra.(*Command).execute(0x40005d2600, {0x400003a1a0, 0x0, 0x0})
	/Users/mick/go/pkg/mod/github.com/spf13/cobra@v1.6.1/command.go:916 +0x5e0
github.com/spf13/cobra.(*Command).ExecuteC(0x40000c6000)
	/Users/mick/go/pkg/mod/github.com/spf13/cobra@v1.6.1/command.go:1044 +0x368
github.com/spf13/cobra.(*Command).Execute(...)
	/Users/mick/go/pkg/mod/github.com/spf13/cobra@v1.6.1/command.go:968
github.com/MickMake/GoUnify/Unify.(*Commands).Execute(...)
	/Users/mick/go/pkg/mod/github.com/!mick!make/!go!unify/!unify@v0.0.0-20221125023651-ff4a37b1928a/struct.go:258
github.com/MickMake/GoUnify/Unify.(*Unify).Execute(0x40000010e0)
	/Users/mick/go/pkg/mod/github.com/!mick!make/!go!unify/!unify@v0.0.0-20221125023651-ff4a37b1928a/struct.go:201 +0x304
GoSungrow/cmd.Execute(...)
	/Users/mick/Documents/GoLang/Solar/GoSungrow/cmd/commands.go:90
main.main()
	/Users/mick/Documents/GoLang/Solar/GoSungrow/main.go:11 +0x6c
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

andreas870, 2023-02-13:
Hello,
I have the following problem, does anyone happen to have a solution?

[16:34:12] INFO: Login to iSolarCloud using gateway https://Gateway.isolarcloud.eu ...
Email:	andreas@nuding.name
Create Date:	Tue Jul 26 03:41:07 CST 2022
Login Last Date:	2023-02-14 00:34:13
Login Last IP:	
Login State:	1
User Account:	e2z2mk933u
User Id:	78178
User Name:	Nuding
Is Online:	false
Token:	78178_c4dec8aafb6747109110aefb477d0405
Token File:	/data/.GoSungrow/AppService_login.json
Email:	andreas@nuding.name
Create Date:	Tue Jul 26 03:41:07 CST 2022
Login Last Date:	2023-02-14 00:34:13
Login Last IP:	
Login State:	1
User Account:	e2z2mk933u
User Id:	78178
User Name:	Nuding
Is Online:	false
Token:	78178_75e8ab4a0c5b4320b1165989d200def9
Token File:	/data/.GoSungrow/AppService_login.json
[16:34:13] INFO: Syncing data from gateway https://Gateway.isolarcloud.eu ...
Email:	andreas@nuding.name
Create Date:	Tue Jul 26 03:41:07 CST 2022
Login Last Date:	2023-02-14 00:34:13
Login Last IP:	
Login State:	1
User Account:	e2z2mk933u
User Id:	78178
User Name:	Nuding
Is Online:	false
Token:	78178_75e8ab4a0c5b4320b1165989d200def9
Token File:	/data/.GoSungrow/AppService_login.json
2023/02/13 16:34:13 INFO: Connecting to MQTT HASSIO Service...
2023/02/13 16:34:13 INFO: Connecting to SunGrow...
2023/02/13 16:34:14 INFO: Found SunGrow 4 devices
Error: network Error : dial tcp: lookup core-mosquitto on 127.0.0.11:53: no such host
Usage:
  GoSungrow mqtt run [flags]
Aliases:
  run, 
Examples:
	GoSungrow mqtt run  
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: network Error : dial tcp: lookup core-mosquitto on 127.0.0.11:53: no such host
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped