MickMake/GoSungrow #21: HASSIO "Energy"

DrLEEi, 2022-12-18:
Application utility entities cannot be added to HASSIO's built-in "Energy" section.

Either there is no match between the entities that can be added, or it indicates an error:
"The date of the last reset is missing
The following entities have a "measurement" state class but are missing a "last_reset" date.

I've tried changing the entity's "state_class" to "total_increasing", but the app seems to reset it to its original state every time the data is updated, so that's not a solution.

It would be useful to add real-time power (current power) and daily yield return entities to the HASSIO "Energy" section for long-term statistics.

MickMake, 2022-12-18:
Ah yes. This is one thing I didn't want to add in this release as it was already a major one.

It's not a complicated change, but it's silly season. I'll try and get it in this week.

MickMake, 2022-12-21:
I'll add this into v3.0.3 release.

DrLEEi, 2022-12-21:
ok!looking forward to your update and thanks for your work!

MickMake, 2022-12-23:
v3.0.3 just released. Read the DOCS.md to see how to add it.

DrLEEi, 2022-12-23:
HA:uptodate
GoSungrow:3.0.3.
Sungrow inverter:SG5.0RT

I only have solar panels.

The entry specified in the installation document
Solar Panels
sensor.gosungrow_virtual_XXXXXXXXXXXX_pv_energy

I do not have such an entity, it is not available.

I thoroughly checked all my available entities (both hidden and disabled) and only found my real-time performance data in the following places:

state_class: measurement
unit_of_measurement: kW
device_class: power
icon: mdi:solar-power-variant-outline
friendly_name: GoSungrow virtual.devices.XXXXXXXXX.curr_power

state_class: measurement
unit_of_measurement: kW
device_class: power
icon: mdi:solar-power-variant-outline
friendly_name: GoSungrow getPsList.devices.XXXXXXXX.curr_power

state_class: measurement
unit_of_measurement: kW
device_class: power
icon: mdi:solar-power-variant-outline
friendly_name: GoSungrow getPsDetail.XXXXXXX.curr_power

state_class: measurement
unit_of_measurement: kW
device_class: power
icon: mdi:solar-power-variant-outline
friendly_name: GoSungrow XXXXXXXX_1_1_1 - p24 - Overview - Total Active Power

However, these cannot be matched to HA "Erergy".

I wrote it because I just watched it, but don't work anymore, celebrate now!

Thanks for your effort!

MickMake, 2022-12-27:
I've just published v3.0.4-beta, (not released yet). This fixes some problems with the energy dashboard, but it's now working for me.

What mode/unit do you have installed? Looks like it's a device_type 1, which is a single phase unit without battery, (I think).

MickMake, 2022-12-27:
`Sungrow inverter:SG5.0RT`

Ah! Didn't see this at the top of your message. I think the points I'm using are possibly not available for that unit.
My brother has a similar single phase unit without batteries; so I'll see where the points are defined.

nvx, 2022-12-29:
I'm in a similar boat, I have a whole tonne of energy entities (maybe 50 or so) but none that match the docs for setting up the energy stuff.

Solar only without battery, SG5K inverter with Smart Meter.

I think the data is all there, just not being recognised what's what.

MickMake, 2023-01-03:
The good thing is that a point_id remains the same within a specific model, but it's just finding the right one.

If you download GoSungrow and use the following command, it'll show you all the defined points for your model.

`GoSungrow show ps points`

I think I have to provide the capability for people to define their own "virtual" entities that they can share with others to make it easier. Kinda thinking of a simple JSON file format.

nvx, 2023-01-05:
I imagine having some defaults based on the model# of inverter would likely work for most people.

chris3081, 2023-04-24:
I had issues with my Sungrow SG8K-D as well, I could see the stats in developer tools but they would fail under the energy dash 

I put the following in my customize.yaml which got me sorted (I've dropped the comments in for anyone chasing which ones to use)

# Smart Meter - Purchased from Grid
sensor.gosungrow_virtual_xxxxxxx_7_1_1_p8062:
  state_class: "total_increasing"
# Invertor Power Input Daily Yield
sensor.gosungrow_virtual_xxxxxxx_1_1_1_p1:
  state_class: "total_increasing"
# Smart Meter - Self-consumption of PV
sensor.gosungrow_virtual_xxxxxxx_7_1_1_p8084:
  state_class: "total_increasing"
# Smart Meter - Sold to Grid
sensor.gosungrow_virtual_xxxxxxx_7_1_1_p8063:
  state_class: "total_increasing"