MickMake/GoSungrow #22: Power units don't change when dropping below 1kW

jimmybish, 2022-12-19:
[image]
[image]

When generation or usage drops below 1kW, instead of the unit of measurement switching to W, it will stay in kW, shooting the usage up to the point where my legit usage is barely distinguishable in the graph. My house definitely didn't peak at over 950kW last night! 😆 

I haven't been at my PC at a time when this problem is actively occurring, so I haven't yet determined whether this is an issue with GoSungrow or the iSolarCloud API, but I should hopefully be available and able to catch it in the act later tonight and update this issue when I know more.

MickMake, 2022-12-19:
Yeah, I noticed that as well. The problem is that Sungrow have their list of points data, but it can change based on whether it's sub-1000 or not.

IE The official unit is `kW`, yet sometimes the unit comes in as `W`.

I'm working on this fix ATM.

MickMake, 2022-12-19:
I've discovered it's within the getPsDetail endpoint - which returns a duplicate entry from queryDeviceList, but lacks decent units.
p13149 & p13119

MickMake, 2022-12-19:
OK. And I now have a fix for it. Not the best, but works well enough.

I have a few other fixes to make, so will roll this up along with others as v3.0.2

jimmybish, 2022-12-21:
Awesome, thanks! Sorry I hadn't been active until now, Christmas preps and all... So extra kudos to you for progressing everything this week - it certainly wasn't expected! 

I'm running this as a service outside of HA, so will upgrade the binary later on to confirm, but if you're confident 3.0.2 sorts it, feel free to close 🙇

MickMake, 2022-12-21:
And released!

jimmybish, 2022-12-21:
Nice one! The binary's in and running as a service now. My solar generation should drop considerably over the next hour or 2, so will be able to confirm shortly.

MickMake, 2022-12-21:
Could you try the Lovelace command and see if it comes up with the correct layout?

I've only tried on my ps_id.

jimmybish, 2022-12-21:
```
$ GoSungrow ha lovelace
Can't find DeviceType 14 attached to ps_id 1181861.
Can't find DeviceType 14 attached to ps_id 1181868.

$ GoSungrow show ps list
┏━━━━━━━━━━━━━━━━━━┳━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━┓
┃ Ps Key           ┃ Ps Id   ┃ Device Type ┃ Device Code ┃ Channel Id ┃ Serial #    ┃ Factory Name ┃ Device Model   ┃
┣━━━━━━━━━━━━━━━━━━╇━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━┫
┃ 1181868_1_1_2    │ 1181868 │ 1           │ 1           │ 2          │ [REDACTED] │ SUNGROW      │ SG5K-D         ┃
┃ 1181868_1_1_1    │ 1181868 │ 1           │ 1           │ 1          │ [REDACTED] │ SUNGROW      │ SG5K-D         ┃
┃ 1181868_22_247_2 │ 1181868 │ 22          │ 247         │ 2          │ [REDACTED] │ SUNGROW      │ WiFi V31       ┃
┃ 1181868_22_247_1 │ 1181868 │ 22          │ 247         │ 1          │ [REDACTED] │ SUNGROW      │ WiFi V31       ┃
┃ 1181868_7_1_1    │ 1181868 │ 7           │ 1           │ 1          │ [REDACTED] │ SUNGROW      │ SG Smart Meter ┃
┗━━━━━━━━━━━━━━━━━━┷━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━━━┷━━━━━━━━━━━━━━━━┛
```

So it looks like it's made it past [here](https://github.com/MickMake/GoSungrow/blob/706306a963ae74180a73cbbf186250d9fc228cb4/cmd/cmd_ha.go#L124), but didn't return any YAML. Will the debug flag enable more output to see where it's falling over?

jimmybish, 2022-12-21:
[image]

Confirmed measurements are looking happier on my side now! 🥳

MickMake, 2022-12-21:
> ```
> 
> $ GoSungrow ha lovelace
> 
> Can't find DeviceType 14 attached to ps_id 1181861.
> 
> Can't find DeviceType 14 attached to ps_id 1181868.
> 
> 
> 
> $ GoSungrow show ps list
> 
> ┏━━━━━━━━━━━━━━━━━━┳━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━┓
> 
> ┃ Ps Key           ┃ Ps Id   ┃ Device Type ┃ Device Code ┃ Channel Id ┃ Serial #    ┃ Factory Name ┃ Device Model   ┃
> 
> ┣━━━━━━━━━━━━━━━━━━╇━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━┫
> 
> ┃ 1181868_1_1_2    │ 1181868 │ 1           │ 1           │ 2          │ [REDACTED] │ SUNGROW      │ SG5K-D         ┃
> 
> ┃ 1181868_1_1_1    │ 1181868 │ 1           │ 1           │ 1          │ [REDACTED] │ SUNGROW      │ SG5K-D         ┃
> 
> ┃ 1181868_22_247_2 │ 1181868 │ 22          │ 247         │ 2          │ [REDACTED] │ SUNGROW      │ WiFi V31       ┃
> 
> ┃ 1181868_22_247_1 │ 1181868 │ 22          │ 247         │ 1          │ [REDACTED] │ SUNGROW      │ WiFi V31       ┃
> 
> ┃ 1181868_7_1_1    │ 1181868 │ 7           │ 1           │ 1          │ [REDACTED] │ SUNGROW      │ SG Smart Meter ┃
> 
> ┗━━━━━━━━━━━━━━━━━━┷━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━━━┷━━━━━━━━━━━━━━━━┛
> 
> ```
> 
> 
> 
> So it looks like it's made it past [here](https://github.com/MickMake/GoSungrow/blob/706306a963ae74180a73cbbf186250d9fc228cb4/cmd/cmd_ha.go#L124), but didn't return any YAML. Nothing in the logs besides the HASSIO sync entries. Will the debug flag enable more output to see where it's falling over?
> 
> 

Ah, so you don't have a battery. I'll need to adjust to compensate for that.

Currently, if it can't find device_type 14, it'll abort.

MickMake, 2022-12-21:
> [image]
> 
> Confirmed measurements are looking happier on my side now! 🥳 

Excellent!

closed 2022-12-21 by MickMake