MickMake/GoSungrow #23: "ha lovelace" without battery

MickMake, 2022-12-21:
Need to support production of Lovelace dashboard even if there's no battery.

This will require changing the layout of the images as well, (because it looks different).

```
$ GoSungrow ha lovelace
Can't find DeviceType 14 attached to ps_id 1181861.
Can't find DeviceType 14 attached to ps_id 1181868.

$ GoSungrow show ps list
┏━━━━━━━━━━━━━━━━━━┳━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━┓
┃ Ps Key           ┃ Ps Id   ┃ Device Type ┃ Device Code ┃ Channel Id ┃ Serial #    ┃ Factory Name ┃ Device Model   ┃
┣━━━━━━━━━━━━━━━━━━╇━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━┫
┃ 1181868_1_1_2    │ 1181868 │ 1           │ 1           │ 2          │ [REDACTED] │ SUNGROW      │ SG5K-D         ┃
┃ 1181868_1_1_1    │ 1181868 │ 1           │ 1           │ 1          │ [REDACTED] │ SUNGROW      │ SG5K-D         ┃
┃ 1181868_22_247_2 │ 1181868 │ 22          │ 247         │ 2          │ [REDACTED] │ SUNGROW      │ WiFi V31       ┃
┃ 1181868_22_247_1 │ 1181868 │ 22          │ 247         │ 1          │ [REDACTED] │ SUNGROW      │ WiFi V31       ┃
┃ 1181868_7_1_1    │ 1181868 │ 7           │ 1           │ 1          │ [REDACTED] │ SUNGROW      │ SG Smart Meter ┃
┗━━━━━━━━━━━━━━━━━━┷━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━━━┷━━━━━━━━━━━━━━━━┛
```

So it looks like it's made it past [here](https://github.com/MickMake/GoSungrow/blob/706306a963ae74180a73cbbf186250d9fc228cb4/cmd/cmd_ha.go#L124), but didn't return any YAML. Nothing in the logs besides the HASSIO sync entries. Will the debug flag enable more output to see where it's falling over?

_Originally posted by @jimmybish in https://github.com/MickMake/GoSungrow/issues/22#issuecomment-1360917295_

MickMake, 2022-12-21:
[image]

scraja69, 2023-03-30:
My inverter setup does not have a battery.I cannot enter Ha lovelace command and it fails.. Is there anthing to do with user?. Can you please explain.