MickMake/GoSungrow #24: 3.0.2 not starting any more in HA

djp3k05, 2022-12-22:
here is the error:

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[15:34:41] INFO: Setting up GoSungrow config ...
/usr/local/bin/run.sh: line 52: GOSUNGROW_MQTT_HOST: unbound variable
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped;`

Nothing changed in the config options from 3.0.1 to 3.0.2.

Thank you.

djp3k05, 2022-12-22:
hmmm... Fixed it! I wanted to revert back to 301, but right before doing that I tried the rebuilt option... and surprise. after that it magically working. so something in the update process had failed.

Issue can be closed.

MickMake, 2022-12-23:
I going to suggest trying that.

closed 2022-12-23 by MickMake