MickMake/GoSungrow #28: API put example to change configuration

Chefkoch81, 2023-01-03:
Hi,

Thx for that great piece of software.

Not really an issue, but a documentation thing (I guess).

I am unsure how I can modify some settings. For example I would like to change the IncomeSetting on a daily basis. When I call get like this:
```
./GoSungrow api get getIncomeSettingInfos '{"ps_id":"1234567"}'
{
        "code_type": 405,
        "enviormentPowerChargeList": [],
        "param_income_unit": 405,
        "powerElectricalChargeMap": {
                "city_allowance_money": null,
                "code_type": 405,
                "county_allowance_money": null,
                "default_charge": 0.2686,
                "electric_charge_id": 157712,
                "end_time": "2099-12-31",
                "income_style": null,
                "interval_time_charge": null,
                "nation_allowance_money": null,
                "param_income_unit": 405,
                "province_allowance_money": null,
                "ps_id": 1234567,
                "start_time": "2023-01-01",
                "use_sharp_peek_valley_flat": null,
                "valid_flag": true
        },
        "powerIntevalTimesChargeMap": null,
        "powerSelfUseTimesChargeMap": {
                "default_charge": 0.3907,
                "end_time": "2099-12-31",
                "interval_time_charge": "13:00-14:00|0.3906,14:00-15:00|0.3908",
                "online_electricity_percent": 0,
                "ps_id": 1234567,
                "start_time": "2023-01-01",
                "use_electricity_discount": 100
        },
        "ps_id": 5113485
}
```

I only would like to change powerSelfUseTimesChargeMap."interval_time_charge". How would I need to do that? I guess with put, but without an example I was afraid of breaking something.

MickMake, 2023-01-03:
Yup, I haven't really tested modification endpoints, and not that one specifically. It's a case of trial and error, but I'll add this one to the next release. I think it's simple enough to add and probably one that people would want.

manny4566, 2023-02-13:
Hello,

it is possible change the discharge cut-off SOC via the api?
My goal is to change the soc so (100%) that my battery is not discharge for a special time. After that i want to change the soc back so that the battery is used by  my house again. 
It is possible in the app. But i search a solution to trigger that externally.
Thanks