MickMake/GoSungrow #29: Parsing Error calling "api get getPsDetailWithPsType"

siftbo, 2023-01-10:
Hey! 
I'm really really thankfull for this project. I want to embed my smart-meter data into my grafana boards, and want to use the information to trigger heating etc. 

I was doing a lot of try and error, to find the right API where the smart-meter data is provided. And I only found this one here. But the implementation is broken :-(.

I'm not a Go guy - sorry therefore I can not help to fix the problem. But maybe, you find time to fix. 
Or maybe you can help me out if there are other API methods where I can access the smart meter data like, active power, etc.

`/GoSungrow api get getPsDetailWithPsType '{"ps_id":"5076346"}'
PsId: required
JSON request:	{"ps_id":5076346}

Error: strconv.ParseInt: parsing "[\n\t\t\t{\n\t\t\t\t\"ameter_device_type\":7,\n\t\t\t\t\"ameter_ps_key\":\"5076346_7_1_1\",\n\t\t\t\t\"ameter_uuid\":1886463,\n\t\t\t\t\"communication_dev_sn\":\"B21C0204055\",\n\t\t\t\t\"dev_status\":1,\n\t\t\t\t\"device_code\":1,\n\t\t\t\t\"device_model_code\":\"SG20RT\",\n\t\t\t\t\"device_name\":\"SG20RT(COM1-001)_001_001\",\n\t\t\t\t\"device_state\":\"3\",\n\t\t\t\t\"device_type\":1,\n\t\t\t\t\"energy_flow\":[\n\t\t\t\t\t\"3\"\n\t\t\t\t],\n\t\t\t\t\"has_ammeter\":1,\n\t\t\t\t\"installer_dev_fault_status\":3,\n\t\t\t\t\"inverter_sn\":\"A21C0703986\",\n\t\t\t\t\"owner_dev_fault_status\":3,\n\t\t\t\t\"p14_map\":{\n\t\t\t\t\t\"unit\":\"W\",\n\t\t\t\t\t\"value\":\"0\"\n\t\t\t\t},\n\t\t\t\t\"p14_map_virgin\":{\n\t\t\t\t\t\"unit\":\"W\",\n\t\t\t\t\t\"value\":\"0\"\n\t\t\t\t},\n\t\t\t\t\"p24_power_map\":{\n\t\t\t\t\t\"unit\":\"W\",\n\t\t\t\t\t\"value\":\"0\"\n\t\t\t\t},\n\t\t\t\t\"p24_power_map_virgin\":{\n\t\t\t\t\t\"unit\":\"W\",\n\t\t\t\t\t\"value\":\"0\"\n\t\t\t\t},\n\t\t\t\t\"p8018\":\"455.0\",\n\t\t\t\t\"p8074_map\":{\n\t\t\t\t\t\"unit\":\"W\",\n\t\t\t\t\t\"value\":\"455\"\n\t\t\t\t},\n\t\t\t\t\"p8074_map_virgin\":{\n\t\t\t\t\t\"unit\":\"W\",\n\t\t\t\t\t\"value\":\"455\"\n\t\t\t\t},\n\t\t\t\t\"p8075_map\":{\n\t\t\t\t\t\"unit\":\"W\",\n\t\t\t\t\t\"value\":\"0\"\n\t\t\t\t},\n\t\t\t\t\"p8075_map_virgin\":{\n\t\t\t\t\t\"unit\":\"W\",\n\t\t\t\t\t\"value\":\"0\"\n\t\t\t\t},\n\t\t\t\t\"p8079_map\":{\n\t\t\t\t\t\"unit\":\"W\",\n\t\t\t\t\t\"value\":\"455\"\n\t\t\t\t},\n\t\t\t\t\"p8079_map_virgin\":{\n\t\t\t\t\t\"unit\":\"W\",\n\t\t\t\t\t\"value\":\"455\"\n\t\t\t\t},\n\t\t\t\t\"ps_key\":\"5076346_1_1_1\",\n\t\t\t\t\"uuid\":1088855\n\t\t\t}\n\t\t]": invalid syntax
`

MickMake, 2023-01-20:
Ah yes. This is a bug.
Problem is; I'm having to guess fields and their types which often flip/flop between string, float and integer.

I'll see if I can replicate.

siftbo, 2023-01-30:
many thanks