MickMake/GoSungrow #30: isolarcloud app is not working when using gosungrow

rey900, 2023-01-12:
Hello

everytime i use the gosungrow Addon all data is pulled and displayed correctly. But after ~6hours the isolarcloud says that my device is offline and so no actual data gets pulled. 

If i disable the addon after ~10hours the device is back online in the app. 

Maybe they are too many requests ? my refresh intervall is 2m

MickMake, 2023-01-20:
Yeah, I'd set it to 5m, which is the minimum change rate.

Interesting that it's coming up as disabled though. I don't see that with mine.

I'll leave this open and see if anyone else has the same issue.

harrison6987, 2023-01-24:
I'm not sure if this is related but rather than my iSolarCloud app not working, the GoSungrow add-on stops due to a timeout error after working fine for a day or so.

> 2023/01/19 01:53:09 ERROR: Post "https://augateway.isolarcloud.com/v1/powerStationService/getPsList": dial tcp: lookup augateway.isolarcloud.com on 127.0.0.11:53: read udp 127.0.0.1:59311->127.0.0.11:53: i/o timeout
Error: Post "https://augateway.isolarcloud.com/v1/powerStationService/getPsList": dial tcp: lookup augateway.isolarcloud.com on 127.0.0.11:53: read udp 127.0.0.1:59311->127.0.0.11:53: i/o timeout
Usage:
  GoSungrow mqtt run [flags]
Aliases:
  run, 
Examples:
	GoSungrow mqtt run  
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: Post "https://augateway.isolarcloud.com/v1/powerStationService/getPsList": dial tcp: lookup augateway.isolarcloud.com on 127.0.0.11:53: read udp 127.0.0.1:59311->127.0.0.11:53: i/o timeout
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped