MickMake/GoSungrow #32: Can I use this in a HA installation without supervisor?

stalakerob, 2023-02-08:
Hi, I'd like to use this component in my HA installation but I'm running HA as a docker installation on a raspberry Pi 4. There is no supervisor, so I cannot use add-ons. Is there a docker file for Linux (raspberry PI OS) on arm ? If not, can I create one and if so, how?
Thx

scarafaggio, 2023-02-21:
Hi, did you make any progress on this?  I have a similar situation, HA in a docker container on another box.  
I've been fiddling trying to get a Dockerfile working, but my main challenge is that I've never touched GO before, so learning that at the same time.  
There are 2x Dockerfiles in the repo.  One uses local files and the other remote.   In the local case the docker build seems to work until the copy step can't find a dist/  directory.   In the remote case wget  falls over with a 404/not found error.   In either case I can't find a GoSungrow executable.  
Alpine doesn't have GO installed, and there is no mention that I can see elsewhere of a "build" step to generate the binary.   So where does that GO build/compile step happen?  Presumably I'm misunderstanding how this is supposed to work.  
I haven't got as far as the ssh certs yet ... but that is looking tricky too!
Anyway, if you make any progress please share ... good luck.

stalakerob, 2023-02-21:
Thanks for posting your interest as well. Unfortunately no progress so far. It should not be hard to do that since a HA add-on is also a container image. Most add-ons are also available as a docker image but this one is not. I do not have enough technical skills to convert this add-on. It should be in the interest of the developer to offer that. It could then be used not only for HA installations.

Paraphraser, 2023-02-24:
@scarafaggio

> Alpine doesn't have GO installed, and there is no mention that I can see elsewhere of a "build" step to generate the binary. So where does that GO build/compile step happen? Presumably I'm misunderstanding how this is supposed to work.

I can help you with this bit. See [this gist](https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37).

Once you have compiled GoSungrow, you can move the compiled binary into the directory alongside the Dockerfile and use the Dockerfile COPY command to copy it into the image as you build it.

There are some rules about what gets copied into the image as part of the build context. I don't have a full understanding but I interpret it to mean "minimal is best". In other words, it is better to move the compiled binary next to the Dockerfile, rather than clone the GoSungrow repo into the same directory as the Dockerfile, compile it there, and then have COPY reference `./GoSungrow/GoSungrow` - if that makes sense.

> I haven't got as far as the ssh certs yet ... but that is looking tricky too!

I'm very very new to Sungrow and GoSungrow so I haven't gotten my brain around very much of anything - including this bit.

What are the "ssh certs" for?

If it's just so the process inside the container can run commands outside the container using `ssh` then you should be able to follow the instructions [here](https://sensorsiot.github.io/IOTstack/Containers/Node-RED/#sshOutside) to generate any necessary keys in situ.

Assuming the process inside the container runs as root, its home directory will be `/root`. If that internal path is mapped to a persistent store outside the container, both `/root/.ssh` and `/root/.GoSungrow` should persist across invocations.

> Alpine doesn't have …

On the topic of Alpine, I know it's lightweight and fashionable as an image base but it seems to come with a lot of gotchas. I spend a lot of time working on [IOTstack](https://github.com/SensorsIot/IOTstack) and, over the past 3-odd years, Alpine has wound up being implicated in more than its fair share of issues.

At the moment I'm following a discussion on Node-RED where [DNS failover](https://github.com/node-red/node-red-docker/issues/348) doesn't work. That has been tracked to Alpine.

There's also a [proposal](https://github.com/node-red/node-red-docker/issues/340) to build a Debian version which is based on that having solved a [fairly weird problem](https://github.com/node-red/node-red-docker/issues/305) that showed up on Alpine. The proposal is getting pushback on the basis of workload. That's not an unreasonable position to take when you're the one who will have to do the work but it isn't actually an argument on the merits of the proposal.

I base my own containers on Ubuntu. Once I get them working they become "just works, rock solid, set and forget".

To put this another way, when you're building an image for your own use, the basic principle is "your container, your rules". I think "your rules" should also include "your decision" about an appropriate base OS.

Anyway, hope this helps you make progress.

scarafaggio, 2023-02-25:
Thanks, your comments are very helpful.   I have a bunch of containers running in a home environment.   Most of these I can pull directly from dockerhub and restrict my local customisation to a few config files and docker-compose.  I've done this deliberately for ease of maintenance, I can easily recreate any container or even the whole docker host.  I have a few home-grown containers where the dockerfile does the work but even then I try to do everything inside the container so I'm not dependent on any other system.   So in this case with GoSungrow I'd like to be able to do the compile as part of building the image.   There seems to be an official image called golang which is itself based on alpine.    So I'll see if I can base my container on that.

MickMake, 2023-02-25:
> Hi, I'd like to use this component in my HA installation but I'm running HA as a docker installation on a raspberry Pi 4. There is no supervisor, so I cannot use add-ons. Is there a docker file for Linux (raspberry PI OS) on arm ? If not, can I create one and if so, how?
> 
> Thx

Yes, it's fairly trivial to do this. I did have a Docker image file that I used to use as part of this repo. I'll have a look at updating it for general use.

MickMake, 2023-02-25:
> Hi, did you make any progress on this?  I have a similar situation, HA in a docker container on another box.
>
> I've been fiddling trying to get a Dockerfile working, but my main challenge is that [...]

Yeah, these are the docker files that I mentioned previously. I used them for testing, but I should update them properly.

Instead of rebuilding from scratch, I would use the prebuilt binaries that I create with every release.

That way, it's a much quicker docker build for you.

stalakerob, 2023-03-27:
Anyone got a docker build running? If so, how?

Altau38650, 2023-05-07:
Hello,

I'm new to HA and also to Docker and I don't speak English very well. However, I think I understood that, MickMake, you were working on a version of your tool to be installed on HA Docker?

So I am interested in this solution. When do you think you can provide this tool?

Many thanks for your work and for your future reply.

--
Alain
France
Inverter : SG3-0RS

stalakerob, 2023-05-09:
@Altau38650 Alternatively you can use [https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant](url)

Altau38650, 2023-05-09:
Thank you very much for the link.
I have managed to get this system to work in order to track my solar production.
Good luck with that.
--
Alain

EvoSems, 2024-02-01:
ok. compiled myself on a rpi! big yay! now what to do to make a docker file?

EvoSems, 2024-02-02:
awesome thanks for the response! I have got some progress but I am getting an error logging in? I have checked my credentials over and over. any ideas?

$ sudo docker run -d --name gosungrow --restart unless-stopped -v /home/evo/gosungrow:/data/ -v /home/evo/gosungrow/certs:/etc/ssl/certs:ro -e TZ=Australia/Brisbane --network=host triamazikamno/armv7-addon-gosungrow:3.0.7
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
83c3f10683d150e974a5597ceba4b5d9c4da9d13701213aacce24e8a06e7ecdc

manually added the following structure/files
home/evo/gosungrow/.GoSungrow/config.json (with my details in full included)
home/evo/gosungrow/options.json (details included)

error logs.
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[02:18:39] INFO: Setting up GoSungrow config ...
[02:18:40] WARNING: MQTT user is empty. I'm guessing this is what you wanted?
[02:18:40] WARNING: MQTT password is empty. I'm guessing this is what you wanted?
[02:18:40] INFO: Writing GoSungrow config ...
New config:
Using config file '/data/.GoSungrow/config.json'
+-------------------+------------+---------------------------+--------------------------------+-----------------------------------+
|       FLAG        | SHORT FLAG |        ENVIRONMENT        |          DESCRIPTION           |        VALUE (* = DEFAULT)        |
+-------------------+------------+---------------------------+--------------------------------+-----------------------------------+
| --config          |            | GOSUNGROW_CONFIG          | GoSungrow: config file.        | /data/.GoSungrow/config.json      |
| --debug           |            | GOSUNGROW_DEBUG           | GoSungrow: Debug mode.         | false *                           |
| --quiet           |            | GOSUNGROW_QUIET           | GoSungrow: Silence all         | false *                           |
|                   |            |                           | messages.                      |                                   |
| --timeout         |            | GOSUNGROW_TIMEOUT         | Web timeout.                   | 1m0s                              |
| --user            | -u         | GOSUNGROW_USER            | SunGrow: api username.         | xxxxxxxxxxxxxxx               |
| --password        | -p         | GOSUNGROW_PASSWORD        | SunGrow: api password.         | xxxxxxxxxxxxxxxx                      |
| --appkey          |            | GOSUNGROW_APPKEY          | SunGrow: api application key.  | B0455FBE7AA0328DB57B59AA729F05D8  |
| --host            |            | GOSUNGROW_HOST            | SunGrow: Provider API URL.     | https://augateway.isolarcloud.com |
|                   |            |                           |                                | *                                 |
| --token-expiry    |            | GOSUNGROW_TOKEN_EXPIRY    | SunGrow: last login.           |  *                                |
| --save            | -s         | GOSUNGROW_SAVE            | Save output as a file.         | false *                           |
| --dir             |            | GOSUNGROW_DIR             | Save output base directory.    |  *                                |
| --mqtt-user       |            | GOSUNGROW_MQTT_USER       | HASSIO: mqtt username.         | xxxxxx                              |
| --mqtt-password   |            | GOSUNGROW_MQTT_PASSWORD   | HASSIO: mqtt password.         | xxxxxxxxxxxx                         |
| --mqtt-host       |            | GOSUNGROW_MQTT_HOST       | HASSIO: mqtt host.             | 192.xxx.xxx.xxx                     |
| --mqtt-port       |            | GOSUNGROW_MQTT_PORT       | HASSIO: mqtt port.             |                              1883 |
| --modbus-user     |            | GOSUNGROW_MODBUS_USER     | Modbus username.               |  *                                |
| --modbus-password |            | GOSUNGROW_MODBUS_PASSWORD | Modbus password.               |  *                                |
| --modbus-host     |            | GOSUNGROW_MODBUS_HOST     | Modbus host.                   | 192.xxx.xxx.xxx                    |
| --modbus-port     |            | GOSUNGROW_MODBUS_PORT     | Modbus port.                   | 502 *                             |
+-------------------+------------+---------------------------+--------------------------------+-----------------------------------+
[02:18:43] INFO: Login to iSolarCloud using gateway https://augateway.isolarcloud.com ...
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x8 pc=0x4b69a8]
goroutine 1 [running]:`

EvoSems, 2024-02-02:
also any news on modbus support? I see it in the config fields. : )

and another thing is. I can't add my solar to HA energy panel. the sensors I need are not options in the selection panel?

ex.
sensor.gosungrow_virtual_xxx_1_1_2_p24 (My Solar stats)

this is not selectable or even visible in the HA energy settings page for solar. same with grid.
sensor.gosungrow_virtual_xxx_7_1_2_p8018
sensor.gosungrow_virtual_xxx_p83106

Paraphraser, 2024-02-02:
I can't say for certain but perhaps try again save that, this time, leave the `/etc/ssl/certs:/etc/ssl/certs:ro` volume mapping **exactly as written**. What it's doing is mapping your host's certs file (which is maintained by the OS vendor - eg Raspberry Pi Foundation) through to the container (with the `:ro` read-only suffix protecting the host's folder from interference from the container - always guard against rogue processes). I suppose you *could* always copy the host's `/etc/ssl/certs` to a folder of your choice and map that Into the container but then you would also be responsible for keeping that copy in parallel with the original. It's much simpler to let routine `apt update ; apt upgrade` do the work for you.

Paraphraser, 2024-02-02:
> also any news on modbus support? I see it in the config fields. : )

I see it too but I've no idea. I can't speak for MickMake. I just hang out here (plus a few other GitHub repos) and answer the odd question if it's within my skill set - my main motivation is that I always learn something when I attempt to answer a question.
 
> and another thing is. I can't add my solar to HA energy panel. the sensors I need are not options in the selection panel?
>
> ex. sensor.gosungrow_virtual_xxx_1_1_2_p24 (My Solar stats)
>
> this is not [...]

Look, I had the same problem too. Just over a year ago I replaced a SolaX with the Sungrow and all I really wanted to do with GoSungrow was continue the same data set. It took me a while to figure out which metrics in the vast quantity of what's available were the ones I needed. A side effect of that process brought the surprising realisation that much of what was useful to me didn't actually show up in any of what I'll call GoSungrow's standard outputs. Kinda weird.

Anyway, I just have a cron job which fires at 3am to invoke GoSungrow to download "yesterday's" data and munge it into a CSV which winds up in an SQLite database. Ultimately, all I really care about is, for a given billing period, what did the panels produce (both DC and AC), what was self-use, what was the grid usage, and the feed-in (for a pissy 7 cents/kWh). That's enough to calculate both "what will the next bill be?" and "what would it have been without solar?". Also, I'm never going to be more than a day from discovering if the system has gone on the blink (hasn't happened yet but...).

If I need "live", I just use the iSolarCloud app.

In short, I'm a bit of a low-rent user; definitely lacking in sophistication. 🤓

rohanrichards, 2024-04-23:
> @EvoSems You don't actually have to create your own Dockerfile. You can just re-use the images that were placed on [DockerHub by triamazikamno](https://hub.docker.com/u/triamazikamno). They are just "Docker images" so it doesn't really matter all that much whether it's Home Assistant or docker-compose or a `docker pull` command doing the "pulling".
> ...

Just wanted to say a huge thank you, I followed this guide and it worked perfectly first time. It sort of half works in the energy tab, the graphs look fine but the card data seems off. My recommendation is to just peak at all the sensor info and try to find the numbers that make sense. Though I never know if the energy tab wants totals or real-time data. I have just ended up making my own energy tab:
[image]

Really happy with this result, considering I went down the modbus rabbit hole, and learned the hard way my wiNet module did not like having modbus over wifi connections and refused to reconnect 😬. I'm so glad to have found goSungrow as an alternative for people who can't get LAN access to their inverter.