MickMake/GoSungrow #36: Templates don't work for newly-installed SG5.0RS

Paraphraser, 2023-02-20:
Last Thursday (2023-02-16) I had an SG5.0RS installed, replacing an older inverter from a different vendor.

This is my first Sungrow. I've been trying to follow your README but I'm stuck.

I'm fine with these steps:

```
$ GoSungrow config write --user=USERNAME --password=PASSWORD
$ GoSungrow api login
$ GoSungrow show ps list
$ GoSungrow show ps tree
$ GoSungrow show ps points 
$ GoSungrow show ps data 1203625 22 20230217000000 20230217235959 60
$ GoSungrow show ps graph 1203625 22 20230217000000 20230217235959 60
```

But when I try:

```
$ GoSungrow show template list
No data table results for 'getTemplateList.ResultData.PageList'

#  - has no data.
```

On a hunch, I tried:

```
$ GoSungrow api get getTemplateList
{
	"pageList": [],
	"rowCount": 0
}
```

If I just assume the template Ids would be the same for everyone and try the example:

```
$ GoSungrow show template points 8040
$
```

Silence. Ditto any other Template Ids I try. 

I can pick up the README again at:

```
$ GoSungrow show device list
$ GoSungrow show point data 20230217000000 20230217235959 120 1203625_1_1_1.p1
```

It's really the "no templates" that's the problem. I don't know whether to conclude:

* There are no templates for the SG5.0RS; or
* There's some setup step I've missed somewhere which would initialise the templates; or
* This bit of GoSungrow is broken at the moment so I may as well just use `show point data` to pick out what I need; or
* I've somehow uncovered a bug somewhere.

No other issue seems to mention templates so my guess is it's something peculiar to my site (maybe the installing electrician forgot a step). I don't have another Sungrow inverter so I've nothing to compare with.

On the off-chance it would make a difference, I compiled on both macOS and Raspberry Pi OS. Same behaviour.

Any guidance would be much appreciated.

scarafaggio, 2023-02-28:
FWIW I have a SH6.0RS and  _GoSungrow show template list_  lists 5 templates which seem to work well.   

I haven't figured out where these are defined, or whether it is possible to define custom templates which would be nice.

Paraphraser, 2023-03-01:
I think I'll assume this will either come good in time (possibly as a result of me poking the right button in iSolarCloud, or maybe these things just take time to populate), or it's simply not a feature of the SGx.0RS.

closed 2023-03-01 by Paraphraser