MickMake/GoSungrow #38: Multiple inverters crash GoSunGrow

bjornhi, 2023-02-24:
I have two Sungrow inverters connected to iSolarcloud through the Sungrow Datalogger 10000. But during parsing GoSunGrow crashes. I suspect it is due to the multiple inverters:

```
Token:	115573_6fbf31c8ddcc4edc920e374162efd8bb
Token File:	/data/.GoSungrow/AppService_login.json
2023/02/24 06:48:42 INFO: Found SunGrow 3 devices
2023/02/24 06:48:42 INFO: Caching Sungrow metadata...
2023/02/24 06:48:42 INFO: Cached 750 Sungrow data points...
PsId: required
JSON request:	{"ps_id":5103517}
2023/02/24 06:48:43 ERROR: strconv.Atoi: parsing "2146431_2146432": invalid syntax
Error: strconv.Atoi: parsing "2146431_2146432": invalid syntax
Usage:
  GoSungrow mqtt run [flags]
Aliases:
  run, 
Examples:
	GoSungrow mqtt run  
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: strconv.Atoi: parsing "2146431_2146432": invalid syntax
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

```

If it would be of any help I can give you access to our installation. 

Thanks in advance!

w2cker, 2023-03-14:
I have 3 inverters and also an error on start. Can this be the same issue?

`Token File:	/data/.GoSungrow/AppService_login.json
[12:18:13] INFO: Syncing data from gateway https://gateway.isolarcloud.eu ...
2023/03/14 12:18:13 INFO: Connecting to MQTT HASSIO Service...
2023/03/14 12:18:13 INFO: Connecting to SunGrow...
2023/03/14 12:18:13 INFO: Found SunGrow 13 devices
2023/03/14 12:18:13 INFO: Caching Sungrow metadata...
2023/03/14 12:18:14 INFO: Cached 810 Sungrow data points...
PsId: required
JSON request:	{"ps_id":5052959}
2023/03/14 12:18:14 ERROR: strconv.Atoi: parsing "null": invalid syntax
Error: strconv.Atoi: parsing "null": invalid syntax
Usage:
  GoSungrow mqtt run [flags]
Aliases:
  run, 
Examples:
	GoSungrow mqtt run  
Flags: Use "GoSungrow help flags" for more info.`

bleughb, 2023-05-01:
I have two SG5.0RS inverters here and it works fine

varpans, 2023-09-06:
I have the same issue. Have 2 inverters as well. 
I have two SG40CX on my isolarcloud account connected through WiNet-S dongles (1 for each inverter). **Anyone have a clue?** 

**From log:**
*
_[10:19:45] INFO: Syncing data from gateway https://gateway.isolarcloud.eu ...
2023/09/06 10:19:45 INFO: Connecting to MQTT HASSIO Service...
2023/09/06 10:19:45 INFO: Connecting to SunGrow...
2023/09/06 10:19:45 INFO: Found SunGrow 4 devices
2023/09/06 10:19:45 INFO: Caching Sungrow metadata...
2023/09/06 10:19:45 INFO: Cached 769 Sungrow data points...
PsId: required
JSON request:	{"ps_id":5242884}
2023/09/06 10:19:46 ERROR: strconv.Atoi: parsing "3127468_3127469": invalid syntax
Error: strconv.Atoi: parsing "3127468_3127469": invalid syntax
Usage:
  GoSungrow mqtt run [flags]
Aliases:
  run, 
Examples:
	GoSungrow mqtt run  
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: strconv.Atoi: parsing "3127468_3127469": invalid syntax
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped_
*