MickMake/GoSungrow #45: PS ID Required?

scraja69, 2023-03-26:
Hi Mick,
Thank you for your project. I have installed the addon and it is asking for  PS ID? Where should I Enter that data?
please see log below. Can anyone please help. I am quiet newbie to Go lang

 --quiet         |            | GOSUNGROW_QUIET         | GoSungrow: Silence all         | false *                           |
|                 |            |                         | messages.                      |                                   |
| --timeout       |            | GOSUNGROW_TIMEOUT       | Web timeout.                   | 1m0s                              |
| --user          | -u         | GOSUNGROW_USER          | SunGrow: api username.         |##########@gmail.com             |
| --password      | -p         | GOSUNGROW_PASSWORD      | SunGrow: api password.         |########                       |
| --appkey        |            | GOSUNGROW_APPKEY        | SunGrow: api application key.  | 93D72E60331ABDCDC7B39ADC2D1F32B3  |
|                 |            |                         |                                | *                                 |
| --host          |            | GOSUNGROW_HOST          | SunGrow: Provider API URL.     | https://augateway.isolarcloud.com |
|                 |            |                         |                                | *                                 |
| --token-expiry  |            | GOSUNGROW_TOKEN_EXPIRY  | SunGrow: last login.           | 2023-03-26T15:05:32               |
| --save          | -s         | GOSUNGROW_SAVE          | Save output as a file.         | false *                           |
| --mqtt-user     |            | GOSUNGROW_MQTT_USER     | HASSIO: mqtt username.         | mqttscr                           |
| --mqtt-password |            | GOSUNGROW_MQTT_PASSWORD | HASSIO: mqtt password.         | mqttscr                           |
| --mqtt-host     |            | GOSUNGROW_MQTT_HOST     | HASSIO: mqtt host.             | core-mosquitto                    |
| --mqtt-port     |            | GOSUNGROW_MQTT_PORT     | HASSIO: mqtt port.             |                              1883 |
+-----------------+------------+-------------------------+--------------------------------+-----------------------------------+
[07:05:44] INFO: Login to iSolarCloud using gateway https://augateway.isolarcloud.com ...
Email:	##########@gmail.com
Create Date:	Wed Dec 23 16:37:22 CST 2020
Login Last Date:	2023-03-26 15:05:32
Login Last IP:	
Login State:	1
User Account:	hxdb6m5x
User Id:	183472
User Name:	Chanaka
Is Online:	false
Token:	183472_b8f295b48bf146d594e4a092eba69b35
Token File:	/data/.GoSungrow/AppService_login.json
Email:	##########@gmail.com
Create Date:	Wed Dec 23 16:37:22 CST 2020
Login Last Date:	2023-03-26 15:05:45
Login Last IP:	
Login State:	1
User Account:	hxdb6m5x
User Id:	183472
User Name:	Chanaka
Is Online:	false
Token:	183472_b8f295b48bf146d594e4a092eba69b35
Token File:	/data/.GoSungrow/AppService_login.json
[07:05:45] INFO: Syncing data from gateway https://augateway.isolarcloud.com ...
Email:	###########@gmail.com
2023/03/26 07:05:45 INFO: Connecting to MQTT HASSIO Service...
2023/03/26 07:05:45 INFO: Connecting to SunGrow...
Create Date:	Wed Dec 23 16:37:22 CST 2020
Login Last Date:	2023-03-26 15:05:45
Login Last IP:	
Login State:	1
User Account:	hxdb6m5x
User Id:	183472
User Name:	Chanaka
Is Online:	false
Token:	183472_b8f295b48bf146d594e4a092eba69b35
Token File:	/data/.GoSungrow/AppService_login.json
2023/03/26 07:05:45 INFO: Found SunGrow 2 devices
2023/03/26 07:05:45 Option[homeassistant/select/GoSungrow/GoSungrow-option-mqtt_debug/cmd] set to 'enabled'
2023/03/26 07:05:45 MQTT[homeassistant/select/GoSungrow/GoSungrow-option-loglevel/state] -> info
2023/03/26 07:05:45 MQTT[homeassistant/select/GoSungrow/GoSungrow-option-fetchschedule/config] -> {"command_template":"{{ value }}","command_topic":"homeassistant/select/GoSungrow/GoSungrow-option-fetchschedule/cmd","device":{"connections":[["GoSungrow","GoSungrow-options"],["GoSungrow-options","GoSungrow"]],"identifiers":["GoSungrow"],"manufacturer":"GoSungrow","model":"GoSungrow","name":"GoSungrow options - Options","suggested_area":"GoSungrow","sw_version":"GoSungrow https://github.com/MickMake/GoSungrow","via_device":"GoSungrow"},"enabled_by_default":true,"encoding":"utf-8","icon":"mdi:format-list-group","name":"GoSungrow Option Fetch Schedule","object_id":"GoSungrow-option-fetchschedule","options":["2m","3m","4m","5m","6m","7m","8m","9m","10m"],"retain":true,"state_topic":"homeassistant/select/GoSungrow/GoSungrow-option-fetchschedule/state","unique_id":"GoSungrow-option-fetchschedule","value_template":"{{ value }}"}
2023/03/26 07:05:45 INFO: Option[loglevel] set to 'info'
2023/03/26 07:05:45 MQTT[homeassistant/select/GoSungrow/GoSungrow-option-loglevel/state] -> info
2023/03/26 07:05:45 INFO: Option[fetchschedule] set to '4m'
2023/03/26 07:05:45 MQTT[homeassistant/select/GoSungrow/GoSungrow-option-fetchschedule/state] -> 4m
2023/03/26 07:05:45 MQTT[homeassistant/select/GoSungrow/GoSungrow-option-fetchschedule/state] -> 4m
2023/03/26 07:05:45 MQTT[homeassistant/select/GoSungrow/GoSungrow-option-sleepdelay/config] -> {"command_template":"{{ value }}","command_topic":"homeassistant/select/GoSungrow/GoSungrow-option-sleepdelay/cmd","device":{"connections":[["GoSungrow","GoSungrow-options"],["GoSungrow-options","GoSungrow"]],"identifiers":["GoSungrow"],"manufacturer":"GoSungrow","model":"GoSungrow","name":"GoSungrow options - Options","suggested_area":"GoSungrow","sw_version":"GoSungrow https://github.com/MickMake/GoSungrow","via_device":"GoSungrow"},"enabled_by_default":true,"encoding":"utf-8","icon":"mdi:format-list-group","name":"GoSungrow Option Sleep Delay After Schedule","object_id":"GoSungrow-option-sleepdelay","options":["0s","10s","20s","30s","40s","50s","60s"],"retain":true,"state_topic":"homeassistant/select/GoSungrow/GoSungrow-option-sleepdelay/state","unique_id":"GoSungrow-option-sleepdelay","value_template":"{{ value }}"}
2023/03/26 07:05:45 MQTT[homeassistant/select/GoSungrow/GoSungrow-option-sleepdelay/state] -> 40s
2023/03/26 07:05:45 MQTT[homeassistant/select/GoSungrow/GoSungrow-option-servicestate/config] -> {"command_template":"{{ value }}","command_topic":"homeassistant/select/GoSungrow/GoSungrow-option-servicestate/cmd","device":{"connections":[["GoSungrow","GoSungrow-options"],["GoSungrow-options","GoSungrow"]],"identifiers":["GoSungrow"],"manufacturer":"GoSungrow","model":"GoSungrow","name":"GoSungrow options - Options","suggested_area":"GoSungrow","sw_version":"GoSungrow https://github.com/MickMake/GoSungrow","via_device":"GoSungrow"},"enabled_by_default":true,"encoding":"utf-8","icon":"mdi:format-list-group","name":"GoSungrow Option Service State","object_id":"GoSungrow-option-servicestate","options":["Run","Restart","Stop"],"retain":true,"state_topic":"homeassistant/select/GoSungrow/GoSungrow-option-servicestate/state","unique_id":"GoSungrow-option-servicestate","value_template":"{{ value }}"}
2023/03/26 07:05:45 INFO: Option[sleepdelay] set to '60s'
2023/03/26 07:05:45 MQTT[homeassistant/select/GoSungrow/GoSungrow-option-sleepdelay/state] -> 60s
2023/03/26 07:05:45 MQTT[homeassistant/select/GoSungrow/GoSungrow-option-servicestate/state] -> Run
2023/03/26 07:05:45 INFO: Caching Sungrow metadata...
2023/03/26 07:05:45 INFO: Option[servicestate] set to 'run'
2023/03/26 07:05:45 MQTT[homeassistant/select/GoSungrow/GoSungrow-option-servicestate/state] -> run
2023/03/26 07:05:45 INFO: Cached 681 Sungrow data points...
PsId: required
JSON request:	{"ps_id":1204183}
2023/03/26 07:05:45 ERROR: strconv.Atoi: parsing "1382562_1382563": invalid syntax
Error: strconv.Atoi: parsing "1382562_1382563": invalid syntax
Usage:
  GoSungrow mqtt run [flags]
Aliases:
  run, 
Examples:
	GoSungrow mqtt run  
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: strconv.Atoi: parsing "1382562_1382563": invalid syntax
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

scraja69, 2023-03-26:
I tried from Gosungrow tool for mqqt run and got following log..
C:\Users\scraj\Downloads\GoSungrow-windows_amd64>GoSungrow mqtt run
Logging in again
2023/03/26 17:32:11 INFO: Connecting to MQTT HASSIO Service...
2023/03/26 17:32:11 INFO: Connecting to SunGrow...
2023/03/26 17:32:11 INFO: Found SunGrow 2 devices
Error: network Error : dial tcp: lookup Core-mosquitto: no such host
←[96;1;4mUsage:←[0m
  ←[37;1mGoSungrow←[0m mqtt run [flags]

←[96;1;4mAliases:←[0m
  run,

←[96;1;4mExamples:←[0m
←[37;3m GoSungrow mqtt run
←[0m

Flags: Use "GoSungrow help flags" for more info.

←[96;1;4mAdditional help topics:←[0m

ERROR: network Error : dial tcp: lookup Core-mosquitto: no such host

C:\Users\scraj\Downloads\GoSungrow-windows_amd64>

PurserS, 2023-05-30:
HI scraja69,

I get the same issue of invalid syntax after succesfully logging to isolar
I use a Raspberry PI 

2023/05/30 05:51:44 INFO: Cached 687 Sungrow data points...
PsId: required
JSON request:	{"ps_id":5095269}
2023/05/30 05:51:45 ERROR: strconv.Atoi: parsing "1879717_1879718": invalid syntax
Error: strconv.Atoi: parsing "1879717_1879718": invalid syntax
Usage:
  GoSungrow mqtt run [flags]
Aliases:
  run, 
Examples:
	GoSungrow mqtt run  
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: strconv.Atoi: parsing "1879717_1879718": invalid syntax
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped