MickMake/GoSungrow #47: How to access predefined / user-defined reports

thankthemaker, 2023-04-10:
I use 3.0.4 and I'd like to automate sending a daily report via mail. In the web UI it's possible to use predifined reports as well as user-defined reports. How can I use these reports via GoSungrow?

I saw there's an api "getReportMsg" but I wasn't able to find out how to use it. I tried to call it like this

`./GoSungrow api get AppService.queryReportMsg`

but it returns an error message:

`Error: unknown error 'er_missing_parameter:menu_code'`

Is getting the reports supported by GoSungrow?

thankthemaker, 2023-04-10:
BTW: I know there's also such a feature available in the UI.

thankthemaker, 2023-04-10:
Ok, found it in the examples
`GoSungrow api get getPsReport '{"report_type":"1","date_id":"20220201","date_type":"1","ps_id":"1171348"}'`

closed 2023-04-10 by thankthemaker

scraja69, 2023-04-19:
Hi,
How do you install version 3.0.4 through addons. It will not allow me. Please See below..
Can you please help?

[image]

thankthemaker, 2023-04-19:
I don't use GoSungrow with HomeAssistent, instead I'm using it standanlon. Maybe someone else can answer your question.