MickMake/GoSungrow #54: Error: identifier rejected - Found SunGrow 0 devices

hboltz, 2023-05-11:
I installed your HA Add-On, but I didn't get it working, The problem is always the same:

[18:35:51] INFO: Login to iSolarCloud using gateway https://gateway.isolarcloud.eu ...
Email:	xxxxxxxxx@xxxxx.de
Create Date:	Fri May 05 20:03:41 CST 2023
Login Last Date:	2023-05-12 02:35:51
Login Last IP:	
Login State:	1
User Account:	xxxxxxxxx
User Id:	xxxxxx
User Name:	xxxxxxxx
Is Online:	false
Token:	216703_85bf00ad816c4c2f9fd4fa87320c6a9e
Token File:	/data/.GoSungrow/AppService_login.json
[18:35:51] INFO: Syncing data from gateway https://gateway.isolarcloud.eu ...
2023/05/11 18:35:51 INFO: Connecting to MQTT HASSIO Service...
2023/05/11 18:35:51 INFO: Connecting to SunGrow...
2023/05/11 18:35:52 INFO: Found SunGrow 0 devices
Error: identifier rejected
Usage:
  GoSungrow mqtt run [flags]
Aliases:
  run, 
Examples:
	GoSungrow mqtt run  
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: identifier rejected
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Afterwards, the Add-On stopped and no data is seen. Any idea on this?

hboltz, 2023-05-11:
#Update1: I configured a MQTT user and put the credentials into the the GoSungrow config. Now it looks a little bit other but I think it's still not working:

[19:14:11] INFO: Syncing data from gateway https://gateway.isolarcloud.eu ...
2023/05/11 19:14:11 INFO: Connecting to MQTT HASSIO Service...
2023/05/11 19:14:11 INFO: Connecting to SunGrow...
2023/05/11 19:14:11 INFO: **Found SunGrow 0 devices**
2023/05/11 19:14:11 INFO: Caching Sungrow metadata...
2023/05/11 19:14:11 INFO: Cached 0 Sungrow data points...
No results found.
2023/05/11 19:14:11 INFO: Syncing 1 entries with HASSIO from getPsList.
CU
2023/05/11 19:14:11 INFO: Starting ticker...
2023/05/11 19:14:11 INFO: Fetch Schedule: 5m
2023/05/11 19:14:11 INFO: Sleep Delay:    40s

hboltz, 2023-06-06:
Problem solved by myself

closed 2023-06-06 by hboltz

Simon-Str, 2024-01-29:
> #Update1: I configured a MQTT user and put the credentials into the the GoSungrow config. Now it looks a little bit other but I think it's still not working:
>
> [19:14:11] INFO: Syncing data from [...]

This looks like my log. But it doesn't seem to be working. Did you get it working?