MickMake/GoSungrow #57: Update Entity only all others are missing

keensy, 2023-06-04:
Hello , I seem to only have one entity "update.gosungrow" and no othe

hboltz, 2023-06-13:
Same for me!

hboltz, 2023-06-30:
unfortnunatelly there is no further documentation about this HA addon. In my installation, I configured the addon in the right way because in the logs, I can see, that there is a lot of stuff discovered:

2023/06/30 06:17:18 INFO: Found SunGrow 2 devices
2023/06/30 06:17:18 INFO: Caching Sungrow metadata...
2023/06/30 06:17:19 INFO: Cached 250 Sungrow data points...
2023/06/30 06:17:19 INFO: Syncing 148 entries with HASSIO from getPsList.
2023/06/30 06:17:19 INFO: Syncing 246 entries with HASSIO from getPsDetail.
2023/06/30 06:17:19 INFO: Syncing 1565 entries with HASSIO from queryDeviceList.

My question is: how to proceed with this addon? Which steps are the next to do?