MickMake/GoSungrow #58: How to get PS ID??

hboltz, 2023-06-06:
Hi Mick,

I installed your Addon to HA and it worked so far:

2023/06/06 14:29:38 INFO: Connecting to SunGrow...
2023/06/06 14:29:38 INFO: Found SunGrow 2 devices
2023/06/06 14:29:38 INFO: Caching Sungrow metadata...
2023/06/06 14:29:38 INFO: Cached 256 Sungrow data points...
2023/06/06 14:29:39 INFO: Syncing 1565 entries with HASSIO from queryDeviceList.
CU--CUCUCU-CUCUCU--CUCUCUCUCUCUCUCU-CU-CUCU-CUCUCUCUCUCUCUCUCU-CUCUCUCUCUCU-CUCU-CUCUCUCUCU-CUCU--CUCU-CUCU-CUCUCUCUCUCU---CUCU-CU-CUCU---CU-CUCU-CU-CU-CUCU---CU-CUCUCU--CU--------CUCUCUCUCUCUCUCUCU-CUCU--CUCUCUCUCUCUCUCUCUCUCU-----CUCUCU-CUCU-CUCU--CUCUCUCUCU-CU--CUCUCUCU----CU-CUCU
2023/06/06 14:29:39 INFO: Syncing 148 entries with HASSIO from getPsList.
2023/06/06 14:29:39 INFO: Syncing 246 entries with HASSIO from getPsDetail.
CUCUCUCUCUCUCUCUCUCUCUCUCU??CUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCU?CUCU?CUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCU
?CUCU?CUCUCU?CUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCU?CU?CUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCUCUCU?CUCUCU?CUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCU?CUCUCUCUCUCUCUCU
2023/06/06 14:29:39 INFO: Starting ticker...
2023/06/06 14:29:39 INFO: Fetch Schedule: 5m
2023/06/06 14:29:39 INFO: Sleep Delay:    40s

To go on with the graphical part, I need to know my PS ID but unfortunatelly I can't issue any command from your description in the Terminal (see attachment). Any hint for me??

[image]

Paraphraser, 2023-06-08:
Well, I don't know if this will help but, when I did all this shortly after my SG5.0RS was installed last Feb, I'm pretty sure I just followed the README.

> For me, the tricky bit came *after* establishing basic connectivity, figuring out which metrics I actually wanted to extract every day.

To make sure the basic setup still works "as documented" (well, for me, at least), I just did the following. 

<hr>

Move the existing GoSungrow configuration out of the way so I'm starting from a clean slate.

```
$ cd
$ mv .GoSungrow .GoSungrow.off
```

Initialise a new configuration. If I remember correctly, the email address and password were set up as part of getting the iPad app going and connected to the inverter.

```
$ GoSungrow config write --user=my.email@my.domain.com --password=mypassword
Using config file '~/.GoSungrow/config.json'
New config:
+-----------------+------------+-------------------------+--------------------------------+-----------------------------------+
|      FLAG       | SHORT FLAG |       ENVIRONMENT       |          DESCRIPTION           |        VALUE (* = DEFAULT)        |
+-----------------+------------+-------------------------+--------------------------------+-----------------------------------+
| --config        |            | GOSUNGROW_CONFIG        | GoSungrow: config file.        | ~/.GoSungrow/config.json   |
| --debug         |            | GOSUNGROW_DEBUG         | GoSungrow: Debug mode.         | false *                           |
| --quiet         |            | GOSUNGROW_QUIET         | GoSungrow: Silence all         | false *                           |
|                 |            |                         | messages.                      |                                   |
| --timeout       |            | GOSUNGROW_TIMEOUT       | Web timeout.                   | 30s *                             |
| --user          | -u         | GOSUNGROW_USER          | SunGrow: api username.         | my.email@my.domain.com             |
| --password      | -p         | GOSUNGROW_PASSWORD      | SunGrow: api password.         | mypassword                |
| --appkey        |            | GOSUNGROW_APPKEY        | SunGrow: api application key.  | AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA  |
|                 |            |                         |                                | *                                 |
| --host          |            | GOSUNGROW_HOST          | SunGrow: Provider API URL.     | https://augateway.isolarcloud.com |
|                 |            |                         |                                | *                                 |
| --token-expiry  |            | GOSUNGROW_TOKEN_EXPIRY  | SunGrow: last login.           |  *                                |
| --save          | -s         | GOSUNGROW_SAVE          | Save output as a file.         | false *                           |
| --mqtt-user     |            | GOSUNGROW_MQTT_USER     | HASSIO: mqtt username.         |  *                                |
| --mqtt-password |            | GOSUNGROW_MQTT_PASSWORD | HASSIO: mqtt password.         |  *                                |
| --mqtt-host     |            | GOSUNGROW_MQTT_HOST     | HASSIO: mqtt host.             |  *                                |
| --mqtt-port     |            | GOSUNGROW_MQTT_PORT     | HASSIO: mqtt port.             |  *                                |
+-----------------+------------+-------------------------+--------------------------------+-----------------------------------+
```

Trigger a login.

```
$ GoSungrow api login
Email:	my.email@my.domain.com
Create Date:	Thu Feb 09 13:14:55 CST 2023
Login Last Date:	2023-06-08 10:24:05
Login Last IP:	
Login State:	1
User Account:	xxxxxxxxxx
User Id:	999999
User Name:	myname
Is Online:	false
Token:	123456_123456789abcdef123456789abcedf00
Token File:	~/.GoSungrow/AppService_login.json
```

Show my devices.

```
$ GoSungrow show ps list
┏━━━━━━━━━━━━━━━━━━┳━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━┓
┃ Ps Key           ┃ Ps Id   ┃ Device Type ┃ Device Code ┃ Channel Id ┃ Serial #    ┃ Factory Name ┃ Device Model   ┃
┣━━━━━━━━━━━━━━━━━━╇━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━┫
┃ 9999999_1_1_1    │ 9999999 │ 1           │ 1           │ 1          │ B9999999999 │ SUNGROW      │ SG5.0RS        ┃
┃ 9999999_22_247_1 │ 9999999 │ 22          │ 247         │ 1          │ B9999999999 │ SUNGROW      │ WiNet-S        ┃
[... 32 lines cut ...]
```
$ mkdir -p /root/bin
$ cd /root/bin
$ wget -q https://github.com/MickMake/GoSungrow/releases/download/v3.0.4/GoSungrow-linux_arm64.tar.gz
$ tar -xzf GoSungrow-linux_arm64.tar.gz
$ rm GoSungrow-linux_arm64.tar.gz
$ cd ..
$ ./bin/GoSungrow config write --user=YOUR-EMAIL-ADDRESS --password=YOUR-PASSWORD
$ ./bin/GoSungrow api login
$ ./bin/GoSungrow show ps list
```

You'll need to replace the `GoSungrow-linux_arm64` with the correct architecture for your system. In words:

1. Create the `bin` directory if it does not exist.
2. Change into the `bin` directory.
3. Download the distribution from GitHub.
4. Unpack the compressed archive into the current directory (ie `bin`).
5. Remove the distribution.
6. Change into the parent directory (ie `/root`).
7. Run the command to create the configuration.
8. Trigger an API login.
9. Fetch the PS list and make a note of your ID.

I hope all this helps you get started. My experience with GoSungrow is that it hasn't missed a beat since I got it to do what I wanted. That's a daily `cron` job to pull down "yesterday's" data for the 12 metrics I'm actually interested in.

hboltz, 2023-06-09:
Thanks a lot for this hints! But unfortunately I can't see the .GoSungrow directory from the first point of your explanation. 

[image]

Also the directory which was mentioned in the log of the Addon, is not visible for me:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[06:37:07] INFO: Setting up GoSungrow config ...
[06:37:08] INFO: Writing GoSungrow config ...
New config:
Using config file '/data/.GoSungrow/config.json'

I cannot find the config.json file! But I think that's not the real problem. I can't issue any of the Golang commands in my terminal of HA. So I am not able to get my PS ID.

hboltz, 2023-06-12:
> However, your output says the .GoSungrow directory is in the /data directory, rather than in /root. Are you saying it isn't there when you go looking in /data?

That's right! Her is the content of /data:

[image]

And you're right - I am running HA as an appliance in a VM under Proxmox. I'll try to install the GoSungrow on my MAC. Maybe there's a possibility to get the PS ID......

hboltz, 2023-06-13:
Good News today: I setup a Ubuntu VM on my Proxmox and within this, I could issue all GoSungrow commands. So now I have my PS ID!! Is anyone aware how to proceed with the graphical part of the installation of the GoSungrow Addon in Home            Assistant? Any help is welcome.

hboltz, 2023-06-20:
See above

closed 2023-06-20 by hboltz