MickMake/GoSungrow #62: Just can't get this to work, panic: runtime error: invalid memory address or nil pointer dereference

Brettkelstar, 2023-07-02:
Hi there,

Brand new user to HA so please forgive my lack of understanding if I have missed something basic

I have completed all the instructions multiple times and continue to be unable to find the data in my Home Assistant to use with integrations.   I feel like its to do with this part of the log:
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0xc8 pc=0x441a60]

Any help would be hugely appreciated

This is the full log below

15:27:01] INFO: Syncing data from gateway https://augateway.isolarcloud.com ...
2023/07/02 15:27:01 INFO: Connecting to MQTT HASSIO Service...
2023/07/02 15:27:01 INFO: Connecting to SunGrow...
2023/07/02 15:27:01 INFO: Found SunGrow 2 devices
2023/07/02 15:27:01 INFO: Caching Sungrow metadata...
2023/07/02 15:27:01 INFO: Cached 250 Sungrow data points...
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0xc8 pc=0x441a60]
goroutine 1 [running]:
GoSungrow/iSolarCloud/AppService/queryDeviceList.(*EndPoint).SetPvPoints(_, {{_, _, _}}, {0x40006ba630, {0x40004233b6, 0x6}, {0xc12083b18021e35e, 0x90c1628, 0x411eca0}, ...})
	/Users/mick/Documents/GoLang/Solar/GoSungrow/iSolarCloud/AppService/queryDeviceList/data.go:419 +0x630
GoSungrow/iSolarCloud/AppService/queryDeviceList.(*EndPoint).GetEnergyStorageSystem(_, {0x40006ba630, {0x40004233b6, 0x6}, {0xc12083b18021e35e, 0x90c1628, 0x411eca0}, {{0x40005a4860, 0x2, 0x2}}, ...})
	/Users/mick/Documents/GoLang/Solar/GoSungrow/iSolarCloud/AppService/queryDeviceList/data.go:299 +0x2e0
GoSungrow/iSolarCloud/AppService/queryDeviceList.(*EndPoint).GetData(_)
	/Users/mick/Documents/GoLang/Solar/GoSungrow/iSolarCloud/AppService/queryDeviceList/data.go:259 +0x94
GoSungrow/iSolarCloud/AppService/queryDeviceList.EndPoint.GetEndPointData(...)
	/Users/mick/Documents/GoLang/Solar/GoSungrow/iSolarCloud/AppService/queryDeviceList/struct.go:367
GoSungrow/iSolarCloud.(*SunGrowData).CallEndpoint(_, {_, _}, {{0x40005dced0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, ...}, ...})
	/Users/mick/Documents/GoLang/Solar/GoSungrow/iSolarCloud/data.go:161 +0x344
GoSungrow/iSolarCloud.(*SunGrowData).getDataSinglePsIdRequired(0x4000c8f460, {0x27661a8, 0x4000022e00})
	/Users/mick/Documents/GoLang/Solar/GoSungrow/iSolarCloud/data.go:281 +0x254
GoSungrow/iSolarCloud.(*SunGrowData).GetDataSingle(0x4000c8f460, {0x40008cad20, 0xf})
	/Users/mick/Documents/GoLang/Solar/GoSungrow/iSolarCloud/data.go:236 +0x124
GoSungrow/iSolarCloud.(*SunGrowData).GetData(0x4000c8f460)
	/Users/mick/Documents/GoLang/Solar/GoSungrow/iSolarCloud/data.go:207 +0x134
GoSungrow/cmd.(*CmdMqtt).Cron(0x4000126140)
	/Users/mick/Documents/GoLang/Solar/GoSungrow/cmd/cmd_mqtt.go:370 +0x200
GoSungrow/cmd.(*CmdMqtt).CmdMqttRun(0x4000126140, 0x0?, {0x0?, 0x0?, 0x0?})
	/Users/mick/Documents/GoLang/Solar/GoSungrow/cmd/cmd_mqtt.go:268 +0x70
github.com/spf13/cobra.(*Command).execute(0x400013a600, {0x4000188170, 0x0, 0x0})
	/Users/mick/go/pkg/mod/github.com/spf13/cobra@v1.6.1/command.go:916 +0x5e0
github.com/spf13/cobra.(*Command).ExecuteC(0x40000fa000)
	/Users/mick/go/pkg/mod/github.com/spf13/cobra@v1.6.1/command.go:1044 +0x368
github.com/spf13/cobra.(*Command).Execute(...)
	/Users/mick/go/pkg/mod/github.com/spf13/cobra@v1.6.1/command.go:968
github.com/MickMake/GoUnify/Unify.(*Commands).Execute(...)
	/Users/mick/go/pkg/mod/github.com/!mick!make/!go!unify/!unify@v0.0.0-20221125023651-ff4a37b1928a/struct.go:258
github.com/MickMake/GoUnify/Unify.(*Unify).Execute(0x4000000b40)
	/Users/mick/go/pkg/mod/github.com/!mick!make/!go!unify/!unify@v0.0.0-20221125023651-ff4a37b1928a/struct.go:201 +0x304
GoSungrow/cmd.Execute(...)
	/Users/mick/Documents/GoLang/Solar/GoSungrow/cmd/commands.go:90
main.main()
	/Users/mick/Documents/GoLang/Solar/GoSungrow/main.go:11 +0x6c
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

nickallan, 2023-08-05:
Having the exact same issue here.

burnsy180, 2023-08-14:
Same errors here. Haven't gotten this working yet

dbr18210, 2023-10-25:
I have the same issue as well.