MickMake/GoSungrow #65: isolarcloud goes offline at night

helikopter1909, 2023-07-06:
Hello everyone,

I have now noticed the following issue for the third night in a row: the communication module /isolarcloud app always goes offline after sunset (currently around 10 p.m.) and in the morning at sunrise (currently around 5:20 a.m.) the communication automatically reconnects and is online. So far not bad, but this means that the network consumption is no longer displayed correctly in the statistics.

I have not activated energy saving mode in my router. Up until three days ago, I didn't know about the abnormality either.

Do you know the issue?

I have the following hardware:

Inverter: SG10KTL-M

Communication: e-net v11

regards, Matthias

helikopter1909, 2023-07-12:
Normally the communication module e-net doesn't have a night/sleep.mode in opposition to the winet module. Does the app have the power to activate a sleeping/night mode? How can I deactivate it?

helikopter1909, 2023-07-18:
Does anyone else have this problem?  Or maybe it's a bug on Sungrow side, but I don't know... Short answer please.

Paraphraser, 2023-07-18:
Well, the most I can say is that I have a SG5.0RS and it does **not** exhibit this pattern. I am 100% certain of that because (a) I am reliably getting 288 log entries per day (288 entries * 5 minute intervals * 60 = 86,400 seconds) and (b) I log "commits" from my DHCP server and the only time the Sungrow shows up is if the lease expires and is renewed. It definitely doesn't happen every 24 hours. I haven't done anything special to make that happen. It's all just as the inverter came out of the box.

Does that help?

helikopter1909, 2023-07-18:
yes for sure. I guess i have to switch to another communication dongle, from e-net V11 to winet s

bleughb, 2023-07-19:
I found my two SG5.0RS going offline regularly
I went through the received entities and deleted all the ones not relevant to me - and dropped the interrogation to 60s

This reduced the number of queries to the inverters enough so they’re now super stable online

I’m pulling about 30 entities every minute via the wifi

helikopter1909, 2023-07-19:
Thanks for the idea to reduce the number of entities. I will try it, but I think I have an other issue. Because my inverter goes offline at sunset - every day. And go online at sunrise. But you help me, at first I thought the app set the inverter offline at sunset ...