MickMake/GoSungrow #67: MQTT: trying to acces host 127.0.0.11:53

HansH2807, 2023-07-10:
I had the HA Sungrow-Addon already running some weeks ago, however I could not see any entity related to it.

So I suspected that my MQTT environment with a second MQTT-broker was making issues and I reconfigered my network so that all MQTT clients now are using the MQTT-broker provided by HA. However now  the HA-Addon cannot be started correctly anymore. The protocol says at the end:

```
.....
Token File:	/data/.GoSungrow/AppService_login.json
2023/07/10 16:27:54 INFO: Found SunGrow 3 devices
Error: network Error : dial tcp: lookup mqtt2 on 127.0.0.11:53: no such host
Usage:
  GoSungrow mqtt run [flags]
Aliases:
  run, 
Examples:
	GoSungrow mqtt run  
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: network Error : dial tcp: lookup mqtt2 on 127.0.0.11:53: no such host
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```
FYI: 
1. I have the GoSungrow.exe application running correctly, it show me correctly e.g. my PsID and also reports all the correct login data!
2. I can verify by use of MQTT-Explorer that the data I configured in the HA-addon are correct.

Can anybody tell me why the HA-Addon tries to use the MQTT-broker with address 127.0.0.11:53 and what I can do to make the Addon start correctly?

Thanks for any help!

Best Regards
Hans

Paraphraser, 2023-07-10:
I will start by saying that I don't run Home Assistant so what I say next might not be applicable.

I *have* seen the 127.0.0.11:53 address+port combination before.

In theory, the whole 127/8 network (ie all hosts from 127.0.0.1 through 127.255.255.254) is reserved for loopback. Using 127.0.0.1 is just a convention.

Docker (mis)uses 127.0.0.11 to mean "the DNS resolver for this container":

```
$ docker exec nodered cat /etc/resolv.conf
nameserver 127.0.0.11
options ndots:0
$ 
```

If you are running multiple Docker containers in non-host mode then, providing those containers are attached to the same internal bridged network, each container can reach every other container via `«containerName»:«port»` syntax. The Docker resolver on 127.0.0.11 provides the lookup service that makes that work.

```
$ docker exec nodered dig mosquitto

; <<>> DiG 9.18.13 <<>> mosquitto
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 65393
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 0

;; QUESTION SECTION:
;mosquitto.			IN	A

;; ANSWER SECTION:
mosquitto.		600	IN	A	172.30.0.5

;; Query time: 0 msec
;; SERVER: 127.0.0.11#53(127.0.0.11) (UDP)
;; WHEN: Tue Jul 11 08:32:15 AEST 2023
;; MSG SIZE  rcvd: 52
```

In the third-to-last line, you can see the 127.0.0.11 as the address of the name server that's responding to the query. The answer it's providing is that the Mosquitto container is associated with the IP address 172.30.0.5.

> When the Node-RED container uses `172.30.0.5:1883` to reach the Mosquitto container, the traffic is going direct (ie layer 2, not layer 3).

In my case, that Mosquitto container is also reachable using a fully-qualified domain name: 

```
$ docker exec nodered dig mosquitto.my.domain.com

; <<>> DiG 9.18.13 <<>> mosquitto.my.domain.com
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 59855
;; flags: qr aa rd ra; QUERY: 1, ANSWER: 2, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
; COOKIE: 5e2673842f8580d10100000064ac87f3ff828d624cd01435 (good)
;; QUESTION SECTION:
;mosquitto.my.domain.com.	IN	A

;; ANSWER SECTION:
mosquitto.my.domain.com. 60	IN	CNAME	iot-hub.my.domain.com.
iot-hub.my.domain.com. 10800	IN	A	   192.168.132.60

;; Query time: 0 msec
;; SERVER: 127.0.0.11#53(127.0.0.11) (UDP)
;; WHEN: Tue Jul 11 08:36:35 AEST 2023
;; MSG SIZE  rcvd: 120
```

The answer is still coming from 127.0.0.11 but the query is actually being relayed through the host's (Raspberry Pi's) resolver:

```
$ cat /etc/resolv.conf
# Generated by resolvconf
domain my.domain.com
nameserver 192.168.132.55
nameserver 192.168.132.57
```

The host where Node-RED and Mosquitto are running is .60 while .55 and .57 are running redundant copies of BIND. So, inside the Node-RED container, the DNS query is being directed to 127.0.0.11 (the container's resolver), which relays it to the resolver of the Pi on which the container is running (.60) which, in turn, relays it to one of the DNS servers running on .55 and .57.

> The answer is .60 so when the Node-RED container tries to reach 192.168.132.60:1883, the traffic will be routed (layer 3) out of container-space, and then be routed back into container-space.

To try to draw all of that together with what I'm seeing in your output, *something* (and I don't know what that *something* is) is asking 127.0.0.11 to provide an IP address for the host named "mqtt2".

Not having any idea of your actual environment (and, in particular, whether Docker is involved or not), it looks to me like this is a Docker container and there is:

1. No non-host-mode container named "mqtt2" attached to the same internal bridged network; or
2. The unqualified name "mqtt2" should be expressed as a fully-qualified domain name (so the Docker resolver knows it isn't a host name and will relay the query to the host's resolver); or
3. The host's resolver is unable to resolve the name.

It is not that the HA-Addon is trying to use the MQTT-broker with the address 127.0.0.11:53. It is that the HA-Addon is asking the DNS resolver at 127.0.0.11:53 to answer the question, "what is the IP address for `mqtt2`?" and the resolver is saying, "absolutely no idea".

Does that help?

HansH2807, 2023-07-11:
Hello Paraphraser,

thank you a lot for your (very) detalled explanation. I have not enough skills in docker configuration, however I roughly can  follow your explanation. It guides me to a direction which I already suspected. 
However the key point of your explanation is for me that you confirm to me that HA is looking for the host "mqtt2", which really is the user-name which I  wanted to use to contact the broker. The hostname of my broker would be "rpi5" or "rpi5.fritz.box", the Raspberry4 which I use to run HA. In any case I will try to make the HA addon to configure to use the correct mqtt-host. Maybe some one else is able to help me how to do this.  

Many thanks!!
Best Regards
Hans

khkissel, 2023-09-14:
Maybe there is a special character in the mqtt-password. See open issue #85

Shibby7986, 2024-06-10:
I had a similar error
'''2024/06/10 04:06:24 INFO: Connecting to MQTT HASSIO Service...
2024/06/10 04:06:24 INFO: Connecting to SunGrow...
2024/06/10 04:06:24 INFO: Found SunGrow 3 devices
Error: network Error : dial tcp: lookup core-mosquitto on 127.0.0.11:53: no such host
Usage:
  GoSungrow mqtt run [flags]
Aliases:
  run, 
Examples:
	GoSungrow mqtt run  

Flags: Use "GoSungrow help flags" for more info.

Additional help topics:
ERROR: network Error : dial tcp: lookup core-mosquitto on 127.0.0.11:53: no such host'''
and resolved it by installing the Mosquitto broker addon via the addon store in Home Assistant