MickMake/GoSungrow #68: Energy Dashboard Configuration struggling with missing last_reset

joelbassett, 2023-07-11:
Hi,
Love your add-on and really appreciate your hard work.
I have a Sungrow SG10RS - single phase system. No battery.

When I try and configure the built-in HA Energy Dashboard, it throws an error saying that the sensor's are configured as state_class 'measurement' but are missing 'last_reset'

Any idea on how to fix this?

[image]

helikopter1909, 2023-07-18:
I created my own sensor. Do need an example?

joelbassett, 2023-07-22:
> I've got the same problem, it seems like GoSungrow is using state classes measurement/total on sensors, which are resetting to zero at midnight. According to the [HA documentation](https://developers.home-assistant.io/docs/core/entity/sensor/#available-state-classes) total_increasing would be the correct state class instead.
> 
> Anyway, I added template sensors for all the required entities just like @helikopter1909. Here is my configuration for grid, PV and battery entities if somebody needs it. Just insert your ps_key and register the template entities in your energy dashboard settings.
> 
> ```yaml
> template:
>   - sensor:
>       - name: Grid to Load Energy (Energy Dashboard)
>         state: >
>           {{ float(states('sensor.gosungrow_virtual_XXXXXXXXXXXXXX_grid_to_load_energy')) }}
>         unit_of_measurement: kWh
>         icon: mdi:transmission-tower
>         device_class: energy
>         state_class: total_increasing
>       - name: PV to Grid Energy (Energy Dashboard)
>         state: >
>           {{ float(states('sensor.gosungrow_virtual_XXXXXXXXXXXXXX_pv_to_grid_energy')) }}
>         unit_of_measurement: kWh
>         icon: mdi:transmission-tower
>         device_class: energy
>         state_class: total_increasing
>       - name: PV Daily Energy (Energy Dashboard)
>         state: >
>           {{ float(states('sensor.gosungrow_virtual_XXXXXXXXXXXXXX_pv_daily_energy')) }}
>         unit_of_measurement: kWh
>         icon: mdi:transmission-tower
>         device_class: energy
>         state_class: total_increasing
>       - name: Battery Charge Energy (Energy Dashboard)
>         state: >
>           {{ float(states('sensor.gosungrow_virtual_XXXXXXXXXXXXXX_battery_charge_energy')) }}
>         unit_of_measurement: kWh
>         icon: mdi:transmission-tower
>         device_class: energy
>         state_class: total_increasing
>       - name: Battery Discharge Energy (Energy Dashboard)
>         state: >
>           {{ float(states('sensor.gosungrow_virtual_XXXXXXXXXXXXXX_battery_discharge_energy')) }}
>         unit_of_measurement: kWh
>         icon: mdi:transmission-tower
>         device_class: energy
>         state_class: total_increasing
> ```

This looks like it might be the fix.

However my biggest issue is that I don't have any entities with the sensibly named endings like your grid_to_load_energy or pv_to_grid_energy so I don't know what information your variables are presenting.

Are they Daily totals in each, or are the running totals (ever increasing) in each of those sensors that you've used to create the total_increasing sensor?

joelbassett, 2023-07-27:
> @joelbassett
> 
> The sensors that you have to choose for the energy dashboard are listed here: https://github.com/MickMake/HomeAssistantAddons/blob/main/GoSungrow/DOCS.md (all the way at the bottom)
> 
> I'm relatively new to Sungrow, but it looks like they all reset to zero at midnight (although the names don't have "Daily" in them like some other sensors do). But those resets wouldn't be a problem at all if the correct device class total_increasing is used, because Home Assistant then treats sensor drops as meter resets and doesn't count them for statistics. So just creating template sensors that copy the value but have a fixed device_class did work for me.
> 
> The only thing a had to change was the sensor for PV energy. "PV Energy" seems to count energy twice sometimes (when it is charged to and consumed from the battery), insted I use the sensor "PV Daily Energy" that looks like to count as I want.
> 
> When setting up GoSungrow, I got a ton of sensors and it has been difficult to find what I wanted at first (especially because the same sensors exist multiple times with different names). Nothing is documented because Sungrow is proprietary. But make sure you havn't disabled any sensors in HA, then you should able to find the right ones using the sensor id. Maybe try to search for "sensor.gosungrow_virtual" or "grid_to_load_energy".

Thanks for trying. The issue is that my inverter is not reporting any of those listed sensor names.
Best example I can give is that my sensor for Self-Consumption of PV is called sensor.gosungrow_getpsdetail_xxxxxx_zfzy_map
So unfortunately its not quite as intuitive as picking the sensor that has the correct name, as none do.

I'm moreso chasing what sensors I should be using to create the faux sensors for the energy dashboard.... ie can I use the sensor that reports the daily pv yield or do I have to find the one that is reporting the all time pv yield.

keplusplus, 2023-07-27:
As I said, if your template sensor uses the state class total_increasing Home Assistant will ignore any resets. So daily PV yield should work as well.

copadewe, 2023-09-06:
ALso noticed, that the sensors 
"The sensors that you have to choose for the energy dashboard are listed here: https://github.com/MickMake/HomeAssistantAddons/blob/main/GoSungrow/DOCS.md (all the way at the bottom)"
are not quit well working. Some of them are 4-5-6 times in it and you can´t read the end of the sensor name.
Also it is not correct, today f.e. with "sensor.gosungrow_virtual_xxxxxx_pv_energy" there is a production of MINUS !!!
i will try the other integration the next days....

kooori, 2024-01-31:
I have the same issue, what did you do to eventually resolve it?

GitPat2, 2024-04-13:
@joelbassett same here, did you find a solution?

VikOlliver, 2024-05-05:
I too have this issue. I have names like "Daily Yield of Plant" multiple times, and the last_reset error when I try to use them in the Energy tab. I'll try to attach a file but GitHub is stubborn in that department...
[image]

VikOlliver, 2024-05-09:
This is the error that comes up when I select Daily Yield of Plant above
[image]
.

bamcauliffe, 2025-04-19:
VikOlliver,
Per what keplusplus or joelbassett have shown above, if you change the entities “state_class” to “total_increasing”, this will resolve the “last reset is missing error”.

You can the entity details in
Developer tools
States
The find your entity “sensor.gosungrow_virtual_devices_1345812_p83022”.

To change the “state_class” this can be done I the configuration.yaml file, see what keplusplus or joelbassett have shown above.
They do it by adding a template sensor.
I have done it by modifying/customizing the entity details by adding to the configuration.yaml file

homeassistant:
  customize:
     sensor.gosungrow_virtual_devices_1345812_p83022:
       state_class: total_increasing