MickMake/GoSungrow #70: Found SunGrow 0 devices

haozgit, 2023-08-02:
Hi, im facing this issue and there is only limited sensor being detected. anyone facing the same issue ? thanks.
[image]
[image]

closed 2023-08-02 by haozgit

sicablondu, 2025-07-20:
i see this issue closed, but how was it solved?

sicablondu, 2025-07-20:
> i see this issue closed, but how was it solved?

apparently i had to user the Owner account , login the iSolarCloud, browse a bit through Live data and then it started to pull data