MickMake/GoSungrow #72: EU server login

rikardkjell, 2023-08-17:
[07:32:40] INFO: Login to iSolarCloud using gateway https://portaleu.isolarcloud.com/ ...
Error: API httpResponse is 405 Method Not Allowed

Above responce in log when trying to use the add-on. Is the host-adress correct for EU server login ?

ottovdv, 2023-08-17:
https://gateway.isolarcloud.eu is the correct one...

rikardkjell, 2023-08-17:
Phew. Thanx a lot. Worked like a charm. 😁

draoZA, 2023-09-01:
added pull request for updated README

skydado, 2023-12-10:
Thanks to @ottovdv's answer, I moved from _"Error: API httpResponse is 405 Method Not Allowed"_ to _"Error: appkey is incorrect 'er_invalid_appkey'"_ ... I did not change the default appkey provided with the addon ... is there a different appkey for the EU gateway?

ottovdv, 2023-12-10:
See https://github.com/MickMake/GoSungrow/issues/101

Paraphraser, 2023-12-10:
You can read the full background to this problem at #101.

If you just want the fix, see [this gist](https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37#encryptionhack) - the link takes you to the part of the gist that summarises what to do for Home Assistant.

If your need is more general (ie you want to use GoSungrow outside of Home Assistant) then start at the top of the gist and follow the instructions for recompiling.

If you have any more trouble with this specific issue (`er_invalid_appkey` or an error saying the content needs to be encrypted) then please add it to #101. In other words, it is **not** specific to the EU server.

xpufx, 2024-03-12:
I've gone through the gist and discussions as much as I can and am wondering why this patch is not included in the actual project ? Is it a license issue? Does the owner not want to incorporate an API key in the code?

Paraphraser, 2024-03-12:
See the last paragraph [here](https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37?permalink_comment_id=4975304#gistcomment-4975304). That's all pure speculation. Nobody knows. The totality of #101, this issue, and other issues, plus the gist, is the best we can do in the absence of actually hearing from the owner.

xpufx, 2024-03-12:
> See the last paragraph [here](https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37?permalink_comment_id=4975304#gistcomment-4975304). That's all pure speculation. Nobody knows. The [...]

So we CAN actually do this:   https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37?permalink_comment_id=4983908