MickMake/GoSungrow #74: JSON error with type of struct field ResultData.result_data.org_id

sungrow-firmware, 2023-08-30:
I'm getting the following error when I try to log in:
`ERROR: json: cannot unmarshal number in go struct field ResultData.result_data.org_id of type string`.

In the file iSolarCloud\AppService\login\data.go I had to change the type of the struct field OrgId from _string_ to _valueTypes.Integer_.
This solved the problem for me.

sungrow-firmware, 2023-08-30:
Here a PR for the change: #75

copadewe, 2023-08-30:
> Here a PR for the change: #75

Same here today morning. 
Sorry noob question, where to change it?

Coketrd, 2023-08-30:
> > Here a PR for the change: #75
> 
> Same here today morning. Sorry noob question, where to change it to "valueTypes.Integer"

yes same here. i cant find the file where i can change this.

sungrow-firmware, 2023-08-30:
Go to the file iSolarCloud\AppService\login\data.go, line 78
--> https://github.com/MickMake/GoSungrow/blob/db9e0bf6c6d124a8e7b18bdc6fa9391246c38cec/iSolarCloud/AppService/login/data.go#L78
There exchange _string_ to _valueTypes.Integer_, simmilar to line 81. 
Or you can use this file as a reference: https://github.com/sungrow-firmware/GoSungrow/blob/patch-1/iSolarCloud/AppService/login/data.go

gaetan-p, 2023-08-30:
I have the same problem and I am not sure where to do the change.
I can ssh to my home assistant box but from there I only have the "ha [command]" commands. I am not sure how / where to enter the addons (or container ?) to do actual changes. Help will be much appreciated : )

grf692, 2023-08-30:
I gitcloned the repo on my HA, edited the file to make the change described above.
Now how do I start the container correctly without breaking my HA and with the right config etc. please ?

Bartleby1980, 2023-08-30:
> > Go to the file iSolarCloud\AppService\login\data.go, line 78
> 
> Like Coketrd - where is this file :-)?

Die Datei ist unter der folgenden URL zu finden.
https://github.com/sungrow-firmware/GoSungrow/tree/master/iSolarCloud/AppService/login

Ich hab auch das gesamte Projekt heruntergeladen und nach Github-Desktop importiert.
Dann kann ich zwar ein neues Projekt mit der geänderten Datei erstellen, bekomme das in HomeAssistant aber nicht ersetzt.

TaxmanNo1, 2023-08-30:
@MickMake 
maybe here´s some work for you here...

oleg-d, 2023-08-30:
@MickMake please merge #75

jrbconsulting, 2023-09-01:
> I have the same problem and I am not sure where to do the change. I can ssh to my home assistant box but from there I only have the "ha [command]" commands. I am not sure how / where to enter the addons (or container ?) to do actual changes. Help will be much appreciated : )

Try 'login' when you get to that point. Should drop you to a shell

copadewe, 2023-09-01:
> Try 'login' when you get to that point. Should drop you to a shell

and than?
[image]

grf692, 2023-09-01:
Yes, I managed to get login already but it does not help much because then I would have to somehow modify what's inside the addons, that is the docker images or containers and I don't know how to do or if it's even possible : )

[image]

draoZA, 2023-09-01:
Same error here. I'm happy to update the source. But I'm not familiar with Go. So not sure how I compile again. Or run from source?

Bartleby1980, 2023-09-01:
Mick Make apparently I and a few others can't make the change. Changing the file on the local PC is not a problem, but exchanging it with the installed version (3.0.3) in HomeAssistant is a problem.

So it would be great if a new version could be made to allow the change to be made in HomeAssistant when there is an addon update.

Thank you.

jrbconsulting, 2023-09-01:
> and than? [image]

Looks like the answer might be 'root' with no password - https://community.home-assistant.io/t/default-password-for-the-pi-account/23432

copadewe, 2023-09-02:
> Mick Make apparently I and a few others can't make the change. Changing the file on the local PC is not a problem, but exchanging it with the installed version (3.0.3) in HomeAssistant is a [...]

Think  we need to wait until it is fixed in a new version :-///

copadewe, 2023-09-02:
> 

no is not working

copadewe, 2023-09-02:
> > 
> Looks like the answer might be 'root' with no password - https://community.home-assistant.io/t/default-password-for-the-pi-account/23432
>
 no is not working

Paraphraser, 2023-09-02:
OK everyone. I believe I have found a workaround.

Please see [this gist](https://gist.github.com/Paraphraser/bf574d8c804c381816e02e0a8fcfbdad).

I'd appreciate feedback.

wermerb, 2023-09-02:
> OK everyone. I believe I have found a workaround.
> 
> Please see [this gist](https://gist.github.com/Paraphraser/bf574d8c804c381816e02e0a8fcfbdad).
> 
> I'd appreciate feedback.

Hi @Paraphraser,

Much appreciate your gist its very well put together.
I've just went over it and it works like a charm. I had only one issue, I had to add one more line to the docker file:

```docker
FROM gosungrow:base
USER root
COPY GoSungrow /usr/local/bin/GoSungrow
RUN chmod +x /usr/local/bin/GoSungrow
```
Otherwise i got a permission denied.

Thanks again for your gist saved me a lot of trouble :).

grf692, 2023-09-02:
Not sure if I am doing something wrong (probably) but it does not work for me:

[image]

Paraphraser, 2023-09-03:
@wermerb Thanks for the feedback. I've updated the gist to include a `chmod 755` step (setting it in the context directory rather than in the Dockerfile - but that's just my personal preference).

copadewe, 2023-09-03:
Thx for this, but not work, stopped for me at point 3.
I will and need to wait for an update of the original.
BUT THANK YOU for your kind work!!!

TaxmanNo1, 2023-09-04:
New Update 3.0.7. ist out.

The Add-on works again.

Thank you @MickMake

MickMake, 2023-09-06:
> Here a PR for the change: #75

Thanks mate. Yup, Sungrow have changed things yet again.

I might make these strings that keep flip-flopping between integers and strings more generic.

closed 2023-10-02 by sungrow-firmware