MickMake/GoSungrow #78: JSON error on HA

brockgoodall, 2023-09-01:
I get the following error when starting the service gosungrow add on in HA:
INFO: Login to iSolarCloud using gateway https://augateway.isolarcloud.com ...
Error: json: cannot unmarshal number into Go struct field ResultData.result_data.org_id of type string

Krikilion, 2023-09-02:
Got same error here. What am I doing wrong?

[06:30:16] INFO: Login to iSolarCloud using gateway https://gateway.isolarcloud.eu ...
Error: json: cannot unmarshal number into Go struct field ResultData.result_data.org_id of type string
Usage:
  GoSungrow api login [flags]
Examples:
	GoSungrow api login  
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: json: cannot unmarshal number into Go struct field ResultData.result_data.org_id of type string
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

jamiellie, 2023-09-02:
Smae error here
[07:31:30] INFO: Login to iSolarCloud using gateway https://augateway.isolarcloud.com ...
Error: json: cannot unmarshal number into Go struct field ResultData.result_data.org_id of type string

Everything worked pefectly until this morning. Looks like Sungrow updated the app with some breaking changes.

Paraphraser, 2023-09-02:
OK everyone. I believe I have found a workaround.

Please see [this gist](https://gist.github.com/Paraphraser/bf574d8c804c381816e02e0a8fcfbdad).

I'd appreciate feedback.

ericedberg, 2023-09-03:
Same for me.

metawops, 2023-09-03:
Same here. I think @MickMake needs to release a new version that fixes this problem. Until then, we're doomed (or we try and use the above mentioned workaround hack).

lupus78, 2023-09-03:
Same here. Can't wait for the new version.

jamiellie, 2023-09-04:
Just installed the MickMade update. Everything seems to be fine.
Thanks for all your time and effort. I really apreciate the help.

metawops, 2023-09-04:
Yes, everything works fine again! 👍👍 ❤️ 🙏

mroxso, 2023-09-04:
New version works for me as well ⚡🎉

lupus78, 2023-09-05:
> How would I install the fix?

just install the latest version.
https://github.com/MickMake/GoSungrow/releases

monojk, 2023-09-06:
I tried different approaches and all failes, like:
```
go install github.com/MickMake/GoSungrow@v3.0.7
go: github.com/MickMake/GoSungrow@v3.0.7: github.com/MickMake/GoSungrow@v3.0.7: invalid version: module contains a go.mod file, so module path must match major version ("github.com/MickMake/GoSungrow/v3")
```

metawops, 2023-09-06:
> I tried different approaches and all fails, like:
> 
> ```
> go install github.com/MickMake/GoSungrow@v3.0.7
> go: github.com/MickMake/GoSungrow@v3.0.7: github.com/MickMake/GoSungrow@v3.0.7: invalid version: module contains a go.mod file, so module path must match major version ("github.com/MickMake/GoSungrow/v3")
> ```

I just clicked on the "Install Update" button in Home Assistant (in Settings > Add-ons) ... 🤷‍♂️ No need to fiddle with the commandline.

monojk, 2023-09-06:
I use it with commandline, I don't use Home Assistant

monojk, 2023-09-08:
Ok, so installation does not work with go install, but from https://github.com/MickMake/GoSungrow/releases the executable GoSungrow can be downloaded.