MickMake/GoSungrow #80: Mqtt

khkissel, 2023-09-04:
How to use mqtt directly without a HA Installation?

Paraphraser, 2023-09-05:
If you:

1. Download the relevant [binary](https://github.com/MickMake/GoSungrow/releases/tag/v3.0.7) for your architecture and stick it in your PATH;
2. Do the [Config and login](https://github.com/MickMake/GoSungrow#config-and-login) steps from the README;
3. Run as many of these commands as you need:

	```
	$ GoSungrow config write --mqtt-host=«HOST»
	$ GoSungrow config write --mqtt-port=«PORT»
	$ GoSungrow config write --mqtt-user=«USER»
	$ GoSungrow config write --mqtt-passwprd=«PASSWORD»
	```
	
	such that:
	
	* `«HOST»` is the fully-qualified domain name, multicast domain name, host name or IP address of the host running your MQTT broker
	* `«PORT»` is 1883 or whatever you have changed it to
	* `«USER»` is a valid username known to your MQTT broker (if you don't have credential checking enabled on your broker, just don't run the command); and
	* `«PASSWORD»` pairs with `«USER»` (same comment about not running this if you don't need to set a password);

4. Then just running the following command gets the job done:

	```
	$ GoSungrow mqtt run
	```

You can fire that off by hand, or stick it in your `crontab` as an `@reboot`, or build a container and let Docker run it for you.

Hope that helps.

khkissel, 2023-09-05:
Thanks a lot for your detailed desription. I've done this before and wasn't able to connect to my mosquitto-broker. IP, port, mosuitto-userid and password are ok.  But after starting the connection, I've got the following messages:

./GoSungrow mqtt run
2023/09/05 21:21:39 Unknown log level, setting to default.
2023/09/05 21:21:39 INFO: Connecting to MQTT HASSIO Service...
2023/09/05 21:21:39 INFO: Connecting to SunGrow...
2023/09/05 21:21:39 INFO: Found SunGrow 2 devices
Error: network Error : dial tcp: lookup XXX on 192.168.1.1:53: no such host

XXX ist the mqtt-userid on the mosquitto-broker. There is no host XXX.

| --mqtt-user          |            | GOSUNGROW_MQTT_USER           | HASSIO: mqtt username.         | XXX                                |
| --mqtt-password |            | GOSUNGROW_MQTT_PASSWORD | HASSIO: mqtt password.         | ***********                      |
| --mqtt-host         |            | GOSUNGROW_MQTT_HOST           | HASSIO: mqtt host.                 | 192.168.1.22                   |
| --mqtt-port         |            | GOSUNGROW_MQTT_PORT            | HASSIO: mqtt port.                 |                               8883 |

khkissel, 2023-09-10:
Thanks a lot for your detailed description. But it seem that there is another problem. All the userids, passwords, mqtt-host ip-address and port are correctly set as you can see below:
`
Using config file '/home/pi42t/.GoSungrow/config.json'
+-----------------+------------+-------------------------+--------------------------------+------------------------------------+
|      FLAG       | SHORT FLAG |       ENVIRONMENT       |          DESCRIPTION           |        VALUE (* = DEFAULT)         |
+-----------------+------------+-------------------------+--------------------------------+------------------------------------+
| --config        |            | GOSUNGROW_CONFIG        | GoSungrow: config file.        | /home/pi42t/.GoSungrow/config.json |
| --debug         |            | GOSUNGROW_DEBUG         | GoSungrow: Debug mode.         | false *                            |
| --quiet         |            | GOSUNGROW_QUIET         | GoSungrow: Silence all         | false *                            |
|                 |            |                         | messages.                      |                                    |
| --timeout       |            | GOSUNGROW_TIMEOUT       | Web timeout.                   | 30s *                              |
| --user          | -u         | GOSUNGROW_USER          | SunGrow: api username.         | *********       |
| --password      | -p         | GOSUNGROW_PASSWORD      | SunGrow: api password.         | *******                     |
| --appkey        |            | GOSUNGROW_APPKEY        | SunGrow: api application key.  | 93D72E60331ABDCDC7B39ADC2D1F32B3   |
|                 |            |                         |                                | *                                  |
| --host          |            | GOSUNGROW_HOST          | SunGrow: Provider API URL.     | https://gateway.isolarcloud.eu     |
| --token-expiry  |            | GOSUNGROW_TOKEN_EXPIRY  | SunGrow: last login.           | 2023-09-10T18:05:22                |
| --save          | -s         | GOSUNGROW_SAVE          | Save output as a file.         | false *                            |
| --dir           |            | GOSUNGROW_DIR           | Save output base directory.    |  *                                 |
| --mqtt-user     |            | GOSUNGROW_MQTT_USER     | HASSIO: mqtt username.         | XXX                                |
| --mqtt-password |            | GOSUNGROW_MQTT_PASSWORD | HASSIO: mqtt password.         | *********                      |
| --mqtt-host     |            | GOSUNGROW_MQTT_HOST     | HASSIO: mqtt host.             | 192.168.1.22                       |
| --mqtt-port     |            | GOSUNGROW_MQTT_PORT     | HASSIO: mqtt port.             |                               8883 |
+-----------------+------------+-------------------------+--------------------------------+------------------------------------+
`
The api login is working well:
`./GoSungrow api login
2023/09/10 12:05:21 Unknown log level, setting to default.
Email:  ***@*****
Create Date:    Mon Aug 28 16:42:24 CST 2023
Login Last Date:        2023-09-10 18:05:22
Login Last IP:
Login State:    1
User Account:   ******
User Id:        *****
User Name:      ******
Is Online:      false
Token:  **********
Token File:     /home/pi42t/.GoSungrow/AppService_login.json
`
Starting the mqtt gives the following messages:
`./GoSungrow mqtt run
2023/09/10 12:24:39 Unknown log level, setting to default.
2023/09/10 12:24:40 INFO: Connecting to MQTT HASSIO Service...
2023/09/10 12:24:40 INFO: Connecting to SunGrow...
2023/09/10 12:24:40 INFO: Found SunGrow 2 devices
Error: network Error : dial tcp: lookup XXX on 192.168.1.1:53: no such host
Usage:
  GoSungrow mqtt run [flags]

Aliases:
  run,

Examples:
        GoSungrow mqtt run

Flags: Use "GoSungrow help flags" for more info.

Additional help topics:

ERROR: network Error : dial tcp: lookup XXX on 192.168.1.1:53: no such host

`
I don`t know why GoSungrow is taking the MQTT_USER for the network lookup. And btw I can`t see any message inside mosquitto.

Paraphraser, 2023-09-10:
I really have no idea but here are some possibilities.

First suggestion. Your config output is a little bit difficult to read so I'm not 100% sure about what I'm seeing (if you do this again, perhaps try putting the output between lines of triple back-ticks - that renders the lines in between "as is"). Anyway, this is what I see from mine:

```
+-------------------+------------+---------------------------+--------------------------------+---------------------------------------+
|       FLAG        | SHORT FLAG |        ENVIRONMENT        |          DESCRIPTION           |          VALUE (* = DEFAULT)          |
+-------------------+------------+---------------------------+--------------------------------+---------------------------------------+
| --config          |            | GOSUNGROW_CONFIG          | GoSungrow: config file.        | /Users/pkflint/.GoSungrow/config.json |
| --debug           |            | GOSUNGROW_DEBUG           | GoSungrow: Debug mode.         | false *                               |
| --quiet           |            | GOSUNGROW_QUIET           | GoSungrow: Silence all         | false *                               |
|                   |            |                           | messages.                      |                                       |
| --timeout         |            | GOSUNGROW_TIMEOUT         | Web timeout.                   | 30s *                                 |
| --user            | -u         | GOSUNGROW_USER            | SunGrow: api username.         | someone@domain.com                    |
| --password        | -p         | GOSUNGROW_PASSWORD        | SunGrow: api password.         | password                              |
| --appkey          |            | GOSUNGROW_APPKEY          | SunGrow: api application key.  | 40E66190FD0B4526A3B87B605F53A18C      |
|                   |            |                           |                                | *                                     |
| --host            |            | GOSUNGROW_HOST            | SunGrow: Provider API URL.     | https://augateway.isolarcloud.com     |
|                   |            |                           |                                | *                                     |
| --token-expiry    |            | GOSUNGROW_TOKEN_EXPIRY    | SunGrow: last login.           | 2023-09-06T16:24:29                   |
| --save            | -s         | GOSUNGROW_SAVE            | Save output as a file.         | false *                               |
| --dir             |            | GOSUNGROW_DIR             | Save output base directory.    |  *                                    |
| --mqtt-user       |            | GOSUNGROW_MQTT_USER       | HASSIO: mqtt username.         |  *                                    |
| --mqtt-password   |            | GOSUNGROW_MQTT_PASSWORD   | HASSIO: mqtt password.         |  *                                    |
| --mqtt-host       |            | GOSUNGROW_MQTT_HOST       | HASSIO: mqtt host.             | sec-dev.my.domain.com                 |
| --mqtt-port       |            | GOSUNGROW_MQTT_PORT       | HASSIO: mqtt port.             |  *                                    |
| --modbus-user     |            | GOSUNGROW_MODBUS_USER     | Modbus username.               |  *                                    |
| --modbus-password |            | GOSUNGROW_MODBUS_PASSWORD | Modbus password.               |  *                                    |
| --modbus-host     |            | GOSUNGROW_MODBUS_HOST     | Modbus host.                   |  *                                    |
| --modbus-port     |            | GOSUNGROW_MODBUS_PORT     | Modbus port.                   | 502 *                                 |
+-------------------+------------+---------------------------+--------------------------------+---------------------------------------+
```

It looks to me like your output ends at the `--mqtt-port` line while mine has four `--modbus` lines. The config file is just JSON so I can't see why it would make any difference but, just maybe, yours is either mal-formed in some way, or from an earlier version, or there's some other logical reason which would explain this formatting difference which might also be implicated in XXX turning up in the wrong place.

Perhaps eliminate all these possibilities by starting from a clean slate:

```
$ cd
$ mv .GoSungrow .GoSungrow.off
$ GoSungrow config write --user=USERNAME --password=PASSWORD
$ GoSungrow api login
$ GoSungrow config write --mqtt-host=192.168.1.22
```

and, if you really are using port 8883 (rather than 1883) then also:

```
$ GoSungrow config write --mqtt-port=8883
```

and then try to fire it up:

```
$ GoSungrow mqtt run
```

Second suggestion. Perhaps read #83. The test script in there would be a good basis for proving to yourself that there really is an MQTT broker on 192.168.1.22 (at either port 1883 or 8883, as appropriate).

As you'll see when you read through that issue, I had a lot of trouble getting multicast domain names to work but no trouble at all with either IP addresses or fully-qualified domain names.

Third suggestion. Truth to tell, I don't actually use RFC1918 addressing on my home network. Whenever you see a 192.168.203.x address in anything I write, it's a sanitised address in the same way as I sanitise usernames, passwords, tokens, and so on. So, I have not *actually* tested any IP address from any RFC1918 range.

[... cut ...]
Fourth suggestion. Another thing I haven't tried but which I'm surmising from the output of `GoSungrow config` is the "ENVIRONMENT" column. I read that as saying that I could override the whatever is in the JSON config using environment variables. For example:

```
$ export GOSUNGROW_MQTT_HOST=XXX
$ GoSungrow mqtt run
```

would override the 192.168.1.22 with XXX

I'm pretty sure you'd remember having done this but it still might be a good idea to double-check that you don't have any conflicting environment variables. Something like:

```
$ env | grep GOSUNGROW
```

If none of those suggestions get you anywhere then &#x1F937;

Paraphraser, 2023-09-10:
Continuing...

If none of those suggestions gets you anywhere, then it might be time for you to explain your actual setup in more detail.

The kinds of questions you should answer are:

1. What, exactly, is your hardware?
2. What operating systems are you running on your client (where you want to run GoSungrow) and server (where you want to run Mosquitto).
3. Is Mosquitto installed on the server natively or is it running in a Docker container?
4. Is your home network a single subnet (eg 192.168.1.0/24)?
5. Can your client computer ping your server computer (and vice versa)?
6. Are you able to set up a `mosuitto_sub` and have it echo a message sent with `mosquitto_pub` (as in the test script example)?

and then add anything else you think might be at all relevant.

khkissel, 2023-09-12:
I'd tried all all your suggested solutions. But no success. At the end I've digged out the problem and got it working. The problem is the mqtt-password. My password contains multiple hashes. Something must be wrong when processing the password. Apparently hashes are not allowed. Unfortunately, the resulting error message is misleading. Everything works fine using a password without hashes. I tested as follows: 

```./GoSungrow config write --mqtt-user=GoSungrow1
./GoSungrow config write --mqtt-password=#Go#Sungrow1-8883
Using config file '/home/pi42t/.GoSungrow/config.json'
New config:
+-------------------+------------+---------------------------+--------------------------------+------------------------------------+
|       FLAG        | SHORT FLAG |        ENVIRONMENT        |          DESCRIPTION           |        VALUE (* = DEFAULT)         |
+-------------------+------------+---------------------------+--------------------------------+------------------------------------+
| --config          |            | GOSUNGROW_CONFIG          | GoSungrow: config file.        | /home/pi42t/.GoSungrow/config.json |
| --debug           |            | GOSUNGROW_DEBUG           | GoSungrow: Debug mode.         | false *                            |
| --quiet           |            | GOSUNGROW_QUIET           | GoSungrow: Silence all         | false *                            |
|                   |            |                           | messages.                      |                                    |
| --timeout         |            | GOSUNGROW_TIMEOUT         | Web timeout.                   | 30s *                              |
| --user            | -u         | GOSUNGROW_USER            | SunGrow: api username.         | *********@t-online.de              |
| --password        | -p         | GOSUNGROW_PASSWORD        | SunGrow: api password.         | *************                      |
| --appkey          |            | GOSUNGROW_APPKEY          | SunGrow: api application key.  | 93D72E60331ABDCDC7B39ADC2D1F32B3   |
|                   |            |                           |                                | *                                  |
| --host            |            | GOSUNGROW_HOST            | SunGrow: Provider API URL.     | https://gateway.isolarcloud.eu     |
| --token-expiry    |            | GOSUNGROW_TOKEN_EXPIRY    | SunGrow: last login.           | 2023-09-13T01:51:10                |
| --save            | -s         | GOSUNGROW_SAVE            | Save output as a file.         | false *                            |
| --dir             |            | GOSUNGROW_DIR             | Save output base directory.    |  *                                 |
| --mqtt-user       |            | GOSUNGROW_MQTT_USER       | HASSIO: mqtt username.         | GoSungrow1                         |
| --mqtt-password   |            | GOSUNGROW_MQTT_PASSWORD   | HASSIO: mqtt password.         | #Go#Sungrow1-8883                  |
| --mqtt-host       |            | GOSUNGROW_MQTT_HOST       | HASSIO: mqtt host.             | 192.168.1.22                       |
| --mqtt-port       |            | GOSUNGROW_MQTT_PORT       | HASSIO: mqtt port.             |                               8883 |
| --modbus-user     |            | GOSUNGROW_MODBUS_USER     | Modbus username.               |  *                                 |
| --modbus-password |            | GOSUNGROW_MODBUS_PASSWORD | Modbus password.               |  *                                 |
| --modbus-host     |            | GOSUNGROW_MODBUS_HOST     | Modbus host.                   |  *                                 |
| --modbus-port     |            | GOSUNGROW_MODBUS_PORT     | Modbus port.                   | 502 *                              |
+-------------------+------------+---------------------------+--------------------------------+------------------------------------+

./GoSungrow mqtt run
ERROR: network Error : dial tcp: lookup GoSungrow1 on 192.168.1.1:53: no such host
```
```./GoSungrow config write --mqtt-password=Go#Sungrow1-8883
Using config file '/home/pi42t/.GoSungrow/config.json'
New config:
+-------------------+------------+---------------------------+--------------------------------+------------------------------------+
|       FLAG        | SHORT FLAG |        ENVIRONMENT        |          DESCRIPTION           |        VALUE (* = DEFAULT)         |
+-------------------+------------+---------------------------+--------------------------------+------------------------------------+
| --config          |            | GOSUNGROW_CONFIG          | GoSungrow: config file.        | /home/pi42t/.GoSungrow/config.json |
| --debug           |            | GOSUNGROW_DEBUG           | GoSungrow: Debug mode.         | false *                            |
| --quiet           |            | GOSUNGROW_QUIET           | GoSungrow: Silence all         | false *                            |
|                   |            |                           | messages.                      |                                    |
| --timeout         |            | GOSUNGROW_TIMEOUT         | Web timeout.                   | 30s *                              |
| --user            | -u         | GOSUNGROW_USER            | SunGrow: api username.         | *********@t-online.de              |
| --password        | -p         | GOSUNGROW_PASSWORD        | SunGrow: api password.         | *************                      |
[... cut ...]
CUCU?CUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCU?CUCU?CUCUCUCUCUCUCUCUCUC...
2023/09/12 22:08:21 INFO: Syncing 1568 entries with HASSIO from queryDeviceList.
CUCUCUCU-CUCUCU-CUCUCU--CUCU---CU-CU-CU---CUCUCUCUCU-CUCUCUCUCU-...
2023/09/12 22:08:21 INFO: Starting ticker...
2023/09/12 22:08:21 INFO: Fetch Schedule: 5m
2023/09/12 22:08:21 INFO: Sleep Delay:    40s
```
What does the many CU.... mean?

Should I open an issue about the password hash problem?

Thanks a lot for your excellent description and help.

Paraphraser, 2023-09-13:
I have no idea what the CU mean.

I agree that the "#" is a bit bizarre. When I try:

```
$ GoSungrow config write --mqtt-password=#Go#Sungrow1-8883
```

the config output shows the password is intact:

```
| --mqtt-password   |            | GOSUNGROW_MQTT_PASSWORD   | HASSIO: mqtt password.         | #Go#Sungrow1-8883                     |
```

I get the correct answer if I query with jq:

```
$ jq '."mqtt-password"' <~/.GoSungrow/config.json
"#Go#Sungrow1-8883"
```

That suggests hashes embedded in strings aren't a problem for JSON parsing, which implies it's internal to GoSungrow so I'd say, yes, this is something worth reporting as an issue - assuming this issue doesn't already serve that purpose.

khkissel, 2023-09-13:
Regarding to the password-problem I've opened the issue #85 [.](https://github.com/MickMake/GoSungrow/issues/85)
Again, thanks a lot for your help and the nice conversation. BTW, I wrote my first line of COBOL-code in early 1974.
Best regards from the middle of Germany to down-under.

closed 2023-09-13 by khkissel

Paraphraser, 2023-09-13:
Well, you've got me beat. If you don't count actual machine code programmed via panel switches into an SC/MP, my first line of code was Pascal. COBOL didn't feature on my radar until 1977 when I was given a listing that took over half a box of continuous stationery and told to add functionality. This was on CDC Cybers where the line numbers got embedded in the lower 18 bits of 30-bit NO-OP instructions so you could work back to the offending line in crash dumps. That COBOL program busted 262144 lines so the line counter wrapped. Ah, the good old days.