MickMake/GoSungrow #84: Complete newbie to Go - how to get started?

baljo, 2023-09-08:
Not sure how to get started, this is what I've done:
- cloned the repo on my Win10-machine
- installed Go and ensured it is working

When trying this in the Command Prompt: 
`./bin/GoSungrow config write --user=USERNAME --password=PASSWORD`
I get:
`'bin\GoSungrow' is not recognized as an internal or external command, operable program or batch file.`

The reason is probably that the only files in the GoSungrow\bin-folder are: docker.sh, release.sh, strip.sh

What am I missing here? Do I need to somehow compile/build something?

baljo, 2023-09-08:
Just shooting in the dark and tried with `go build`

Got this:

**C:\Users\thv\Dropbox\Github\GoSungrow>go build**
iSolarCloud\api\GoStruct\gojson\funcs.go:45:2: github.com/MickMake/GoUnify@v0.0.0-00010101000000-000000000000: replacement directory ../../GoUnify does not exist
iSolarCloud\api\file.go:9:2: github.com/MickMake/GoUnify@v0.0.0-00010101000000-000000000000: replacement directory ../../GoUnify does not exist
cmdHassio\config.go:6:2: github.com/MickMake/GoUnify@v0.0.0-00010101000000-000000000000: replacement directory ../../GoUnify does not exist
cmd\commands.go:6:2: github.com/MickMake/GoUnify@v0.0.0-00010101000000-000000000000: replacement directory ../../GoUnify does not exist
cmd\cmd_api.go:10:2: github.com/MickMake/GoUnify@v0.0.0-00010101000000-000000000000: replacement directory ../../GoUnify does not exist
cmd\cmd_api.go:11:2: github.com/MickMake/GoUnify@v0.0.0-00010101000000-000000000000: replacement directory ../../GoUnify does not exist
`

Inca1777, 2023-09-12:
I had the same issue today as well.

GoSunGrow "go build" depends on https://github.com/MickMake/GoUnify, so I downloaded that first.

Make sure you get the location right as it expects it "../../GoUnify"  (for which I've used a soft link to match my download directory) 

With the GoUnify dependency met, GoSungGrow built OK for me. 

As last step run "go install" - which given proper go setup should put the binary in $GOPATH/bin

closed 2023-09-17 by baljo

Paraphraser, 2023-11-29:
I've mentioned this on #87 but I'm also mentioning it here in case anyone else who is having trouble compiling GoSungrow reads this issue:

* Please try following [this gist](https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37) and see if it helps solve your problem