MickMake/GoSungrow #87: replacement directory ../../GoUnify does not exist

khkissel, 2023-09-16:
Tried to compile GoSungro, but got the following error:
```
iSolarCloud/api/GoStruct/gojson/funcs.go:45:2: github.com/MickMake/GoUnify@v0.0.0-00010101000000-000000000000: replacement directory ../../GoUnify does not exist
```
Downloading:
```
 git clone https://github.com/MickMake/GoSungrow.git
Klone nach 'GoSungrow'...
remote: Enumerating objects: 63505, done.
remote: Counting objects: 100% (3967/3967), done.
remote: Compressing objects: 100% (1562/1562), done.
remote: Total 63505 (delta 2534), reused 3791 (delta 2398), pack-reused 59538
Empfange Objekte: 100% (63505/63505), 33.28 MiB | 9.66 MiB/s, fertig.
Löse Unterschiede auf: 100% (37134/37134), fertig.
Aktualisiere Dateien: 100% (2514/2514), fertig.
```
Compiling:
```
go version
go version go1.21.0 linux/arm64
cd GoSungrow
go build
go: downloading github.com/go-co-op/gocron v1.33.1
go: downloading github.com/spf13/cobra v1.7.0
go: downloading github.com/spf13/viper v1.16.0
go: downloading github.com/eclipse/paho.mqtt.golang v1.4.3
go: downloading github.com/wcharczuk/go-chart/v2 v2.1.1
go: downloading github.com/google/uuid v1.3.1
go: downloading go.uber.org/atomic v1.9.0
go: downloading github.com/spf13/afero v1.9.5
go: downloading github.com/spf13/cast v1.5.1
go: downloading github.com/gorilla/websocket v1.5.0
go: downloading golang.org/x/net v0.14.0
go: downloading github.com/mattn/go-runewidth v0.0.15
go: downloading golang.org/x/sys v0.12.0
go: downloading github.com/subosito/gotenv v1.4.2
go: downloading github.com/magiconair/properties v1.8.7
go: downloading github.com/pelletier/go-toml/v2 v2.0.8
go: downloading golang.org/x/text v0.12.0
go: downloading golang.org/x/image v0.11.0
iSolarCloud/api/GoStruct/gojson/funcs.go:45:2: github.com/MickMake/GoUnify@v0.0.0-00010101000000-000000000000: replacement directory ../../GoUnify does not exist
iSolarCloud/api/file.go:9:2: github.com/MickMake/GoUnify@v0.0.0-00010101000000-000000000000: replacement directory ../../GoUnify does not exist
cmdHassio/config.go:6:2: github.com/MickMake/GoUnify@v0.0.0-00010101000000-000000000000: replacement directory ../../GoUnify does not exist
cmd/commands.go:6:2: github.com/MickMake/GoUnify@v0.0.0-00010101000000-000000000000: replacement directory ../../GoUnify does not exist
cmd/cmd_api.go:10:2: github.com/MickMake/GoUnify@v0.0.0-00010101000000-000000000000: replacement directory ../../GoUnify does not exist
cmd/cmd_api.go:11:2: github.com/MickMake/GoUnify@v0.0.0-00010101000000-000000000000: replacement directory ../../GoUnify does not exist
'''

baljo, 2023-09-16:
Have you downloaded GoUnify? Pls see https://github.com/MickMake/GoSungrow/issues/84#issuecomment-1715979300

khkissel, 2023-09-16:
That's it. Now the compilation is running without errors.

closed 2023-09-16 by khkissel

miettal, 2023-11-29:
@MickMake It seems build script is broken.

GoUnify package is directly specified by relative filepath in go.mod.
https://github.com/MickMake/GoSungrow/blob/master/go.mod#L5

this change is from this commit.
https://github.com/MickMake/GoSungrow/commit/c4b870b7c7de647b757d61295dd7d66d11f5bba4#

Paraphraser, 2023-11-29:
Hi @miettal - I'm pretty sure the build script is not broken. The documentation just doesn't make it all that clear how you need to create the conditions for a successful compile.
 
Please see if [this gist](https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37) helps.

I have just re-tested on several platforms (listed in the gist) and successfully compiled on all of them.

miettal, 2023-11-29:
Thank you quick response and sharing/making document gist.
I'll follow this document. It seems build is ok in my environment. very Thank you.

In other hands, I think hardcoding of path to outside repository has problem of portability.
It seems past version is not kind of this hardconding. do you have reason of this change?
(I'm not familiar with go building system. so I just interested in)

Paraphraser, 2023-11-29:
> do you have reason of this change?

No. I just looked at the error message showing the relative path construction and worked backwards to how the folder structure needed to be set up so the compile would succeed. I have no idea whether my structure is "right" in any absolute sense. The most I can say is "it works".

Kouda-Titanium, 2023-12-12:
Hi there,

I've realized an issue when installing GoSungrow in the root or home directory ( `/` or `~/` ). In such cases, the reference `../../GoUnify` becomes invalid and points to inaccessible or non-existent paths.

Looking forward to a resolution. Thanks!

Paraphraser, 2023-12-12:
Hi @Kouda-Titanium - please see [this gist](https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37). As the gist explains, you just have to set up the appropriate directory structure and then it will compile OK.