MickMake/GoSungrow #88: Strange sum values of previous day at 00:00 each day

rikardkjell, 2023-09-17:
A couple of days this addon worked just fine and I got the values as it should but today it freaked out adding the previous days numbers as a post at 00:00.... What is happening and how can I correct it ?
It has happend earlier but disappered when I read the instructions more carefully and added the correct sensors as input to the Energy Dashboard, as intructed...
[image]
[image]

copadewe, 2023-09-17:
Hi, have the same problem, at 0:00 is a huge stripe and don´t know why.

rikardkjell, 2023-09-19:
Solved (in a way) in community HASS community post ["[Strange values in Energy Dashboard](https://community.home-assistant.io/t/strange-values-in-energy-dashboard/615232)"](https://community.home-assistant.io/t/strange-values-in-energy-dashboard/615232/3)

copadewe, 2023-09-20:
> Solved (in a way) in community HASS community post ["[Strange values in Energy Dashboard](https://community.home-assistant.io/t/strange-values-in-energy-dashboard/615232)"]

Ok thank you, but i don´t understand what to do - any idea - any helping step by step guide?

rikardkjell, 2023-09-20:
"Developer tools → Statistics, click on the ramp icon next to your entity to edit the statistics."
Find the value that sticks out and zero it out...

r77s, 2023-10-26:
But this is only the solution for the one day when this happens, but not a permanent solution. Will there be a fix?

rikardkjell, 2023-10-26:
I solved the issue connecting the inverters ModBus-port to a WiFi-extender in the garage, where the inverter is located. My initial problem with that location was that I didn't wanna lay a LAN-cable all the way to the inverter. That issue a Extender solved. This also gave me snappier and more frequent updates of the actual values than going via the iSolarCloud login integration.
I use this instead https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant.

mazvone, 2024-07-23:
Has anyone found a permanently solution for this problem?