MickMake/GoSungrow #89: Add-on stops every few days

KLucky13, 2023-09-23:
Every few days I need to manually start this add-on again. It seems worse since the latest update and I'm not sure what is causing it. Everything is up to date and I posted the log below. It completely messed up the data every time it turns off since it creates a massive spike of solar production whenever I manually start up the add-on again. Very annoying issue and I would like to fix this. Let me know if more info is needed to troubleshoot this

[02:05:40] INFO: Login to iSolarCloud using gateway https://gateway.isolarcloud.eu ...
Error: Post "https://gateway.isolarcloud.eu/v1/userService/login": dial tcp 192.168.0.1:443: connect: connection refused
Usage:
GoSungrow api login [flags]
Examples:
GoSungrow api login
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: Post "https://gateway.isolarcloud.eu/v1/userService/login": dial tcp 192.168.0.1:443: connect: connection refused
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

[02:05:40] INFO: Login to iSolarCloud using gateway https://gateway.isolarcloud.eu ...
Error: Post "https://gateway.isolarcloud.eu/v1/userService/login": dial tcp 192.168.0.1:443: connect: connection refused
Usage:
GoSungrow api login [flags]
Examples:
GoSungrow api login
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: Post "https://gateway.isolarcloud.eu/v1/userService/login": dial tcp 192.168.0.1:443: connect: connection refused
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

KLucky13, 2023-10-01:
@MickMake Could you help? For now the add-on is completely unusable for me since it just keeps stopping on it's own.

KLucky13, 2023-10-18:
bump, yesterday it happened again... Still no fix for this?

KLucky13, 2023-10-18:
In the meantime I made an "ugly fix" which prevents the add-on from being inactive for longer periods. It basically just restarts it in case it was stopped (for unknown reasons)

alias: Restart GoSunGrow if Stopped
description: ""
trigger:
  - type: not_running
    platform: device
    device_id: 7a8dd90a84e17a390e6d66e36b7c05d6
    entity_id: 32e01c95af128148080233a7ca419962
    domain: binary_sensor
action:
  - service: hassio.addon_start
    data:
      addon: ba22da74_gosungrow

anthony-spruyt, 2023-11-22:
Seeing the same problem here. It stops randomly and doesnt restart even with the watchdog enabled.

KLucky13, 2023-11-22:
> Seeing the same problem here. It stops randomly and doesnt restart even with the watchdog enabled.

You could implement the workaround above, it has worked flawless for me so far. I also set up a notification every time it goes offline, by the time I check, the service was already rebooted and everything was already back up and running. So I'm going to disable the notification now and just trust on the implementation above

DerMahlus, 2025-07-11:
This automation provides a simple fallback mechanism for the GoSunGrow add-on. It triggers once every hour and attempts to start the add-on. If the add-on is already running, the command is ignored by Home Assistant without causing any issues. This is a lightweight and effective way to recover from occasional crashes or unexpected shutdowns without requiring additional sensors or complex logic.

alias: Restart GoSunGrow if Stopped
description: >-
  Automatically restarts the GoSunGrow add-on every hour to recover from
  occasional crashes or unexpected stops. If the add-on is already running, the
  start command is safely ignored.
triggers:
  - trigger: time_pattern
    hours: "1"
conditions: []
actions:
  - action: hassio.addon_start
    data:
      addon: ba22da74_gosungrow
mode: single