MickMake/GoSungrow #97: Seeing a load of exceptions in the logs such as "could not convert string to float"

oleg-d, 2023-11-08:
The add-on seems to be working great otherwise, just that I am seeing a bunch of exceptions in the logs.

ps_type_name ValueError: could not convert string to float: 'Residential PV'
ps_location ValueError: could not convert string to float: 'Finland'
ps_holder ValueError: invalid literal for int() with base 10: 'B2290923829'
ps_short_name ValueError: invalid literal for int() with base 10: 'B2290923829'
ps_name ValueError: could not convert string to float: 'Finland'
timezone ValueError: invalid literal for int() with base 10: 'GMT+2'

For example here's the full stack trace for the timezone from the log:

```
2023-11-08 08:08:15.968 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised when updating state of sensor.gosungrow_getpsdetail_5182053_timezone, topic: 'homeassistant/sensor/GoSungrow/GoSungrow-getPsDetail-5182053-timezone/state' with payload: b'{"value":"GMT+2"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 616, in state
    numerical_value = int(value)
                      ^^^^^^^^^^
ValueError: invalid literal for int() with base 10: 'GMT+2'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 619, in state
    numerical_value = float(value)
                      ^^^^^^^^^^^^
ValueError: could not convert string to float: 'GMT+2'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 305, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 743, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 843, in _async_write_ha_state
    state, attr = self._async_generate_attributes()
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 784, in _async_generate_attributes
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 749, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 623, in state
    raise ValueError(
ValueError: Sensor sensor.gosungrow_getpsdetail_5182053_timezone has device class 'None', state class 'measurement' unit 'None' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'GMT+2' (<class 'str'>)

```

GitPat2, 2024-05-25:
same here, did you find a solution?

jsanchezdelvillar, 2025-01-17:
Same here. When I start the integration I get many warnings and one error. I have an SG0.4RS inverter, the comunication module and I have pending the installation of the general meter:
``````
2025-01-17 12:17:27.746 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.gosungrow_getpsdetail_5694969_safe_start_date (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('timestamp') it is using; expected None; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-01-17 12:17:27.765 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.gosungrow_getpsdetail_5694969_actual_energy_9 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-01-17 12:17:27.796 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.gosungrow_getpsdetail_5694969_actual_energy_0 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-01-17 12:17:27.804 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.gosungrow_getpsdetail_5694969_actual_energy_7 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-01-17 12:17:27.810 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.gosungrow_getpsdetail_5694969_plan_energy_5 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-01-17 12:17:27.831 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.gosungrow_getpsdetail_5694969_plan_energy_9 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-01-17 12:17:27.875 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.gosungrow_getpsdetail_5694969_plan_energy_11 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-01-17 12:17:27.930 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.gosungrow_getpsdetail_5694969_data_last_update_time (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('timestamp') it is using; expected None; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-01-17 12:17:27.945 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.gosungrow_getpsdetail_5694969_actual_energy_1 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-01-17 12:17:27.995 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.gosungrow_getpsdetail_5694969_record_create_time (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('timestamp') it is [... cut ...] iguration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-01-17 12:17:31.746 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised while updating state of sensor.gosungrow_getpslist_devices_5694969_location, topic: 'homeassistant/sensor/GoSungrow/GoSungrow-getPsList-devices-5694969-location/state' with payload: b'{"value":"full address"}'
``````