MickMake/GoSungrow #98: iSolarCloud -> Account does not exist

plumpy80, 2023-11-20:
Hi!

Try to start, but it's stop always. In the log i found this: ( I have an iSolarCloud account, and it's working now on my phone, or on the web. I think it's try to login on the chineese server, but my account on the European servers)

[14:22:21] INFO: Login to iSolarCloud using gateway https://augateway.isolarcloud.com ...
Error: Account does not exist '账号不存在'
Usage:
  GoSungrow api login [flags]

Examples:
	GoSungrow api login  

Flags: Use "GoSungrow help flags" for more info.

Additional help topics:

ERROR: Account does not exist '账号不存在'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

irqnet, 2023-11-20:
same here, european account seems not to work

plumpy80, 2023-11-21:
https://augateway.isolarcloud.com/  is for Australian users only... "AU"gateway.... I tried to change it any european server, but doesn't work.....

irqnet, 2023-11-21:
> https://augateway.isolarcloud.com/  is for Australian users only... "AU"gateway.... I tried to change it any european server, but doesn't work.....

https://gateway.isolarcloud.eu

plumpy80, 2023-11-22:
> > https://augateway.isolarcloud.com/  is for Australian users only... "AU"gateway.... I tried to change it any european server, but doesn't work.....
> 
> https://gateway.isolarcloud.eu

Thanks! It's worked!

plumpy80, 2023-11-22:
Thanks! It's worked!

closed 2023-11-22 by plumpy80

oriolMTo, 2023-11-22:
Hi, i'm from europe too, trying to start with:  GoSungrow api login  , but I have this message  "Error: unknown error 'Missing parameter in request header: x-access-key'" 

any suggestion please?

thanks

plumpy80, 2023-11-22:
> Hi, i'm from europe too, trying to start with: GoSungrow api login , but I have this message "Error: unknown error 'Missing parameter in request header: x-access-key'"
> 
> any suggestion please?
> 
> thanks

Sorry, but I don't know.....  

Can you fill all requested information?  I think it's  something login trouble.
[image]

You need to change the sungrow host to the european server. sungrow_user and passwd is your iSolarCloud account, mqtt-suer and passwd is the login for your mqtt server.... 

That was all I did for me...

oriolMTo, 2023-11-22:
Hi,  I set the parameters directly to config.json file, with european server, user, pass and appkey, currently testing deployment without HA, just to know if can retrieve my data from isolarCloud. Golang build was ok, so I understand it is possible to work on windows computer.
thanks for any suggestion

Oriol.