MickMake/GoSungrow #103: Can't install Sungrow on RPi3B

tombernr, 2023-12-04:
Running HA on RPi3B, Core 2023.11.3 is installed. Can't install Sungrow -->
The command '/bin/ash -o pipefail -c chmod a+x /usr/local/bin/setup.sh && /usr/local/bin/setup.sh' returned a non-zero code: 2

Any suggestion/idea is appreciated.

NoCampersFluffy, 2024-01-07:
Same issue with RP2b

Core 2024.1.2
Supervisor 2023.12.0
Operating System 11.3
Frontend 20240104.0

From Core log
----------------
Logger: homeassistant.components.hassio
Source: components/hassio/websocket_api.py:138
Integration: Home Assistant Supervisor (documentation, issues)
First occurred: 9:12:40 PM (5 occurrences)
Last logged: 9:34:39 PM

Failed to to call /addons/3fa94738_gosungrow/install - The command '/bin/ash -o pipefail -c chmod a+x /usr/local/bin/setup.sh && /usr/local/bin/setup.sh' returned a non-zero code: 1

From supervisor log
----------------------
Step 22/36 : ARG GOSUNGROW_MQTT_PORT

 ---> Using cache

 ---> 483d8edd531c

Step 23/36 : ENV GOSUNGROW_MQTT_PORT ${SUNGROW_MQTT_PORT}

 ---> Using cache

 ---> b49f1978d4ad

Step 24/36 : ARG GOSUNGROW_MQTT_USER

 ---> Using cache

 ---> 9229e91294fc

Step 25/36 : ENV GOSUNGROW_MQTT_USER ${SUNGROW_MQTT_USER}

 ---> Using cache

 ---> 5cf64fcf4b48

Step 26/36 : ARG TZ

 ---> Using cache

 ---> dd02d070abb4

Step 27/36 : ENV TZ ${TZ}

 ---> Using cache

 ---> 45478987500c

Step 28/36 : COPY src/ /usr/local/bin/

 ---> Using cache

 ---> 5923434a78bd

Step 29/36 : RUN chmod a+x /usr/local/bin/setup.sh && /usr/local/bin/setup.sh

 ---> Running in 1798c09d1159

fetch https://dl-cdn.alpinelinux.org/alpine/v3.16/main/armv7/APKINDEX.tar.gz

fetch https://dl-cdn.alpinelinux.org/alpine/v3.16/community/armv7/APKINDEX.tar.gz

v3.16.8-44-g4759ec58415 [https://dl-cdn.alpinelinux.org/alpine/v3.16/main]
v3.16.8-43-g6fced584282 [https://dl-cdn.alpinelinux.org/alpine/v3.16/community]
OK: 16567 distinct packages available

fetch https://dl-cdn.alpinelinux.org/alpine/v3.16/main/armv7/APKINDEX.tar.gz

fetch https://dl-cdn.alpinelinux.org/alpine/v3.16/community/armv7/APKINDEX.tar.gz

(1/4) Installing libunistring (1.0-r0)

(2/4) Installing libidn2 (2.3.2-r2)

(3/4) Installing wget (1.21.3-r0)

(4/4) Installing gosungrow.persist (20240107.103437)

Executing busybox-1.35.0-r17.trigger

OK: 20 MiB in 46 packages

bashio::log.error Hardware not supported

Removing intermediate container 1798c09d1159

24-01-07 21:42:05 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
24-01-07 21:42:05 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
24-01-07 21:42:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
24-01-07 21:42:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
24-01-07 21:42:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
24-01-07 21:42:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
24-01-07 21:42:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
24-01-07 21:42:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
24-01-07 21:42:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
24-01-07 21:42:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
24-01-07 21:42:07 INFO (MainThread) [supervisor.resolution.check] System checks complete
24-01-07 21:42:07 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
24-01-07 21:42:08 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
24-01-07 21:42:08 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
24-01-07 21:42:08 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
24-01-07 21:43:16 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
24-01-07 22:13:16 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token

popdanne, 2024-02-08:
Same issue. can't install add-on fails during install with following error message

[image]