MickMake/GoSungrow #107: strconv.Atoi error

illuzn, 2024-02-06:
Using the patched gosungrow build with the new API and APP key.
I get the following error:
```
[04:37:56] INFO: Syncing data from gateway https://augateway.isolarcloud.com ...
2024/02/06 04:37:56 INFO: Connecting to MQTT HASSIO Service...
2024/02/06 04:37:56 INFO: Connecting to SunGrow...
2024/02/06 04:37:56 INFO: Found SunGrow 5 devices
2024/02/06 04:37:56 INFO: Caching Sungrow metadata...
2024/02/06 04:37:56 INFO: Cached 1015 Sungrow data points...
PsId: required
JSON request:	{"ps_id":1259467}

2024/02/06 04:37:57 ERROR: SetResponse: strconv.Atoi: parsing "1559776_1559777": invalid syntax
Error: SetResponse: strconv.Atoi: parsing "1559776_1559777": invalid syntax
```

I have no clue about Golang but it appears this gets called in only a few places:

* [iSolarCloud/api/GoStruct/valueTypes/uuid.go](https://github.com/MickMake/GoSungrow/blob/391253aaadd2cae9df32c95bec9e6b9bbf83f4d6/iSolarCloud/api/GoStruct/valueTypes/uuid.go#L93)
* [iSolarCloud/api/GoStruct/valueTypes/uv.go](https://github.com/MickMake/GoSungrow/blob/391253aaadd2cae9df32c95bec9e6b9bbf83f4d6/iSolarCloud/api/GoStruct/valueTypes/uv.go#L457)
* [iSolarCloud/api/GoStruct/valueTypes/point.go](https://github.com/MickMake/GoSungrow/blob/391253aaadd2cae9df32c95bec9e6b9bbf83f4d6/iSolarCloud/api/GoStruct/valueTypes/point.go#L251)
* [iSolarCloud/api/GoStruct/valueTypes/integers.go](https://github.com/MickMake/GoSungrow/blob/391253aaadd2cae9df32c95bec9e6b9bbf83f4d6/iSolarCloud/api/GoStruct/valueTypes/integers.go#L104)

Debug logging doesn't reveal anymore information. Not sure how to provide further information to diagnose this issue.

Paraphraser, 2024-02-06:
Well, on the face of it, the string "1559776_1559777" has been passed to the "convert string to integer" function. That function has (correctly) returned an error because a string containing an underscore doesn't represent a valid integer.

I think the real question is where is "1559776_1559777" coming from?

From your question it's not clear exactly how you are running GoSungrow. When you say "patched gosungrow build" do you mean you have just recompiled the binary as per part 1 of [this gist](https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37). Or do you mean you're running the pre-built Docker image (as per part 2 of the gist). The log messages don't look the same as I see in Home Assistant so I'm guessing you're not doing this in HA.

To put that last para another way, if you don't explain exactly *what* you are doing, you make it difficult for anyone else to try to replicate to see if they **do** hit the same problem (meaning it's something that would likely be able to be fixed by the community), or they **don't** hit the same problem (meaning it's more likely to be something peculiar to your system).

When I look at the container log in Home Assistant, I see:

```
[11:17:34] INFO: Syncing data from gateway https://augateway.isolarcloud.com ...
2024/02/06 11:17:34 INFO: Connecting to MQTT HASSIO Service...
2024/02/06 11:17:34 INFO: Connecting to SunGrow...
2024/02/06 11:17:34 INFO: Found SunGrow 3 devices
2024/02/06 11:17:35 INFO: Caching Sungrow metadata...
2024/02/06 11:17:35 INFO: Cached 958 Sungrow data points...
2024/02/06 11:17:35 INFO: Syncing 461 entries with HASSIO from queryDeviceList.
2024/02/06 11:17:35 INFO: Syncing 148 entries with HASSIO from getPsList.
--CUCUCUCUCUCU-CUCUCU-CU-CU--CU--CUCU-CUCUCU---CU-CU---CUCUCU
…
```

You'll note that after the "Cached" message, I get two "Syncing" messages followed by what I think of as the usual gibberish pattern with the CU stuff. Conversely, your log mentions `PsId: required` at that point.

What do you see when you run:

```
$ GoSungrow api login
$ GoSungrow show ps list
```

What's in the "Ps Id" column? Is it the integer "1259467" or is it that problem string?

If you only just followed Part 1 of the gist then perhaps also confirm the version of Go you are using. My standalone binary was compiled on Jan 22 with Go version 1.21.6. Does that match the compiler version you are using?

illuzn, 2024-02-08:
Sorry to be unclear. Was running the homeassistant addon with only Part 2 of that patch applied.

I'll try installing this stand alone then running those commands directly.

Paraphraser, 2024-02-08:
And sorry to be unclear myself. The point I was trying to make (and obviously failing) was that I wasn't seeing the problem with my HA instance. Therefore assuming your setup didn't differ materially from mine in some way that wasn't immediately obvious from your post, the underlying cause was unlikely to be found in HA or the (replacement) add-on, and more likely to be found in either your configuration (ie the bits in the HA UI where to tell GoSungrow your credentials etc) or something peculiar to your inverter (ie the bits you configure using the iSolarCloud app) or your inverter's relationship with the Sungrow cloud. I think the latter will become evident with those GoSungrow commands.

What's most puzzling to me is the `PsId required` because, as far as I can tell, the  discovery of that is automatic. The fact that mine doesn't show that line while yours does was either going to be explained by you not actually running HA (now eliminated) or perhaps bad credentials. But, assuming you can login correctly then auto discovery of the PsId should occur (ie like mine) rather than `PsId required` (like yours). And how that then gets us to malformed integers...

All very weird.

illuzn, 2024-02-08:
Does it have to do with me having 2 sites connected to my account work and home?

Paraphraser, 2024-02-08:
If you mean that you have inverters at home and work, and they are both connected to the same Sungrow account, then I have no idea. I only have the one Sungrow inverter.

If you mean that you have HA running at both home and work, connected to the same Sungrow account and pulling data associated with your inverter(s) then I can't see why that would make any difference. As far as I can tell, the iSolarCloud app, HA, and non-HA invocations of GoSungrow are all doing the same thing in API terms. I'm using the iSolarCloud app (mainly to view "now" data), the HA version (for nothing other than to prove it's working) plus a cron job firing once a day to call GoSungrow standalone to download data for "yesterday" which I load into a database. I also run standalone GoSungrow when trying to answer questions here on GitHub. So, by my count, I have four totally independent usages of the API and I don't see anything like you're seeing.

copetogo, 2024-02-09:
@Paraphraser  i second the issue raised here. I don't think 2 accounts is an issue because i had 2 inverters in isolar account but it anyways used to show 1 account as the other account was a shared account from another email ID . I am facing the same issue at about the same timeline when this issue has been raised. I tried deleting the whole add on reinstall which gave me the api error for which you had released the patch but the problem continued 

Token File:	/data/.GoSungrow/AppService_login.json
[05:10:36] INFO: Syncing data from gateway https://gateway.isolarcloud.com.hk ...
2024/02/09 05:10:36 INFO: Connecting to MQTT HASSIO Service...
2024/02/09 05:10:36 INFO: Connecting to SunGrow...
2024/02/09 05:10:37 INFO: Found SunGrow 2 devices
2024/02/09 05:10:37 INFO: Option[loglevel] set to 'info'
2024/02/09 05:10:37 INFO: Caching Sungrow metadata...
2024/02/09 05:10:40 INFO: Cached 888 Sungrow data points...
PsId: required
JSON request:	{"ps_id":1228950}

2024/02/09 05:10:41 ERROR: SetResponse: strconv.Atoi: parsing "3071048_3071049": invalid syntax
Error: SetResponse: strconv.Atoi: parsing "3071048_3071049": invalid syntax
Usage:
  GoSungrow mqtt run [flags]

Aliases:
  run, 

Examples:
	GoSungrow mqtt run

illuzn, 2024-02-12:
Not sure if its's safe to show the outputs of:
```bash
./GoSungrow api login
./GoSungrow show ps list
```
But both commands complete sucessfully and show the following general output (santitised):
```plaintext
┏━━━━━━━━━━━━━━━━━━┳━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━┓
┃ Ps Key           ┃ Ps Id   ┃ Device Type ┃ Device Code ┃ Channel Id ┃ Serial #    ┃ Factory Name ┃ Device Model   ┃
┣━━━━━━━━━━━━━━━━━━╇━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━┫
┃ psid1_1_1_1      │ psid1   │ 1           │ 1           │ 1          │ ###         │ SUNGROW      │ SG10RT         ┃
┃ psid1_22_247_1   │ psid1   │ 22          │ 247         │ 1          │ ###         │ SUNGROW      │ WiFi V31       ┃
┃ psid1_7_1_1      │ psid1   │ 7           │ 1           │ 1          │ ###         │ SUNGROW      │ SG Smart Meter ┃
┃ psid2_1_1_1      │ psid2   │ 1           │ 1           │ 1          │ ###         │ SUNGROW      │ SG30CX         ┃
┃ psid2_22_247_1   │ psid2   │ 22          │ 247         │ 1          │ ###         │ SUNGROW      │ EyeM4          ┃
┗━━━━━━━━━━━━━━━━━━┷━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━━━┷━━━━━━━━━━━━━━━━┛
```

Running `./GoSungrow mqtt run` (which seems to the be command which performs a one-shot data pull and is the one which hangs). Generates the same error above.

Now to the meat of what I've discovered.

Running `./GoSungrow mqtt run psid1` generates the strconv.Atoi error which I'm encountering above.

Running `./GoSungrow mqtt run psid2` generates the "Repeated request" error I was encountering previously.

That explains the behaviour I was experiencing previously with different errors. For whatever reason the first site it decided to pull was different between my installs.

Interestingly, I have the "EyeM4" which seems to be relatively new and may not be supported and on the other site I have a Smart Meter (which may also not be supported).

msmitka, 2024-05-30:
Hello to everybody,
i have problem with ERROR: SetResponse: strconv.Atoi: parsing "4050146_4050147": invalid syntax.
With the help of @Paraphraser  I found that this occurs with "show device list" where the error occurs. 4050146_4050147 occurs in "device_area":"4050146_4050147" in the dir cache in the devicelist file

info here: [https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37?permalink_comment_id=5072228#gistcomment-5072228](url)

Can any of you help me or fix the image?

Thank you very much Michal

lordforall, 2024-08-20:
I am getting the same error.
Does anyone know where I can look to get more information on the error and where it is happening?

[01:54:15] INFO: Login to iSolarCloud using gateway https://augateway.isolarcloud.com ...
Logging in again
Email:	xxxxxxxx@gmail.com
Create Date:	Fri Apr 21 11:01:16 CST 2023
Login Last Date:	2024-08-20 09:54:16
Login Last IP:	
Login State:	1
User Account:	a60gxxxxxx
User Id:	392360
User Name:	xxxxxxxxxxxx
Is Online:	false
Token:	392360_32423540d00347e8xxxxxxxxxxxxxx
Token File:	/data/.GoSungrow/AppService_login.json
[01:54:16] INFO: Syncing data from gateway https://augateway.isolarcloud.com ...
2024/08/20 01:54:16 INFO: Connecting to MQTT HASSIO Service...
2024/08/20 01:54:16 INFO: Connecting to SunGrow...
2024/08/20 01:54:16 INFO: Found SunGrow 3 devices
2024/08/20 01:54:17 INFO: Caching Sungrow metadata...
2024/08/20 01:54:17 INFO: Cached 1042 Sungrow data points...
PsId: required
JSON request:	{"ps_id":1219098}

2024/08/20 01:54:18 ERROR: strconv.Atoi: parsing "1423777_1423778": invalid syntax
Error: strconv.Atoi: parsing "1423777_1423778": invalid syntax
Usage:
  GoSungrow mqtt run [flags]

Aliases:
  run, 

Examples:
	GoSungrow mqtt run  

Flags: Use "GoSungrow help flags" for more info.

Additional help topics:

ERROR: strconv.Atoi: parsing "1423777_1423778": invalid syntax
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

copetogo, 2024-09-26:
i guess  no progress on this issue. :(.  i somehow think having 2 accounts one of them being shared with the main account may be the reason

lordforall, 2024-09-27:
> i somehow think having 2 accounts one of them being shared with the main account may be the reason

I had several accounts shared but I unshared them to test the theory. I still get the same error unfortunately.

Paraphraser, 2025-08-01:
Some of what I'm about to say repeats earlier posts in this issue but it will help set the scene.

```
2025/07/31 12:24:55 ERROR: SetResponse: strconv.Atoi: parsing "4783104_4783105": invalid syntax
Error: SetResponse: strconv.Atoi: parsing "4783104_4783105": invalid syntax
```

The [`strconv.Atoi()`](https://pkg.go.dev/strconv#Atoi) function expects a string which is capable of being converted to an Integer (ie an optional leading sign followed by digits).

It has been passed the string "4783104_4783105" which, because of the underscore, isn't capable of being converted to Integer representation, so the function returns an "invalid syntax" error.

Some users (eg you) hit this problem, some don't (eg me).

An earlier hypothesis was that people hitting this error had multiple accounts (or possibly multiple plants). That hypothesis seems to have been discounted so we're stuck.

I've previously searched the code looking for any "obvious places" where `strconv.Atoi()` is used in the context of `SetResponse()`. Nothing jumped out.

The situation is complicated by the fact that there is no traceback. If you don't understand that term, please take a look at [this](https://github.com/MickMake/GoSungrow/issues/136#issuecomment-3068752258). There, the first line of traceback is:

```
github.com/MickMake/GoSungrow/iSolarCloud/AppService/queryDeviceList.(*EndPoint).SetBatteryPoints(, {{, , }}, {0xc000a92c00, {0xc000885897, 0x7}, {0xc21553309baaaaf6, 0x46cf303, 0x4b5fa20}, ...})
/Users/isit/git/GoSungrow/iSolarCloud/AppService/queryDeviceList/data.go:336 +0x4d3
```

So that means we can go to the source code at the path:

```
GoSungrow/iSolarCloud/AppService/queryDeviceList/data.go
```

and, from there, to line 336 which is the line of code causing the error. You can either use a debugger to attach a breakpoint just before that line or add print statements to the code to help figure out what is going on.

Conversely, **this** "invalid syntax" problem doesn't have a traceback so it's tricky for someone who **isn't** hitting the problem to figure out what might be going on, let alone guess at a solution.

It's **possible** that you **might** get a bit more information if you run GoSungrow outside of the add-on container. I'm not really all that hopeful but it never hurts to try.

Use SSH to connect to Home Assistant. You can either `ssh` to your HA host from another host, or you can use the "Terminal" in the HA GUI side-bar.

I'm going to make the assumption that the GoSungrow add-on is **not** running (because it's crashing) but if you have it set up to auto-restart, please stop it manually.

Next, run:

```
$ docker images | grep triamazikamno
```

The response will be something like this:

```
triamazikamno/amd64-addon-gosungrow  3.0.7  f2cbc9418287   20 months ago   161MB
```

We're interested in the image ID which, in the above example, is "f2cbc9418287".

Now we are going to use the image ID to instantiate the image as a non-running container:

```
$ docker create --name temp f2cbc9418287
```

Then we will pull the GoSungrow binary out of it:

```
$ docker cp temp:/usr/local/bin/GoSungrow .
```

We don't need the container any more so get rid of it:

```
$ docker rm temp
```

The result is a copy of the GoSungrow binary in the local working directory.

Now you need to configure GoSungrow. These are the same parameters as you set up in the "Configuration" tab for the GoSungrow add-on:

```
$ ./GoSungrow config write \
	--host «sungrow_host» \
	--user «sungrow_user» \
	--password «sungrow_password» \
	--appkey «sungrow_appkey» \
	--mqtt-host «mqtt_host» \
	--mqtt-port «mqtt_port» \
	--mqtt-user «mqtt_user» \
	--mqtt-password «mqtt_password»
```

Try to login:

```
$ ./GoSungrow api login
```

If you are successful, it confirms you are using the correct AppKey. If it doesn't work, try the other AppKey.

> I did note you said you tried both AppKeys and got the same result. That's puzzling.

You can then try commands like:

```
$ GoSungrow show ps list
```

And, finally, mimic the container's command:

```
$ ./GoSungrow mqtt run
```

which should hit the same error. But, if it doesn't (which will be interesting), you can exit with <kbd>control</kbd>+<kbd>c</kbd>.

Last point. If your HA instance restarts then the copy of GoSungrow and the configuration will disappear so you'll have to redo.

Paraphraser, 2025-08-01:
My apologies. I made too many assumptions. I should have thought of that.

So, for any string enclosed in  `«guillemots»` I was intending to signal "replace this with the value in the Configuration tab of the GoSungrow add-on in Home Assistant. My *assumption* was that you would be sufficiently familiar with Unix conventions as to know when values would need to be quoted, and how to go about doing that.

The simplest way to avoid problems with special characters is to wrap the string in **single** quotes (not double quotes). The reason is that the command interpreter (aka "the shell") *will* peer inside and may "do things" with strings wrapped in double quotes but not strings wrapped in single quotes.

If you're using single quotes to enclose your strings, the only rule to remember is:

* if the string *contains* single quotes then **each** single quote needs to be handled in what might seem to be a fairly weird way with the five-character sequence `'"'"'` (one, two, one, two, one).

You can use the `echo` command to test what is going to happen. The screen shot contains examples of passwords, the first with an embedded exclamation mark, the second with that plus a single quote in the role of an apostrophe.

[image]

Here's a better example. I've wrapped every string in single quotes, except the port number (I could still have used quotes around that if I wanted to):

[image]

Finally, what `GoSungrow config write` does is to manipulate a JSON file at `~/.GoSungrow/config.json`. That is just a text file so it can be edited in any Unix text editor. If you have a complex password that is making your life hell when you try to use GoSungrow to set values in the config file, the easiest solution might be to use `GoSungrow config write` to set a placeholder value, then use a text editor to change it to the actual value. Here's a listing of the file I just created in the previous screen shot:

[image]

AndreasDi, 2025-08-02:
I was somehow familiar with Unix conventions but I forgot most of them since I did not use it the last 15 years. But while reading your feedback I start to remember.
Coming back to the topic I was able to run it with appkey 'B0455FBE7AA0328DB57B59AA729F05D8' but not with appkey 'ANDROIDE13EC118BD7892FE7AB5A3F20'.
So I continued with appkey 'B0455FBE7AA0328DB57B59AA729F05D8' and received the following message (as expected):
```
➜  ~ ./GoSungrow api login     
Email:  XXXXXXXXXXXXXXXXXXXXX
Create Date:    Mon Dec 16 20:43:56 CST 2024
Login Last Date:        2025-08-02 16:24:15
Login Last IP:
Login State:    1
User Account:   XXXXXXXXXXXXXXXXXXXXX
User Id:        XXXXXXXXXXXXXXXXXXXXX
User Name:
Is Online:      false
Token:  XXXXXXXXXXXXXXXXXXXXX
Token File:     /root/.GoSungrow/AppService_login.json
➜  ~ ./GoSungrow show ps list
┏━━━━━━━━━━━━━━━━━━┳━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┓
┃ Ps Key           ┃ Ps Id   ┃ Device Type ┃ Device Code ┃ Channel Id ┃ Serial #    ┃ Factory Name ┃ Device Model ┃
┣━━━━━━━━━━━━━━━━━━╇━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━┫
┃ 5705537_1_1_1    │ 5705537 │ 1           │ 1           │ 1          │ B2290922933 │ SUNGROW      │ SG15RT       ┃
┃ 5705537_22_247_1 │ 5705537 │ 22          │ 247         │ 1          │ B2290922933 │ SUNGROW      │ WiNet-S      ┃
┗━━━━━━━━━━━━━━━━━━┷━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━━━┷━━━━━━━━━━━━━━┛

➜  ~ ./GoSungrow mqtt run    
2025/08/02 10:25:14 INFO: Connecting to MQTT HASSIO Service...
2025/08/02 10:25:14 INFO: Connecting to SunGrow...
2025/08/02 10:25:14 INFO: Found SunGrow 2 devices
2025/08/02 10:25:14 INFO: Caching Sungrow metadata...
2025/08/02 10:25:16 INFO: Cached 1011 Sungrow data points...
PsId: required
JSON request:   {"ps_id":5705537}

2025/08/02 10:25:16 ERROR: SetResponse: strconv.Atoi: parsing "4783104_4783105": invalid syntax
Error: SetResponse: strconv.Atoi: parsing "4783104_4783105": invalid syntax
Usage:
  GoSungrow mqtt run [flags]

Aliases:
  run, 

Examples:
        GoSungrow mqtt run  

Flags: Use "GoSungrow help flags" for more info.

Additional help topics:

ERROR: SetResponse: strconv.Atoi: parsing "4783104_4783105": invalid syntax

```
So yes, it its the same error.

AndreasDi, 2025-08-02:
I search in some data ( `~ ./GoSungrow show ps [try all commands])` for the numbers 4783104 and 4783105 mentioned in the error statement:
  `ERROR: SetResponse: strconv.Atoi: parsing "4783104_4783105": invalid syntax`
  
And yes, they are included as Uuid in  DeviceName:Grid-connected point 1 and  DeviceName:1#unit:   

```
  ➜  ~ ./GoSungrow show ps list 
┏━━━━━━━━━━━━━━━━━━┳━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┓
┃ Ps Key           ┃ Ps Id   ┃ Device Type ┃ Device Code ┃ Channel Id ┃ Serial #    ┃ Factory Name ┃ Device Model ┃
┣━━━━━━━━━━━━━━━━━━╇━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━┫
┃ 5705537_1_1_1    │ 5705537 │ 1           │ 1           │ 1          │ B2290922933 │ SUNGROW      │ SG15RT       ┃
┃ 5705537_22_247_1 │ 5705537 │ 22          │ 247         │ 1          │ B2290922933 │ SUNGROW      │ WiNet-S      ┃
┗━━━━━━━━━━━━━━━━━━┷━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━━━┷━━━━━━━━━━━━━━┛

➜  ~ ./GoSungrow show ps tree
+       PsId:5705537    PsName:PVinNorderstapel PsKey:5705537_11_0_0    DeviceName:PVinNorderstapel     Uuid:4782219
+--     PsId:5705537    PsName:PVinNorderstapel PsKey:5705537_3_1_0     DeviceName:Grid-connected point 1       Uuid:4783104
+----   PsId:5705537    PsName:PVinNorderstapel PsKey:5705537_17_1_1    DeviceName:1#unit       Uuid:4783105
+------ PsId:5705537    PsName:PVinNorderstapel PsKey:5705537_1_1_1     DeviceName:Andreas 4    Uuid:4782223
+----   PsId:5705537    PsName:PVinNorderstapel PsKey:5705537_22_247_1  DeviceName:Communication Module1        Uuid:4782222
```

Mayby here is a compareable error I jumped into yesterday? -> not correct handling of '#' ?

Taking a more detailed look the '#' is used here:
```

# Available points:
┏━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━┳━━━━━━━━━━━┳━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━┓
┃ Id       ┃ Name                                                                       ┃ Unit   ┃ Unit Type ┃ Ps Id   ┃ Device Type ┃ Device Name            ┃
┣━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━╇━━━━━━━━━━━╇━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━┫
┃ p81001   │ Inverter AC power normalization                                            │ kW/kWp │ 43        │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81002   │ Inverter AC power                                                          │ kW     │ 3         │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81003   │ Inverter DC power normalization                                            │ kW/kWp │ 43        │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81004   │ Inverter DC power                                                          │ kW     │ 3         │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81005   │ Inverter efficiency                                                        │ %      │ 10        │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81007   │ Inverter total yield                                                       │ kWh    │ 7         │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81011   │ Inverter equivalent hours                                                  │ h      │ 15        │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81012   │ Daily yield by inverter                                                    │ kWh    │ 7         │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81022   │ Unit PR                                                                    │ %      │ 10        │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81023   │ Unit daily yield                                                           │ kWh    │ 7         │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81025   │ Unit yield (Theoretical)                                                   │ kW     │ 3         │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81040   │ Yp-day                                                                     │ h      │ 15        │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81044   │ Number of normal state inverters                                           │        │ 999       │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81045   │ Number of alarm state inverters                                            │        │ 999       │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81046   │ Number of fault state inverters                                            │        │ 999       │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81047   │ Number of off-line inverters                                               │        │ 999       │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81048   │ Standard health status                                                     │        │ 999       │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81050   │ Unit daily yield                                                           │ kWh    │ 999       │ 5705537 │ 17          │ 1#unit                 ┃
┃ p81052   │ Annual yield                                                               │ MWh    │ 999       │ 5705537 │ 17          │ 1#unit                 ┃

```

Paraphraser, 2025-08-05:
I'm sort of the opposite. I first encountered Unix circa 1982. It was on-again, off-again until the first half of the 90s when I started using Linux on a regular basis, ramping up to "daily use" since 2001 after Mac OS X hit the streets.

My mistake was assuming *HA knowledge* implied *Linux knowledge* when, in fact, the good folks at HA go out of their way to avoid any need for that. My bad.

<hr>

For comparison, here's mine.

```
$ GoSungrow show ps list
┏━━━━━━━━━━━━━━━━━━┳━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━┓
┃ Ps Key           ┃ Ps Id   ┃ Device Type ┃ Device Code ┃ Channel Id ┃ Serial #    ┃ Factory Name ┃ Device Model   ┃
┣━━━━━━━━━━━━━━━━━━╇━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━┫
┃ 9999999_1_1_1    │ 9999999 │ 1           │ 1           │ 1          │ B2222334445 │ SUNGROW      │ SG5.0RS        ┃
┃ 9999999_7_1_1    │ 9999999 │ 7           │ 1           │ 1          │ B2222334445 │ SUNGROW      │ SG Smart Meter ┃
┃ 9999999_22_247_1 │ 9999999 │ 22          │ 247         │ 1          │ B2222334445 │ SUNGROW      │ WiNet-S        ┃
┗━━━━━━━━━━━━━━━━━━┷━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━━━┷━━━━━━━━━━━━━━━━┛

$ GoSungrow show ps tree
+	PsId:9999999	PsName:Home	PsKey:9999999_11_0_0	DeviceName:Home	Uuid:1367847
+--	PsId:9999999	PsName:Home	PsKey:9999999_22_247_1	DeviceName:Communication Module1	Uuid:1367849
+--	PsId:9999999	PsName:Home	PsKey:9999999_1_1_1	DeviceName:Inverter1	Uuid:1367850
+	PsId:9999999	PsName:Home	PsKey:9999999_11_0_0	DeviceName:Home	Uuid:1367847
+--	PsId:9999999	PsName:Home	PsKey:9999999_22_247_1	DeviceName:Communication Module1	Uuid:1367849
+--	PsId:9999999	PsName:Home	PsKey:9999999_1_1_1	DeviceName:Inverter1	Uuid:1367850
+----	PsId:9999999	PsName:Home	PsKey:9999999_7_1_1	DeviceName:Meter1	Uuid:1367851

```

Ignoring the PsID prefixes, I think of them like this:

* `1_1_1` is the inverter
* `7_1_1` is the meter
* `11_0_0` is the plant
* `22_247_1` is the WiFi dongle

I don't understand why the plant (`11_0_0`) isn't mentioned in the `show ps list` because it would seem to be a higher-level concept than the "devices" that comprise a "plant". But that might be an artefact of GoSungrow rather than something inherent to the iSolarCloud API.

In your case, `1_1_1` is your inverter, `22_247_1` your WiFi, `11_0_0` your plant, but what the heck `3_1_0` and `17_1_1` mean is a mystery. Does it make sense to you?

Am I correct that "PVinNorderstapel" translates as "PV in northern stack"? Stack of panels? Stack of batteries? Is the "in" intended to mean a direction of energy flow, as in "PV energy going into the northern stack"?

As you can see, I don't have anything like that. On the other hand, my inverter is still "grid tied" so you'd think I would have a `3_1_0`.

All very mysterious.

But to answer your question, I doubt that the embedded `#` is the cause of any problems.

<hr>

As I understand it, the GoSungrow add-on for HA fetches **everything** that it is possible to fetch from iSolarCloud and transcodes the data points into MQTT messages. On the HA side you, the user, pick and choose what you want to monitor or display.

Of necessity, an "**everything**" strategy means all the associated fetching code has got to be bullet-proof. Fairly obviously, it isn't. And the fact that some people can get GoSungrow to work while others can't is almost certainly explained by equipment differences. My simple non-hybrid setup just doesn't exercise the parts of GoSungrow that are failing for you.

I don't actually use either HA or the GoSungrow add-on for anything beyond trying to help out here (usually with not much success). Getting GoSungrow working properly is either going to take MickMake coming back or someone with the relevant skills forking the project.

> I don't have any Go skills so that isn't going to be me.

I've done code searches to see if I can pin down where that error message is coming from. I can't. There are plenty of hits on "invalid syntax" but I've yet to spot that in conjunction with an Atoi call.

If this error was causing problems for me and I was only armed with the tools I have (basically a Mac with the Go compiler and the various chunks of source code needed to compile GoSungrow), I think I would approach the problem like this. The code seems to be distributed across three folders:

```
$HOME
├── go
└── go-projects
    ├── GoUnify
    └── MickMake
```

I'd search that structure like this:

```
for P in $HOME/go $HOME/go-projects/GoUnify $HOME/go-projects/MickMake ; do
  find "$P" -name "*.go" -exec grep -l "invalid syntax" {} \; | sort | uniq
done
```

That produces a list of 18 files. Some of those hits lead to comments like:

```
// invalid syntax
```

but most are strings that can turn up in errors. I'd change each such string to give it a sequence number so every "invalid syntax" message became unique, as in "invalid syntax 1" then 2 and so on.

Then I'd compile the program and run it. If the error changed to include a sequence number I'd at least know where in the code it was coming from.

> If I still got the vanilla message, I'd conclude it was coming from something outside that scope (presumably in some library installed as part of the compiler), in which case I'd change tack and maybe search for `Atoi` (118 files) or `parsing` (262 files) or `SetResponse` (986 files).

Assuming I could nail down the line of code where the message was coming from, I'd insert a call to `panic()` which, according to the documentation, would trigger a traceback. The traceback would give me a detailed view of the lines of code involved and that, in turn, would help lead to the associated data structures and, hopefully, whatever element is currently defined as "integer" which should probably be defined as a string.

But whether this helps or not ... 🤷‍♂️

AndreasDi, 2025-09-24:
Some points and feedback:

### **(1)**

In your case, 

> 1_1_1 is your inverter 	-> yes  
> 22_247_1 your WiFi     -> yes  
> 11_0_0 your plant  	-> yes 

unclear:

> 3_1_0   : DeviceName:Grid-connected point 1     Uuid:4783104 
> 17_1_1 : DeviceName:1#unit                                Uuid:4783105

If you go on the webpage of your iSolarCloud you will find it under Devices -> Settings -> "Device Location" in the upper coloum.
There my entry is "**Grid-connected point 1_1#unit**".
I am not aware that I generated these entry during the setup phase so it seems to be automatically generated.
If you think about the error message: `parsing "4783104_4783105": invalid syntax` and take a look into the Uuid this is exactly the root cause at least for this error message. The current content "**Grid-connected point 1_1#unit**" is tried to be splitted up into two parts like 4783104_4783105 which does not work.
So currently it seems that the "**Grid-connected point 1_1#unit**" should end up in the content of 3_1_0 and 17_1_1.

### **(2)**
"PVinNorderstapel" is just the name of my plant. It means something like “PV at location A,” where Norderstapel is the place where I grew up.
So don't worry about it.

### **(3)** 
I try to dive into some details and search for "atoi" usage in all *.go files of this project and found it appering over all 5 times in the following four files (not taking the BACKUP into account):

> Suche "atoi" (8 Treffer in 6 Dateien von 2514 gesucht) [Erweitert]
>   C:\Users\Dell Optoplex 3060\Downloads\GoSungrow-master\iSolarCloud\api\GoStruct\valueTypes\integers.go (2 Treffer)
> 	Zeile 104: 		v, t.Error = strconv.Atoi(t.string)
> 	Zeile 224: 		v, t.Error = strconv.Atoi(t.string)
>   C:\Users\Dell Optoplex 3060\Downloads\GoSungrow-master\iSolarCloud\api\GoStruct\valueTypes\point.go (1 Treffer)
> 	Zeile 251: 		v, t.Error = strconv.Atoi(t.string)
>   C:\Users\Dell Optoplex 3060\Downloads\GoSungrow-master\iSolarCloud\api\GoStruct\valueTypes\uuid.go (1 Treffer)
> 	Zeile  93: 		v, t.Error = strconv.Atoi(t.string)
>   C:\Users\Dell Optoplex 3060\Downloads\GoSungrow-master\iSolarCloud\api\GoStruct\valueTypes\uv.go (1 Treffer)
> 	Zeile  457: 		v, t.Error = strconv.Atoi(t.StringValue)
>   C:\Users\Dell Optoplex 3060\Downloads\GoSungrow-master\iSolarCloud\api\GoStruct\valueTypes\uv.goBACKUP (1 Treffer)
> 	Zeile 403: 		v, t.Error = strconv.Atoi(t.StringValue)
>   C:\Users\Dell Optoplex 3060\Downloads\GoSungrow-master\iSolarCloud\api\GoStruct\valueTypes\uv.goBACKUP2 (2 Treffer)
> 	Zeile 412: 		v, t.Error = strconv.Atoi(t.StringValue)
> 	Zeile 762: 				val, err := strconv.Atoi(key)
> 

Since I never programmed GO programming language (I was even not aware about it existance) I ask AI to improve it:
I have a go Script which may have a wrong input like: "1559776_1559777". 
Please modify the following go-Sript to handle this wrong input correctly. 

```
func (t *Integer) SetString(value string) Integer {
	for range Only.Once {
		t.string = value
		t.int64 = 0
		t.Valid = false

		if value == "" {
			break
		}

		if value == "--" {
			// value = ""
			break
		}

		var v int
		v, t.Error = strconv.Atoi(t.string)
		if t.Error != nil {
			break
		}
		t.int64 = int64(v)
		t.Valid = true
	}

	return *t
}
```

// To make your Go script more robust and gracefully handle malformed inputs like "1559776_1559777", you'll want to ensure it:
// Detects invalid formats (e.g., strings containing underscores or other non-numeric characters).
// Returns an appropriate error or logs a warning.
// Keeps t.Valid = false and doesn't attempt conversion if input is invalid.

```
func (t *Integer) SetString(value string) Integer {
	for range Only.Once {
		t.string = value
		t.int64 = 0
		t.Valid = false
		t.Error = nil

		if value == "" || value == "--" {
			break
		}

		// Check for valid integer format using a regular expression
		// This ensures no underscores or non-digit characters are present
		matched, err := regexp.MatchString(`^-?\d+$`, value)
		if err != nil {
			t.Error = fmt.Errorf("regex error: %v", err)
			break
		}
		if !matched {
			t.Error = fmt.Errorf("invalid integer format: %q", value)
			break
		}

		// Safe to parse
		var v int
		v, t.Error = strconv.Atoi(value)
		if t.Error != nil {
			break
		}

		t.int64 = int64(v)
		t.Valid = true
	}

	return *t
}
```

// Explanation of Changes
// Regex Check: Ensures the string is strictly a valid integer, allowing optional - for negatives. Blocks strings like "1559776_1559777" or "12abc".
// Error Handling: If the input is invalid or regex fails, t.Error is set with a meaningful message.
// Cleans up assignment: Ensures t.Error is cleared initially.

Since in the GO language files have to be compiled is there anyone how is able to verify the modification and compile it?
The other way is to find out where and how to change the content of Devices -> Settings -> "Device Location" 

Best regards,
Andreas