MickMake/GoSungrow #109: Issue with Energy dashboard in HA

grf692, 2024-03-11:
Hi,

I have an issues I don't manage to fix and wish to ask if one of you had the same behavior maybe.

In the Energy dashboard settings page, I've set the "today_energy" sensor to be used as Solar production.

When I look at the history of this sensor in HA I get the behavior I believe I should expect. That is everyday at midnight it is reset to 0 as you can see in the screenshot here:

[image]

But in my Energy Dashboard I shows as negative value everyday:

[image]

May anyone give me a hint on how to fix this please?
Or what is the full name of the "sensor" you use? (I am in Europe/Sweden)
Mine is GoSungrow xxx - SUNGROW GoSungrow getPsDetail.xxx.today_energy

Thanks!

JohnFCreep, 2024-03-12:
That is because the sensor is resetting every day, but because it doesn't update before 0:00 Home Assistant calculates like that.

You can create an Utility Meter with "Periodically resetting" to address that.

grf692, 2024-03-13:
Thanks, I was not aware of these "creatures"!
Now I've set a helpers as utility meter and it seems to work like a charm.

closed 2024-03-13 by grf692

mazvone, 2024-09-28:
How did you that, can you write step by step?

JohnFLoki, 2024-11-09:
@mazvone You go to settings -> Devices & Services -> Helpers (topside) -> create Helper -> Utility Meter ->
Name: you do you
Input sensor: sensor.[...]today_energy
...
Periodically resetting: yes

Then add it to your energy dashboard when it so its first value.