MickMake/GoSungrow #113: Update for the Lovelace examples doco

igoratencompass, 2024-04-25:
@Paraphraser I was looking at the Lovelace files you have generously provided and just a small note about [Lovelace examples](https://github.com/MickMake/HomeAssistantAddons/blob/main/GoSungrow/DOCS.md#lovelace-examples) section. At least in my case, apart from replacing the `ps_id` and `ps_keys` with my own I think I have to replace `p*` points as well. For example in your case the point for Insulation Resistance is `p13160` but for me it is `p94`:

```
┃ p94      │ Array Insulation Resistance   │ kΩ     │ 23        │ ps_id │ 1           │ Inverter1             ┃
```
Does it make sense?

And thanks for the great job!

igoratencompass, 2024-04-25:
Ah, reading the standalone doc I can see the `ps_keys` refers to the whole string including the `p*` part. Sorry my bad.

closed 2024-04-25 by igoratencompass

Paraphraser, 2024-04-25:
@igoratencompass I am afraid that you are mistaken about my role. I don't have anything to do with the creation of GoSungrow or any of the documentation or files on this site. That was all done by @MickMake - that's the person you should be thanking.

The problem is that MickMake hasn't been active recently. That's why I've started to try to help out by answering questions. But what I have been doing is only a stopgap.